Koenkk/zigbee2mqtt #14651: [New device support]:  Hilo Stelpro thermostat HT402 trying to gather the wattage used

FrancoLoco, 2022-10-25:
### Link

https://www.hiloenergie.com/en-ca/products/energie/thermostat-intelligent-hilo-thermostats_1/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xbc33acfffe3623b5","nwkAddr":24360,"manufId":4759,"manufName":"Stello","powerSource":"Unknown","modelId":"HT402","epList":[25],"endpoints":{"25":{"profId":260,"epId":25,"devId":769,"inClusterList":[0,3,4,513,516,1029],"outClusterList":[3,10,1026],"clusters":{"genBasic":{"attributes":{"modelId":"HT402","manufacturerName":"Stello","powerSource":0,"zclVersion":3,"appVersion":23,"stackVersion":21,"hwVersion":1,"dateCode":"20000000 00000\u0000"}},"hvacThermostat":{"attributes":{"16386":0,"16392":0,"16393":460,"systemMode":4,"StelproSystemMode":4,"occupiedHeatingSetpoint":900,"localTemp":1800,"pIHeatingDemand":0}},"hvacUserInterfaceCfg":{"attributes":{"keypadLockout":0}},"seMetering":{"attributes":{}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1},{"cluster":516,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d15f52","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":28,"minRepIntval":10,"maxRepIntval":3600,"repChange":null},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":516,"attrId":1,"minRepIntval":10,"maxRepIntval":3600,"repChange":null}],"meta":{}}},"appVersion":23,"stackVersion":21,"hwVersion":1,"dateCode":"20000000 00000\u0000","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666715319736,"defaultSendRequestWhen":"immediate"}

### Comments

I have tried setting up a converter for this thermostat, which is essentially a rebranded Stelpro thermostat. I was mostly successful, but I would like to be able to gather the wattage used.  I know this device can do it, cause I can see the instant watts used when I use the official bridge and app.

From what I have seen in the logs, this info seems to be in an attribute of the hvacThermostat:
_Debug 2022-10-25 12:40:32Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,**"16392":1841**,"16393":534}' from endpoint 25 with groupID 0_

My assumption is that: 
- 16392 = instant wattage
- 16393 = increasing consumption wH

How could I map this information to the power of the device?   

--
full log:
--
_2022-10-25 12:38:42Received MQTT message on 'zigbee2mqtt/0xbc33acfffe3623b5/set' with data '{"occupied_heating_setpoint":20}'
Debug 2022-10-25 12:38:42Publishing 'set' 'occupied_heating_setpoint' to '0xbc33acfffe3623b5'
Info 2022-10-25 12:38:42MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":81,"local_temperature":18,"occupied_heating_setpoint":20,"pi_heating_demand":7,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:38:42Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2000}' from endpoint 25 with groupID 0
Info 2022-10-25 12:38:42MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":81,"local_temperature":18,"occupied_heating_setpoint":20,"pi_heating_demand":7,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:38:46Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"pIHeatingDemand":100}' from endpoint 25 with groupID 0
Info 2022-10-25 12:38:46MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":81,"local_temperature":18,"occupied_heating_setpoint":20,"pi_heating_demand":39,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:38:47Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":0,"16392":1841,"16393":478}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:39:17Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":0,"16392":1841,"16393":497}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:40:02Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,"16392":1841,"16393":515}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:40:13Received MQTT message on 'zigbee2mqtt/0xbc33acfffe3623b5/set' with data '{"occupied_heating_setpoint":18.5}'
Debug 2022-10-25 12:40:13Publishing 'set' 'occupied_heating_setpoint' to '0xbc33acfffe3623b5'
Info 2022-10-25 12:40:13MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":81,"local_temperature":18,"occupied_heating_setpoint":18.5,"pi_heating_demand":39,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:40:13Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1850}' from endpoint 25 with groupID 0
Info 2022-10-25 12:40:13MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":81,"local_temperature":18,"occupied_heating_setpoint":18.5,"pi_heating_demand":39,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:40:32Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,"16392":1841,"16393":534}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:41:02Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,"16392":1841,"16393":552}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:41:09Received MQTT message on 'zigbee2mqtt/0xbc33acfffe3623b5/set' with data '{"occupied_heating_setpoint":18}'
Debug 2022-10-25 12:41:09Publishing 'set' 'occupied_heating_setpoint' to '0xbc33acfffe3623b5'
Info 2022-10-25 12:41:09MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":75,"local_temperature":18,"occupied_heating_setpoint":18,"pi_heating_demand":39,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:41:09Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1800}' from endpoint 25 with groupID 0
Info 2022-10-25 12:41:09MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":75,"local_temperature":18,"occupied_heating_setpoint":18,"pi_heating_demand":39,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:41:13Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"pIHeatingDemand":61}' from endpoint 25 with groupID 0
Info 2022-10-25 12:41:13MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe3623b5', payload '{"keypad_lockout":"unlock","linkquality":78,"local_temperature":18,"occupied_heating_setpoint":18,"pi_heating_demand":24,"power":null,"running_state":"heat","system_mode":"heat"}'
Debug 2022-10-25 12:41:13Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,"16392":1123,"16393":552}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:41:59Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,"16392":1123,"16393":570}' from endpoint 25 with groupID 0
Debug 2022-10-25 12:42:29Received Zigbee message from '0xbc33acfffe3623b5', type 'attributeReport', cluster 'hvacThermostat', data '{"16386":1,"16392":1104,"16393":570}' from endpoint 25 with groupID 0
[... cut ...]
		await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
		await reporting.thermostatSystemMode(endpoint);
		await reporting.thermostatPIHeatingDemand(endpoint);
		await reporting.thermostatKeypadLockMode(endpoint);
		
		// cluster 0x0201 attribute 0x401c
		await endpoint.configureReporting('hvacThermostat', [{
			attribute: 'StelproSystemMode',
			minimumReportInterval: constants.repInterval.MINUTE,
			maximumReportInterval: constants.repInterval.HOUR,
			reportableChange: 1,
		}]);

	},
};

module.exports = definition;
```

FrancoLoco, 2022-10-26:
I kept fiddling and managed to get it working.. gonna keep testing this in the next days then I will probably submit my PR.

const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const stello = {
    fz: {
          stello_power: {
            cluster: 'hvacThermostat',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                if (msg.data.hasOwnProperty('16392')) {
                    return {stello_power: msg.data["16392"]};
                }
            },
		  },
		  
		  stello_energy: {
            cluster: 'hvacThermostat',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                if (msg.data.hasOwnProperty('16393')) {
                    return {stello_energy: parseFloat(msg.data["16393"]) / 1000};
                }
            },
        },
    },
};

const definition = {
	zigbeeModel: ['HT402'], 
	model: 'HT402', 
	vendor: 'Stello', 
	description: 'Hilo Stelpro thermostat', 
	fromZigbee: [fz.legacy.stelpro_thermostat, fz.legacy.hvac_user_interface, stello.fz.stello_power, stello.fz.stello_energy],  
	toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint,
		tz.thermostat_temperature_display_mode, tz.thermostat_keypad_lockout, tz.thermostat_system_mode,
		tz.thermostat_running_state, tz.stelpro_thermostat_outdoor_temperature],
	exposes: [e.local_temperature(), e.keypad_lockout(), e.power().withAccess(ea.STATE_GET).withProperty('stello_power'), e.energy().withAccess(ea.STATE_GET).withProperty('stello_energy'), 
		exposes.climate().withSetpoint('occupied_heating_setpoint', 5, 30, 0.5).withLocalTemperature()
			.withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat'])],
	configure: async (device, coordinatorEndpoint, logger) => {
		const endpoint = device.getEndpoint(25);
		const binds = ['genBasic', 'genIdentify', 'genGroups', 'hvacThermostat', 'hvacUserInterfaceCfg', 'msTemperatureMeasurement'];
		await reporting.bind(endpoint, coordinatorEndpoint, binds);

		// Those exact parameters (min/max/change) are required for reporting to work
		await reporting.thermostatTemperature(endpoint);
		await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
		await reporting.thermostatSystemMode(endpoint);
		await reporting.thermostatPIHeatingDemand(endpoint);
		await reporting.thermostatKeypadLockMode(endpoint);
		
		// cluster 0x0201 attribute 0x401c
		await endpoint.configureReporting('hvacThermostat', [{
			attribute: 'StelproSystemMode',
			minimumReportInterval: constants.repInterval.MINUTE,
			maximumReportInterval: constants.repInterval.HOUR,
			reportableChange: 1,
		}]);

	},
};

module.exports = definition;

closed 2022-11-07 by FrancoLoco

RayLation, 2023-01-16:
salut Franco,

Do you sometimes have problems with the power property? I have had cases where the power value received (or displayed) by zigbee2mqtt would make no sense for several days because it is too low even if the heat is on full-time and I set the temperature for 30c. The only solution I found is to reset the thermostats by closing the breaker.

[image]

FrancoLoco, 2023-01-17:
Moi je n’ai aucun problème mais je suspecte que tu as peut être un meuble  devant le calorifère ou un calorifère pas assez puissant ou quelque chose qui fait en sorte que la température ne monte pas et alors le thermostat se met en mode surchauffe et diminue la puissance envoyée au calorifère pour éviter un potentiel de feu

RayLation, 2023-01-17:
hmm intéressant, mais pourquoi ca ferait ca pendant plusieurs jours? Et pourquoi il reviendrait à la normale 10s après avec un reset? (il serait pourtant encore chaud)

Je vais monitorer pour voir si c'est toujours les mêmes calorifères. Ils n'ont pas changé depuis des années, donc ils sont peut-être en train de s'en aller...

Au fait, c'est quel attribut de quel cluster qu'il faut poller pour avoir le power? Pas réussi à trouver à quoi correspond "16392" à partir de la "dev console" de z2m

FrancoLoco, 2023-01-17:
Ouin on ne peut pas poller.. ces thermostats ne font que pusher les
changements de power…

RayLation, 2023-02-04:
Simple petit avancement là-dessus... on voit que le thermostat est à "idle" dans Z2M. C'est corroboré par l'écran du thermostat et aussi corroboré par ma conso générale de la maison au même moment qui est ~nulle.

On voit que le "power" est collé à 2688w depuis plusieurs heures et qu'il saute de 0 à 2688 à 0 depuis quelques jours, alors qu'il y avait une belle courbe avant.

Je pense vraiment que les thermostats ont un bug intermittent... Même avec un reboot (fermé le breaker) il veut pas décoller...

[image]

[image]

Emilef22, 2023-11-22:
As tu finalement réussi à résoudre le problème? Merci

nxor, 2024-01-08:
Salut, je viens davoir le probleme sur un de mes thermostats... moi cest seuelement le "Energy" reported qui est stuck a 499.9 meme apres un reset du breaker...

avez vous trouver une solution ?

nxor, 2024-01-08:
Je pense avoir fixer le miens en poussant une nouvelle configuration des setting specifique.

Energy calibration: J'ai mis: 1
Energy precision : J'ai mis: 1

[image]

RayLation, 2024-01-13:
FrancoLoco m'a confirfmé que c'est lorsque le "energy" atteint 500 que les 2 valeurs gèlent.
C'est ce qui se passe ici aussi pour mon thermostat sous-sol qui a gelé à 259w et 500kwh depuis plus de 2 semaines:

[image]

J'ai essayé torn truc avec les settings specific et ca ne fonctionne pas ici.

francispoisson, 2024-01-24:
Quelqu'un a trouvé la solution pour la puissance du thermostat qui bloque lorsque l'énergie totale arrive à 499.99 kWh?

nxor, 2024-01-24:
De mon côté changé les settings spécifiques semblent avoir résolu le problème. Tout va bien depuis...

J'avais aussi éteint le Breaker quelques secondes...

Mais si ne fonctionne pas chez vous, je ne saurais pas quoi faire d'autre.

FrancoLoco, 2024-01-24:
> Quelqu'un a trouvé la solution pour la puissance du thermostat qui bloque lorsque l'énergie totale arrive à 499.99 kWh?

Pour moi présentement la solution officielle de contournement c'est de fermer le breaker et le repartir.