Koenkk/zigbee2mqtt #14665: [New device support]: Zigbee Model TS0202 Zigbee Manufacturer _TZ3000_y79r2ymz

mgmtotorsperformance, 2022-10-27:
### Link

https://allegro.pl/oferta/mini-czujnik-pir-ruchu-obecnosci-zigbee-wx-tuya-10758878346

### Database entry

{"id":21,"type":"EndDevice","ieeeAddr":"0xa4c138e1873a11d4","nwkAddr":54770,"manufId":4417,"manufName":"_TZ3000_y79r2ymz","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b00258bf905","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666855873716,"defaultSendRequestWhen":"immediate"}

### Comments

Added to zigbee2mqtt but shows as not supported

Friendly name
czujnik obecnosci
Description
Last seen
N/A
Availability
[Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
EndDevice
Zigbee Model
TS0202
Zigbee Manufacturer
_TZ3000_y79r2ymz
Support status
Not supported

IEEE Address
0xa4c138e1873a11d4
Network address
0xD5F2
Power
Battery 
Interview completed
True

Koenkk, 2022-10-27:
Could you check if the device works with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_jytabjkb'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_lltemgsf'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_y79r2ymz'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_mg4dy6z6'}],
    model: 'TS0202_1',
    vendor: 'TuYa',
    description: 'Motion sensor',
    // Requires alarm_1_with_timeout https://github.com/Koenkk/zigbee2mqtt/issues/2818#issuecomment-776119586
    fromZigbee: [fz.ias_occupancy_alarm_1_with_timeout, fz.battery, fz.ignore_basic_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.linkquality(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works

mgmtotorsperformance, 2022-10-27:
it changed to supported only in exposes can see only this:

occupancy
Indicates whether the device detected occupancy
N/A
battery_low
Indicates if the battery of this device is almost empty
N/A
linkquality
Link quality (signal strength)
102lqi
battery
Remaining battery in %
N/A%
voltage
Voltage of the battery in millivolts
N/AmV

mgmtotorsperformance, 2022-10-27:
ok after battery removal and installation all works, thank you

closed 2022-10-27 by mgmtotorsperformance