Koenkk/zigbee2mqtt #14666: Sonoff SNZB-02 round battery

leroyloren, 2022-10-27:
### What happened?

Device battery status is expsosed with a decimal place.

### What did you expect to happen?

For the value to be displayed as an integer

### Zigbee2MQTT version

1.28.0edge

### Adapter firmware version

20220219

### Adapter

zzh!

### Debug log

{
    "battery": **68.5**,
    "humidity": 48,
    "last_seen": "2022-10-27T11:17:41+02:00",
    "linkquality": 87,
    "temperature": 25.1,
    "voltage": 2900
}

value 68.5 => 69

Koenkk, 2022-10-27:
What is wrong with this?

leroyloren, 2022-10-27:
it's not bad, but it would be good to use a value without decimal places.

closed 2022-12-05 by github-actions[bot]