Koenkk/zigbee2mqtt #14670: [Feature request]: Add identifier to a group created in Zigbee2MQTT so that you can identify Zigbee2MQTT groups in Home Assistant.

smarthomejunkie, 2022-10-27:
### Is your feature request related to a problem? Please describe

I wanted to create a template sensor that shows all the lights that are turned on. The thing is that you cannot derive single lightbulbs from light groups if you have creates the light groups in Zigbee2MQTT.

### Describe the solution you'd like

I would like to have an attribute in the Zigbee2MQTT light groups in Home Assistant to be able to filter out light groups in template sensors.

Something like:

is_group: true

### Describe alternatives you've considered

One option is to give light groups a name with the word Group in it and filter on names containing that word. But this feels a little bit iffy to me.

### Additional context

none

danieledwardgeorgehitchcock, 2022-10-27:
Looks like the base work for this is already in Z2M. See [here](https://github.com/Koenkk/zigbee2mqtt/blob/03ba647dc6b5f299f8f3ab441712999fcb3a253e/lib/model/group.ts#L32). I wonder how much extra work is needed (if any) would be required to expose this via Z2M MQTT statuses..

stefanuytterhoeven, 2022-11-01:
Would like to see this implement too! 😄😄
Any timeline?

KevinCathcart, 2022-11-01:
The timeline would be whenever Koenkk decides to implement it, unless somebody provides a pull request first.

My proposal would be something along the lines of adding code like the following to controller.ts's `publishEntityState` method:

```typescript
if (entity. isGroup()) {
  message['group_id']= entity.ID;
}
```
That would not only make it possible to determine that a topic is a group (by the presence of the property), but also provides the group number, so you don't need basically join against the contents of `bridge/group` to determine this.

This does not technically require a change to the core, as it could be done in an extension (built-in or custom) like so (not tested, so there may be typos):
```js
class GroupIDExtension{
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.eventBus = eventBus;
    }

    start() {
    }
    
    stop() {
        this.eventBus.removeListenersExtension(this);
    }
    
    adjustMessageBeforePublish(entity, message) {
        if (entity. isGroup()) {
            message['group_id']= entity.ID;
        }
    }
}

module.exports = GroupIDExtension;
```

JoramHhub, 2022-12-02:
It would be great if this also adds a list of included lights as an attribute.

smarthomejunkie, 2022-12-02:
Please do not close this feature request. It would be great if this can be added! Really missing this functionality atm.

stefanuytterhoeven, 2022-12-02:
I vote for it

closed 2023-01-10 by github-actions[bot]