Koenkk/zigbee2mqtt #14681: Tuya TS0002_switch_module does not trigger individual channels

krissam44, 2022-10-27:
### What happened?

I'm using Z2M 1.28.
I've installed Tuya TS0002_switch_module  which is well recognized.
My issue is when I switch on one channel, the second channel switch on at the same time.
I've read that this issue is supposed to be resolved.
How can I solve it ?

Regards

### Zigbee2MQTT version

1.28.0 commit: 03ba647d

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

info  2022-10-28 00:07:39: MQTT publish: topic 'zigbee2mqtt/ZLinky', payload '{"active_register_tier_delivered":"TH..","apparent_power":700,"available_power":60,"current_summ_delivered":51422,"current_tarif":"BASE","device":{"applicationVersion":5,"dateCode":"20220217","friendlyName":"ZLinky","hardwareVersion":1,"ieeeAddr":"0x00158d0005d2a782","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":20392,"powerSource":"Mains (single phase)","softwareBuildID":"4000-0005","stackVersion":2,"type":"Router","zclVersion":3},"linkquality":255,"meter_serial_number":"061861455877","rms_current":3,"rms_current_max":90,"update":{"progress":29.83,"remaining":1285,"state":"updating"},"update_available":false}'
info  2022-10-28 00:07:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'ZLinky' at 29.83%, â‰ˆ 21 minutes remaining","meta":{"device":"ZLinky","progress":29.83,"status":"update_progress"},"type":"ota_update"}'
debug 2022-10-28 00:07:46: Received MQTT message on 'zigbee2mqtt/0xa4c1384b7dcab88f/set' with data '{"state_left":"ON"}'
debug 2022-10-28 00:07:46: Publishing 'set' 'state' to '0xa4c1384b7dcab88f'
info  2022-10-28 00:07:46: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b7dcab88f', payload '{"device":{"applicationVersion":68,"dateCode":"","friendlyName":"0xa4c1384b7dcab88f","hardwareVersion":1,"ieeeAddr":"0xa4c1384b7dcab88f","manufacturerID":4417,"manufacturerName":"_TZ3000_jl7qyupf","model":"TS0012_switch_module","networkAddress":20327,"powerSource":"Mains (single phase)","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":255,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"ON","state_right":"OFF","switch_type":"toggle"}'
debug 2022-10-28 00:07:46: Received Zigbee message from '0xa4c1384b7dcab88f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
info  2022-10-28 00:07:46: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b7dcab88f', payload '{"device":{"applicationVersion":68,"dateCode":"","friendlyName":"0xa4c1384b7dcab88f","hardwareVersion":1,"ieeeAddr":"0xa4c1384b7dcab88f","manufacturerID":4417,"manufacturerName":"_TZ3000_jl7qyupf","model":"TS0012_switch_module","networkAddress":20327,"powerSource":"Mains (single phase)","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":255,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"ON","state_right":"OFF","switch_type":"toggle"}'
debug 2022-10-28 00:07:47: Received Zigbee message from '0xa4c1384b7dcab88f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID null
info  2022-10-28 00:07:47: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b7dcab88f', payload '{"device":{"applicationVersion":68,"dateCode":"","friendlyName":"0xa4c1384b7dcab88f","hardwareVersion":1,"ieeeAddr":"0xa4c1384b7dcab88f","manufacturerID":4417,"manufacturerName":"_TZ3000_jl7qyupf","model":"TS0012_switch_module","networkAddress":20327,"powerSource":"Mains (single phase)","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":255,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"ON","state_right":"ON","switch_type":"toggle"}'
debug 2022-10-28 00:07:48: Received MQTT message on 'zigbee2mqtt/0xa4c1384b7dcab88f/set' with data '{"state_right":"OFF"}'
debug 2022-10-28 00:07:48: Publishing 'set' 'state' to '0xa4c1384b7dcab88f'
info  2022-10-28 00:07:48: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b7dcab88f', payload '{"device":{"applicationVersion":68,"dateCode":"","friendlyName":"0xa4c1384b7dcab88f","hardwareVersion":1,"ieeeAddr":"0xa4c1384b7dcab88f","manufacturerID":4417,"manufacturerName":"_TZ3000_jl7qyupf","model":"TS0012_switch_module","networkAddress":20327,"powerSource":"Mains (single phase)","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":255,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"ON","state_right":"OFF","switch_type":"toggle"}'
debug 2022-10-28 00:07:48: Received Zigbee message from '0xa4c1384b7dcab88f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2022-10-28 00:07:48: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b7dcab88f', payload '{"device":{"applicationVersion":68,"dateCode":"","friendlyName":"0xa4c1384b7dcab88f","hardwareVersion":1,"ieeeAddr":"0xa4c1384b7dcab88f","manufacturerID":4417,"manufacturerName":"_TZ3000_jl7qyupf","model":"TS0012_switch_module","networkAddress":20327,"powerSource":"Mains (single phase)","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":255,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"OFF","state_right":"OFF","switch_type":"toggle"}'
debug 2022-10-28 00:07:49: Received Zigbee message from '0xa4c1384b7dcab88f', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID null
info  2022-10-28 00:07:49: MQTT publish: topic 'zigbee2mqtt/0xa4c1384b7dcab88f', payload '{"device":{"applicationVersion":68,"dateCode":"","friendlyName":"0xa4c1384b7dcab88f","hardwareVersion":1,"ieeeAddr":"0xa4c1384b7dcab88f","manufacturerID":4417,"manufacturerName":"_TZ3000_jl7qyupf","model":"TS0012_switch_module","networkAddress":20327,"powerSource":"Mains (single phase)","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":255,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"OFF","state_right":"OFF","switch_type":"toggle"}'

Koenkk, 2022-10-29:
Can you provide the data/database.db entry of your device?

krissam44, 2022-12-01:
Sorry, Ididn(t see your request.
Here is the database.db (I've added .txt to be able to upload).
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/10135587/database.db.txt)

Thank for your help

Koenkk, 2022-12-02:
Does the issue still occur with z2m 1.28.4?

theklaoww, 2022-12-03:
i also have this problems

krissam44, 2022-12-04:
Yes, I've just updated to 1.28.4 and I still have the problem.

theklaoww, 2022-12-06:
i updated zigbee2mqtt 
it's fixed my issue

krissam44, 2022-12-06:
Hello, I've just updated and now it's OK.
Thank you for your VERY GOOG JOB!

closed 2023-01-13 by github-actions[bot]