Koenkk/zigbee2mqtt #14682: [Feature request]: OSRAM AC01353010G SMART+ Motion Sensor, add battery power reporting

aioue, 2022-10-28:
### Is your feature request related to a problem? Please describe

I have 4 of these sensors. In the Dev console, they all report `batteryVoltage` in a unit of `volts*10`, but battery level is not present in the State so they get a `?` in the Power column.

### Describe the solution you'd like

Use the already available `batteryVoltage` to show battery status on these already supported devices.

Here are Dev console responses:
```
Endpoint: 1, Cluster: PowerCfg

Sensor1: Read result of 'genPowerCfg': {"batteryVoltage":25,"batteryAlarmState":0}
Sensor2: Read result of 'genPowerCfg': {"batteryVoltage":23,"batteryAlarmState":8}
Sensor3: Read result of 'genPowerCfg': {"batteryVoltage":27,"batteryAlarmState":0}
Sensor4: Read result of 'genPowerCfg': {"batteryVoltage":25,"batteryAlarmState":0}
```

The voltages are 2.5V, 2.3V, 2.7V, 2.5V respectively (Volts * 10).

The batteries are a single lithium CR2 with 3V capacity. The units stop functioning around 1.9V so an acceptable simple calculation for battery % is

`(batteryVoltage - 20) *10`

Less important, but the `batteryAlarmState` is `> 0` when `batteryVoltage < 24`

### Describe alternatives you've considered

I have tried to figure out what code changes to make but do quite understand yet.

### Additional context

This commit removed a low battery and buggy tamper flag. https://github.com/Koenkk/zigbee-herdsman-converters/commit/152abf0e14017631d25769860254b73a4e048d0d

danieledwardgeorgehitchcock, 2022-10-28:
Just slung something together.. Could you try this [external converter](https://gist.github.com/danieledwardgeorgehitchcock/62867b046100b3ba3f6c1e8e1da0efa7#file-osram_ac01353010g-js) and let me know how you get on?

aioue, 2022-10-30:
# External Converter Test

1. Placed osram exernal converter .js file into z2m data directory
2. Added to external converters list in settings
3. Restarted z2m and confirmed file loaded in logs
4. Battery changed from `?` to values as soon as motion was detected (less than 1 minute)

# Results

`voltage` (millivolts) **SUCCESS**, converted correctly to state
`battery` (percentage of battery remaining) **FAIL**, too low e.g. 2700mV is more like 70+%
`battery_low` (bool) **FAIL**, not set to `true` when `batteryAlarmState > 0`

# Results raw data

Endpoint: 1, Cluster: PowerCfg

## Sensor1:
`Read result of 'genPowerCfg': {"batteryVoltage":27,"batteryAlarmState":0}`

State:
```
    "battery": 16,
    "battery_low": false,
    "voltage": 2700
```

## Sensor2:
`Read result of 'genPowerCfg': {"batteryVoltage":23,"batteryAlarmState":8}`

State:
```
    "battery": 4,
    "battery_low": false,
    "voltage": 2300
```

## Sensor3: 
`Read result of 'genPowerCfg': {"batteryVoltage":27,"batteryAlarmState":0}`

State:
```
    "battery": 16,
    "battery_low": false,
    "voltage": 2700
```

## Sensor4: 
`Read result of 'genPowerCfg': {"batteryVoltage":25,"batteryAlarmState":0}`

State:
```
    "battery": 8,
    "battery_low": false,
    "voltage": 2500
```

danieledwardgeorgehitchcock, 2022-10-30:
Thanks for the detailed feedback - I'll have a look tomorrow and see if I can tweak anything

danieledwardgeorgehitchcock, 2022-10-31:
@aioue - I have updated the Gist - could you try with the updated converter?

aioue, 2022-10-31:
# External Converter Test 2

1. Replaced osram .js file with updated version in z2m data directory
2. Restarted z2m
3. Power state did not appear to be updating as quickly as it did before, thinking there was an issue, I removed, then re-added the reference to the external converter in z2m settings. This prompted one sensor to update. When I manually fetched the values in the dev console the power states updated so potentially normal caching/slow update behaviour I am not aware of. Values now coming through successfully.

# Results

**100% SUCCESS**

**This issue resolved using latest converter.** Thank You @danieledwardgeorgehitchcock 

`voltage` (millivolts) **SUCCESS**, converted correctly to state
`battery` (percentage of battery remaining) **SUCCESS**, % values rendering as expected
`battery_low` (bool) **SUCCESS**, set to `true` when `batteryAlarmState > 0`

# Results raw data

Endpoint: 1, Cluster: PowerCfg

## Sensor1:
`Read result of 'genPowerCfg': {"batteryVoltage":27,"batteryAlarmState":0}`

State:
```
{
    "battery": 73,
    "battery_low": false,
    "linkquality": 220,
    "occupancy": false,
    "temperature": 21.07,
    "voltage": 2700
}
```

## Sensor2:
`Read result of 'genPowerCfg': {"batteryVoltage":23,"batteryAlarmState":8}`

State:
```
{
    "battery": 36,
    "battery_low": true,
    "linkquality": 176,
    "occupancy": false,
    "temperature": 20.28,
    "voltage": 2300
}
```

## Sensor3: 
`Read result of 'genPowerCfg': {"batteryVoltage":27,"batteryAlarmState":0}`

State:
```
{
    "battery": 73,
    "battery_low": false,
    "linkquality": 192,
    "occupancy": true,
    "temperature": 19.59,
    "voltage": 2700
}
```

## Sensor4: 
`Read result of 'genPowerCfg': {"batteryVoltage":25,"batteryAlarmState":0}`

State:
```
{
    "battery": 55,
    "battery_low": false,
    "linkquality": 204,
    "occupancy": false,
    "temperature": 20.6,
    "voltage": 2500
}
```

aioue, 2022-10-31:
Seeing the `fz` local override in the gist also helps me understand a bit more about how the code works. If possible, could you leave that online please? It's a useful reference.

danieledwardgeorgehitchcock, 2022-10-31:
Yeah no problem! I have now added the changes to my PR and marked it as ready for review.

It might require a bit of fettling to get the job done properly, but in the meantime, you have a working external converter :-)

closed 2022-11-03 by Koenkk