Koenkk/zigbee2mqtt #14684: IKEA bulbs (LED2003G10) can't switch scenes with IKEA Styrbar remote when binding

hnykda, 2022-10-28:
### What happened?

I followed the same steps as in this thread here: https://community.home-assistant.io/t/binding-ikea-tradfri-5-button-remote-with-ikea-floalt-via-zigbee2mqtt-ui/327806/14?u=kotrfa , basically binding remote with a bulb via group. Unfortunately, it seems that switching the color scenes do not work as expected. Interestingly, I have the exactly same controller+light (on the same firmwares, same network) elsewhere and it works just fine (using the approach with the "magic group"), I just bought them ~10 months earlier. 

Related issues:
* https://github.com/Koenkk/zigbee2mqtt/issues/12611
* https://github.com/Koenkk/zigbee2mqtt/issues/13151
* https://github.com/Koenkk/zigbee2mqtt/issues/14215
* https://github.com/Koenkk/zigbee2mqtt/issues/14392

### What did you expect to happen?

Scenes would switch

### How to reproduce it (minimal and precise)

Try to bind https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10 with https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002 and it might not work.

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20220219

### Adapter

Slaesh

### Debug log

Here is when I press different buttons and what it does, I don't see the changes I would expect
```
zigbee2mqtt/testolea {"brightness":1,"color_mode":"color_temp","color_temp":454,"state":"ON"}
zigbee2mqtt/0x003c84fffe69d3f7 {"brightness":1,"color_mode":"color_temp","color_temp":454,"last_seen":"2022-10-28T12:03:11+02:00","linkquality":116,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"on","battery":100,"last_seen":"2022-10-28T12:03:11+02:00","linkquality":123,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:11+02:00","linkquality":123,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:11+02:00","linkquality":123,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:11+02:00","linkquality":123,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:11+02:00","linkquality":0,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"arrow_left_click","battery":100,"last_seen":"2022-10-28T12:03:24+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:24+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:24+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:24+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:24+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"arrow_right_click","battery":100,"last_seen":"2022-10-28T12:03:29+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:29+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:29+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:29+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:29+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"on","battery":100,"last_seen":"2022-10-28T12:03:33+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:33+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:33+02:00","linkquality":61,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:33+02:00","linkquality":61,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:33+02:00","linkquality":61,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/testolea {"brightness":1,"color_mode":"color_temp","color_temp":454,"state":"OFF"}
zigbee2mqtt/0x003c84fffe69d3f7 {"brightness":1,"color_mode":"color_temp","color_temp":454,"last_seen":"2022-10-28T12:03:34+02:00","linkquality":116,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"off","battery":100,"last_seen":"2022-10-28T12:03:35+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:35+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:35+02:00","linkquality":0,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:35+02:00","linkquality":0,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:35+02:00","linkquality":0,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"on","battery":100,"last_seen":"2022-10-28T12:03:43+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:43+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:43+02:00","linkquality":120,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:43+02:00","linkquality":120,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:43+02:00","linkquality":120,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/testolea {"brightness":1,"color_mode":"color_temp","color_temp":454,"state":"ON"}
zigbee2mqtt/0x003c84fffe69d3f7 {"brightness":1,"color_mode":"color_temp","color_temp":454,"last_seen":"2022-10-28T12:03:43+02:00","linkquality":116,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}
[... 13 lines cut ...]
zigbee2mqtt/0x003c84fffe69d3f7 {"brightness":90,"color_mode":"color_temp","color_temp":454,"last_seen":"2022-10-28T12:03:51+02:00","linkquality":116,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"arrow_left_click","battery":100,"last_seen":"2022-10-28T12:03:55+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":"","battery":100,"last_seen":"2022-10-28T12:03:55+02:00","linkquality":65,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:55+02:00","linkquality":61,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:55+02:00","linkquality":61,"update":{"state":"idle"},"update_available":null}
zigbee2mqtt/0x9035eafffe1a5d56 {"action":null,"battery":100,"last_seen":"2022-10-28T12:03:55+02:00","linkquality":61,"update":{"state":"idle"},"update_available":null}
[... cut ...]
zigbee2mqtt/0x003c84fffe69d3f7 {"brightness":67,"color_mode":"color_temp","color_temp":454,"last_seen":"2022-10-28T12:04:06+02:00","linkquality":112,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}
zigbee2mqtt/testolea {"brightness":60,"color_mode":"color_temp","color_temp":454,"state":"ON"}
zigbee2mqtt/0x003c84fffe69d3f7 {"brightness":60,"color_mode":"color_temp","color_temp":454,"last_seen":"2022-10-28T12:04:11+02:00","linkquality":112,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}
```

hnykda, 2022-10-28:
OK, I am lost... 

I have tried to repeat the same process and now it's working again. I don't know why, I really really followed did the same thing as I had done twice before that... 

So the takeaway for others stumbling upon this: keep trying :shrug: ...

antonio1475, 2022-12-02:
I have the same problem. My office lights are two [LED1732G11](https://www.zigbee2mqtt.io/devices/LED1732G11.html) and one [LED2003G10](https://www.zigbee2mqtt.io/devices/LED2003G10.html).

I'm using an [E1810](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810) remote. 
Binding works (to all 3 devices, not group as per [E1810's binding limitations](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#binding) ), but the LED2003G10 won't change color temperature.

Same thing if I unbind all and just bind the remote to the LED2003G10 (incl. Scenes)...
It does work through the dashboard or HA, incl. changes to the light group.

hnykda, 2022-12-02:
OK, I am reopening, as it seems to work unreliably for quite a few of us for some reason. So something fishy is going on.

antonio1475, 2022-12-02:
Thanks. Workaround for now: remote binded to all 3 bulbs for OnOff & Brightness, but not for Scenes.
For color temp, I set an automation (using a blueprint, although it can be done without):

```
alias: '[Zigbee2MQTT] Tradfri E1810 Despacho'
description: ''
use_blueprint:
  path: EPMatt/ikea_e1524_e1810.yaml
  input:
    integration: Zigbee2MQTT
    controller_device: a1622dd1a187117869016a6803625b51
    controller_entity: sensor.e1810_despacho_action
    helper_last_controller_event: input_text.last_controller_event_e1810_despacho
    action_button_right_long:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Despacho/set
          payload: '{"color_temp_move": -100}'
    action_button_left_long:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Despacho/set
          payload: '{"color_temp_move": 100}'
    action_button_left_release:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Despacho/set
          payload: '{"color_temp_move": "stop"}'
    action_button_right_release:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Despacho/set
          payload: '{"color_temp_move": "stop"}'
    action_button_left_short:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Despacho/set
          payload: '{"color_temp_step": 30,"transition": 0.5}'
    action_button_right_short:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Despacho/set
          payload: '{"color_temp_step": -30,"transition": 0.5}'
```
Works fine... 😅

closed 2023-01-10 by github-actions[bot]

Dielee, 2023-02-13:
Any updates on this ? Is there anything we can do ?