Koenkk/zigbee2mqtt #14687: Unable to pair Friends of Hue Switches (Niko/Jung) with Sonoff Dongle Plus E

tompijls, 2022-10-28:
### What happened?

Clean Z2M install on new home assistant setup on odroid.

Pairing process as explained on https://www.zigbee2mqtt.io/devices/PTM_215ZE.html does not work.

Pairing in deconz and hue app works without problems

### How to reproduce it (minimal and precise)

click permit join
Press top right button on Niko switch for 10+ seconds
release
nothing happens

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle Plus E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9888457/log.txt)

danieledwardgeorgehitchcock, 2022-10-28:
I seem to recall that the Green Power (EnOcean) devices have different buttons mapped to different Zigbee Channels when pairing..

What channel is your Z2M network on?

tompijls, 2022-10-28:
It's on channel 11. 
I actually tried all 4 buttons.

tompijls, 2022-10-28:
Ah ok, that explains 

> Currently, the ezsp adapter does not support working with greenpower devices. I'm working on it. I'll post when you can try.

Ah Ok - that explains :-)

Looking forward to it

kirovilya, 2022-11-02:
@tompijls  try now on the latest version of z2m 1.28.1

tompijls, 2022-11-02:
will be able to do on friday - travelling this week

tompijls, 2022-11-05:
Hi Ilya,

pairing worked fine. 
I can also capture the button press events in node-red. 
Long presses on the other hand don't give a different event (they just show up os a short press)

Not sure if this is to be expected? (phoscon did register short vs long presses with same switch)

Cheers

Tom

tompijls, 2022-11-11:
hey @kirovilya , let me know if anything else is needed?

kirovilya, 2022-11-11:
@tompijls I don't know what else to do. are there any errors in the logs? if not, then everything is working.
no idea yet how to distinguish between long and short presses

danieledwardgeorgehitchcock, 2022-11-11:
As green power devices are only momentarily powered, a press event is a press event on the device (with the exception of the long press combos which put the device into a channel pairing mode). I'd say the Deconz behaviour is not coming from the device, but rather a timer or something similar.. there is some documentation floating about from the manufacturer that lists all of the possible button events.. I'll see if I can dig it out...

kirovilya, 2022-11-11:
@danieledwardgeorgehitchcock if it were possible to distinguish between pressing and releasing the button, then yes, it would be possible to consider the duration as a timer. but I understand that the event is one - only when the button is released. I didn't see the logs - I can't say

danieledwardgeorgehitchcock, 2022-11-11:
Looks like both the press and release events should be being reported by Z2M. See the device converter [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a41e7cf0d78bea89997103a23a613c6c79607810/devices/enocean.js#L31)

tompijls, 2022-11-11:
Correct, both events are captured (although the release event may be dependent on the link quality). With low quality i have seen a few occasions where the release event didnt make it

This is in the log tab of MQTT

` 2022-11-11 08:47:26MQTT publish: topic 'zigbee2mqtt/FOH Black', payload '{"action":"press_1","linkquality":223}'
2022-11-11 08:47:26MQTT publish: topic 'zigbee2mqtt/FOH Black', payload '{"action":"","linkquality":223}'
2022-11-11 08:47:26MQTT publish: topic 'zigbee2mqtt/FOH Black/action', payload 'press_1'
2022-11-11 08:47:26MQTT publish: topic 'zigbee2mqtt/FOH Black', payload '{"action":"release_1","linkquality":224}'
2022-11-11 08:47:26MQTT publish: topic 'zigbee2mqtt/FOH Black', payload '{"action":"","linkquality":224}'
2022-11-11 08:47:26MQTT publish: topic 'zigbee2mqtt/FOH Black/action', payload 'release_1'`

tompijls, 2022-11-11:
In general, it must be possible to distinguish. These are friends of hue devices, and when connecting to a hue hub, i can clearly chose between press or hold events.

@danieledwardgeorgehitchcock @kirovilya let me know if you need more logs

danieledwardgeorgehitchcock, 2022-11-11:
I think that we are getting out of scope now with the original issue / request.

I have many of these different Green Power devices dotted around my home (albeit, I am not using an EZSP device as my co-ordinator) and have never had issue with them (although I've read plenty of reports to the latter).

I'd go as fast as to say that this is expected behaviour for such a device in Z2M so would recommend closing this issue as it has been resolved and if you still feel that the device isn't functioning as expected, open up a new one

kirovilya, 2022-11-11:
now I see that there are hundreds of pressing and releasing. yes, programmatically through the timer, you can implement the definition of the duration of pressing. Unfortunately, I have no plans to improve the converter for this device. I provided support for greenpower devices for the ezsp adapter.

danieledwardgeorgehitchcock, 2022-11-11:
> now I see that there are pressing and releasing. yes, programmatically through the timer, you can implement the definition of the duration of pressing. Unfortunately, I have no plans to improve the converter for this device. I provided support for greenpower devices for the ezsp adapter.

I don't think Z2M supports timer related pseudo attributes anyway but it could be done quite easily in the home automation software itself such as Home Assistant or Node-Red

tompijls, 2022-11-11:
understood and agreed this is potentially a new request - thank you both - will close this issue and open a new one if/when needed.

Cheers

tom

closed 2022-11-11 by tompijls

Hedda, 2022-11-18:
> Currently, the ezsp adapter does not support working with greenpower devices. I'm working on it. I'll post when you can try.

FYI, you guys can follow kirovilya development and help out as a tester in the dedicated zigbee-herdsman issue about EZSP here:

https://github.com/Koenkk/zigbee-herdsman/issues/319

https://github.com/Koenkk/zigbee-herdsman/pulls?q=is%3Apr+ezsp