Koenkk/zigbee2mqtt #14692: IKEA Starkvind error when setting mode & no converter available

Xeevis, 2022-10-28:
### What happened?

Sometimes (not always) when changing `mode` I get this error, but otherwise it appears to work correctly.

```log
error 2022-10-28 18:53:44: Publish 'set' 'fan_mode' to 'Patro Obyvák Čistička vzduchu' failed: 'Error: Write 0xb4e3f9fffe2b154b/1 manuSpecificIkeaAirPurifier({"fanMode":20}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
debug 2022-10-28 18:53:44: Error: Write 0xb4e3f9fffe2b154b/1 manuSpecificIkeaAirPurifier({"fanMode":20}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4476,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:389:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/devices/ikea.js:257:17)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
```

I also get `no converter available` every time I change mode.
> debug 2022-10-28 18:53:48: No converter available for 'E2007' with cluster 'manuSpecificIkeaAirPurifier' and type 'raw' and data '{"data":[24,124,18,1],"type":"Buffer"}'

### What did you expect to happen?

No error and no missing converter.

### How to reproduce it (minimal and precise)

Change fan mode on [IKEA Starkvind E2007](https://www.zigbee2mqtt.io/devices/E2007.html) with firmware 1.1.001.

### Zigbee2MQTT version

1.28.0-dev commit: c943992

### Adapter firmware version

0x26720700

### Adapter

 ConBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9890529/log.txt)

sjorge, 2022-10-29:
I think the no converter might be related to being unable to set fan_mode.

There should be now raw buffer data in manuSpecificIkeaAirPurifier. And for the finegrained fan_mode selection a attribute on the manuSpecificIkeaAirPurifier cluster is used.

I wonder if the deconz adaptor code somehow has issues properly parsing the payload. I've never seen raw buffer messages when using a zstack based coordinator. @Koenkk is that possible? Or is the cluster attribute mapping share between all adaptors?

Koenkk, 2022-10-29:
@sjorge attribute mapping is shared but it could be that the Conbee II does some strange stuff with the payload.

@Xeevis first make sure you are using the latest conbee II firmware, latest firmware is `0x26780700`

Xeevis, 2022-10-29:
> @Xeevis first make sure you are using the latest conbee II firmware, latest firmware is `0x26780700`

Oh boy that seems to be it and I was checking firmware few weeks ago, single digit difference in the middle make for terrible versioning scheme. 😅🤦‍♂️ 

Closing & sending ☕ as apology for wasting your time.

closed 2022-10-29 by Xeevis