Koenkk/zigbee2mqtt #14694: LELLKI WK34 (EU) not recognize as this model.

Minims, 2022-10-28:
### What happened?

Hello,

I have ordered in December 2021 a LELLKI WK34 (EU)

This socket is recognized as 
```
Zigbee Manufacturer: _TZ3000_ps3dmato
Model: [TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3)
```

But the model I have is : https://www.zigbee2mqtt.io/devices/WK34-EU.html
My FW is 1.0.5 on this Socket.

PS: Z2M show this socket in OTA tab, When OTA is checked I have this error : 
```
Failed to check if update available for 'Prise Murale Zigbee - 1' (No image available for imageType '5634')
```

The ON/OFF socket is working but things like LED INDICATOR does not work.
Is it possible to add _TZ3000_ps3dmato as WK34-EU in the code to test ? 

Thanks.

### What did you expect to happen?

Socket recognized as a WK34-EU

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

sonoff zigbee 3.0 usb dongle plus

Minims, 2022-10-31:
I think I am in this case : 

https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#broken-attribute-reporting-functionality

My plug is reported as `[TS011F_plug_3](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3)`

[image]

I have the FW 1.0.5

Z2M says there is an OTA update, but when I try to update I have this error : 

```
Info 2022-10-31 12:19:13MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x540f57fffe74d2e8"},"error":"Update of 'Prise Murale Zigbee - 1' failed (No image available for imageType '5634')","status":"error"}'
```

[image]

closed 2022-12-08 by github-actions[bot]

davidcampos, 2024-09-22:
Same issue 👎