Koenkk/zigbee2mqtt #14698: zigbee2mqtt could not restart with Conbee2 stick

vturekhanov, 2022-10-29:
### What happened?

I use Conbee2 USB stick in my Home Assistant test instance:
- HA OS 9.3 (the same with 9.2)
- HA Core 2022.10.5 (the same with 2022.9.7)
- zigbee2mqtt addon 1.28.0-1

A couple of weeks ago I found that zigbee2mqtt could not restart (log attached).

There are no paired devices at the moment, only the coordinator is in the network.
Tried to use /dev/serial/by-id/... instead of /dev/ttyACM0 with the same result.
There is no problems with restart when I just change Conbee2 to cc2531.

The workaround (always works): unplug and plug again the stick.

Here is my z2m addon config:
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto
  user: user
  password: password
serial:
  port: /dev/ttyACM0
  adapter: deconz
zigbee_herdsman_debug: true
```

### What did you expect to happen?

zigbee2mqtt restarts successfully.

### How to reproduce it (minimal and precise)

1. Plug Conbee2 stick to USB port.
2. Start zigbee2mqtt addon.
3. Be sure addon started successfully.
4. Restart zigbee2mqtt addon.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

0x26780700

### Adapter

Conbee2

### Debug log

[06:45:43] INFO: Preparing to start...
[06:45:43] INFO: Socat not enabled
[06:45:43] INFO: Zigbee Herdsman debug logging enabled
[06:45:44] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2022-10-29 06:45:47: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-10-29 06:45:47: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-29.06-45-47' filename: log.txt
Zigbee2MQTT:debug 2022-10-29 06:45:47: Removing old log directory '/config/zigbee2mqtt/log/2022-10-16.18-25-05'
Zigbee2MQTT:info  2022-10-29 06:45:47: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-10-29 06:45:47: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2022-10-29 06:45:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":0,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
2022-10-29T00:45:47.560Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-10-29T00:45:47.562Z zigbee-herdsman:deconz:driver Set delay to 0
2022-10-29T00:45:47.564Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":0}}'
2022-10-29T00:45:47.565Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2022-10-29T00:45:47.612Z zigbee-herdsman:deconz:driver Serialport opened
2022-10-29T00:45:47.622Z zigbee-herdsman:controller:log Started with result 'resumed'
2022-10-29T00:45:47.624Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2022-10-29T00:45:47.626Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2022-10-29T00:45:50.565Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
2022-10-29T00:45:50.571Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 request id: 1
2022-10-29T00:45:50.595Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 1 status: 0
2022-10-29T00:45:50.596Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-10-29T00:45:50.597Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
2022-10-29T00:45:50.598Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-10-29T00:45:50.599Z zigbee-herdsman:deconz:driver query aps data confirm
2022-10-29T00:45:50.605Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 3
2022-10-29T00:45:50.606Z zigbee-herdsman:deconz:driver query aps data indication
2022-10-29T00:45:50.609Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x0 request id: 1 confirm status: 0
2022-10-29T00:45:50.610Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-10-29T00:45:50.610Z zigbee-herdsman:deconz:driver query aps data indication
2022-10-29T00:45:50.611Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 4
2022-10-29T00:45:50.616Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 4 srcAddr: 0x0 destAddr: 0x0 profile id: 0x0 cluster id: 0x8004 lqi: 69
2022-10-29T00:45:50.617Z zigbee-herdsman:deconz:frameParser response payload: 1,0,0,0,26,1,4,1,0,4,0,5,0,0,6,0,10,0,25,0,1,5,4,1,0,32,0,0,5,2,5
2022-10-29T00:45:50.617Z zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr.
2022-10-29T00:45:50.617Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-10-29T00:45:50.618Z zigbee-herdsman:deconz:adapter RECEIVING SIMPLE_DESCRIPTOR - addr: 0x0 EP:1 inClusters: 0,6,10,25,1281 outClusters: 1,32,1280,1282
2022-10-29T00:45:50.618Z zigbee-herdsman:deconz:adapter EP: 1
2022-10-29T00:45:50.619Z zigbee-herdsman:deconz:adapter profile ID: 260
2022-10-29T00:45:50.619Z zigbee-herdsman:deconz:adapter device ID: 1024
2022-10-29T00:45:50.619Z zigbee-herdsman:deconz:adapter input cluster: 0x0
2022-10-29T00:45:50.620Z zigbee-herdsman:deconz:adapter input cluster: 0x6
2022-10-29T00:45:50.620Z zigbee-herdsman:deconz:adapter input cluster: 0xa
2022-10-29T00:45:50.620Z zigbee-herdsman:deconz:adapter input cluster: 0x19
2022-10-29T00:45:50.621Z zigbee-herdsman:deconz:adapter input cluster: 0x501
2022-10-29T00:45:50.621Z zigbee-herdsman:deconz:adapter output cluster: 0x1
2022-10-29T00:45:50.621Z zigbee-herdsman:deconz:adapter output cluster: 0x20
2022-10-29T00:45:50.622Z zigbee-herdsman:deconz:adapter output cluster: 0x500
2022-10-29T00:45:50.622Z zigbee-herdsman:deconz:adapter output cluster: 0x502
2022-10-29T00:45:50.622Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 5
2022-10-29T00:45:57.631Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
Zigbee2MQTT:error 2022-10-29 06:45:57: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-29 06:45:57: Failed to start zigbee
Zigbee2MQTT:error 2022-10-29 06:45:57: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-29 06:45:57: Exiting...
Zigbee2MQTT:error 2022-10-29 06:45:57: undefined

vturekhanov, 2022-11-29:
The problem still exists.

platini76, 2022-11-30:
same problem here.

closed 2023-02-07 by github-actions[bot]

vturekhanov, 2023-02-19:
The problem still exists.
HA OS 9.5 (ova)
HA Core 2023.2.5
zigbee2mqtt addon 1.30.1-1

The workaround mentioned above also still works: physically unplug and plug again the stick.