Koenkk/zigbee2mqtt #14702: [New device support]: LEDEPLY Zigbee MR16 smart bulbs

radaiming, 2022-10-29:
### Link

https://www.amazon.cn/gp/product/B09SHM3CP8

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0x540f57fffe796777","nwkAddr":61865,"manufId":4098,"manufName":"_TZ3210_vaiyrvd1","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"61440":0,"onOff":1}},"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_vaiyrvd1","powerSource":1,"zclVersion":3,"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"57344":1000,"61447":[],"61448":[],"61449":[],"61453":0,"tuyaRgbMode":0,"currentHue":null}},"genLevelCtrl":{"attributes":{"61440":1000}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1667037611664,"defaultSendRequestWhen":"immediate"}

### Comments

I copy [devices/tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a197538c53680aaa79b0b976213b67a620f56d1c/devices/tuya.js#L882) into the following converter and it works, but for color temperature, seems it's reporting `Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x540f57fffe796777!`

### External converter

```shell
const extend = require('zigbee-herdsman-converters/lib/extend');

module.exports = [
    {
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_vaiyrvd1'}],
        model: 'TS0505B',
        vendor: 'TuYa',
        description: 'Zigbee RGB+CCT light',
        whiteLabel: [{vendor: 'Mercator Ikuü', model: 'SMD4106W-RGB-ZB'},
            {vendor: 'TuYa', model: 'A5C-21F7-01'}, {vendor: 'Mercator Ikuü', model: 'S9E27LED9W-RGB-Z'},
            {vendor: 'Aldi', model: 'L122CB63H11A9.0W', description: 'LIGHTWAY smart home LED-lamp - bulb'},
            {vendor: 'Lidl', model: '14153706L', description: 'Livarno smart LED ceiling light'},
            {vendor: 'Zemismart', model: 'LXZB-ZB-09A', description: 'Zemismart LED Surface Mounted Downlight 9W RGBW'},
        ],
        extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500], disableColorTempStartup: true}),
        meta: {applyRedFix: true, enhancedHue: false},
    },
];
```

### Supported color modes

color temperature; color rgp

Koenkk, 2022-10-29:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-29 by Koenkk

radaiming, 2022-10-30:
It works, thanks!