Koenkk/zigbee2mqtt #14704: TuYa PJ-ZGD01 send message every 30sec.

ALaDoffe, 2022-10-29:
### What happened?

Hello

My garage door opener TuYa PJ-ZGD01 is very verbose.
It send the message below every 30 sec

info 2022-10-29 15:50:19MQTT publish: topic 'zigbee2mqtt/0xa4c138200172b9b6', payload '{"garage_door_contact":true,"linkquality":123,"trigger":true}'
debug 2022-10-29 15:50:35Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":37120}' from endpoint 1 with groupID 0
debug 2022-10-29 15:50:35Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":37120}' from endpoint 1 with groupID 0
info 2022-10-29 15:50:36MQTT publish: topic 'zigbee2mqtt/0xa4c138200172b9b6', payload '{"garage_door_contact":true,"linkquality":127,"trigger":true}'
debug 2022-10-29 15:51:05Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":37632}' from endpoint 1 with groupID 0
debug 2022-10-29 15:51:05Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":37632}' from endpoint 1 with groupID 0
info 2022-10-29 15:51:06MQTT publish: topic 'zigbee2mqtt/0xa4c138200172b9b6', payload '{"garage_door_contact":true,"linkquality":120,"trigger":true}'
debug 2022-10-29 15:51:35Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":38144}' from endpoint 1 with groupID 0
debug 2022-10-29 15:51:35Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":38144}' from endpoint 1 with groupID 0
debug 2022-10-29 15:51:36Received Zigbee message from '0xa4c138200172b9b6', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":

I tried to play with reporting but no way.
Don't know if it is normal and if there is something to do. 

Here below 

Device.yaml : 

 "0xa4c138200172b9b6": {
                "debounce": 0.5,
                "friendly_name": "0xa4c138200172b9b6",
                "homeassistant": {},
                "legacy": false,
                "optimistic": true,
                "retain": true
}

database.db

{"id":15,"type":"Router","ieeeAddr":"0xa4c138200172b9b6","nwkAddr":34211,"manufId":4417,"manufName":"_TZE200_nklqjk62","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":
1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���*i���*i���*i\u0001��*i\u001f��*i=��*i[��*iy��*i","65506":54,"65508":0,"appVersion":70,"modelId":"TS0601","manufa
cturerName":"_TZE200_nklqjk62","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002291d5a3","endpointID":1}],"configuredReportings"
:[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVe
rsion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1667052465729,"defaultSendRequestWhen":"immediate"}

### Zigbee2MQTT version

1.28.0 commit: 03ba647

### Adapter firmware version

20220928

### Adapter

CC2652P

Koenkk, 2022-10-29:
This is normal for TuYa devices and there is no known way to limit this.

ALaDoffe, 2022-10-30:
Ok, thank's for the information.

closed 2022-10-30 by ALaDoffe