Koenkk/zigbee2mqtt #14709: Continuous left of the network and new device not join

maframan, 2022-10-29:
### What happened?

Continuous left of the network and/or new device not join.
Random wrong behavior

### What did you expect to happen?

Devices join on the network without problem

### How to reproduce it (minimal and precise)

Add new device

### Zigbee2MQTT version

version 1.28.0-dev (commit #c943992)

### Adapter firmware version

Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'

### Adapter

Sonoff ZBDongle-E

### Debug log

[zigbee2mqtt_edge.log](https://github.com/Koenkk/zigbee2mqtt/files/9894064/zigbee2mqtt_edge.log)

maframan, 2022-10-29:
I just tried that if I let the same device in through the router rather than the coordinator, I have no problem .
When he enters with the coordinator he always fails the interview and does not display the battery status.

maframan, 2022-10-30:
Nothing, the problem persist.
The device ZBMini left and join of the network contiuosly.

maframan, 2022-10-31:
someone kindly help me?

maframan, 2022-10-31:
Please follow the "Giardino anteriore - Lato strada" device:
[debug_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9900870/debug_log.txt)

maframan, 2022-10-31:
Left of the network occur after 30 seconds about. Continuously enters and exits from the network

maframan, 2022-10-31:
I tried with zbdongle-p on another Raspberry with zigbee2mqtt configured on Channel 25 and the problem not occur with the same device. Zbdongle-e is the problem?

maframan, 2022-11-01:
I've upgraded to 1.28.1 but the problem persist.
Find in attach log file where same behavior, left and join continuously, for device 0x00124b0024c0792c.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9914263/log.txt)

Please help me for resolve my tremendous issue.

Koenkk, 2022-11-02:
Could you make a sniff of this? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

maframan, 2022-11-03:
Please, follow 0x00124b0024c1eb2b device that It left and join from the network two time

[0x00124b0024c1eb2b_device.zip](https://github.com/Koenkk/zigbee2mqtt/files/9926596/0x00124b0024c1eb2b_device.zip)

maframan, 2022-11-03:
Please, consider this snoop, device 0x00124b0024c1eb2b
[snoop_zigbee_0x00124b0024c1eb2b_device.zip](https://github.com/Koenkk/zigbee2mqtt/files/9927299/snoop_zigbee_0x00124b0024c1eb2b_device.zip)

maframan, 2022-11-03:
After your analysis, please explain me how to analyze the snoop with device address for example, thanks a lot

maframan, 2022-11-03:
Who is d68d? A ezsp (zbdongle-e) router or coodinator?
Why if pair this same device near to coordinator the problem not occur?
Very strange behavior, maybe it is caused by interference?

Koenkk, 2022-11-03:
0x0000 = always the coordinator. 

> Why if pair this same device near to coordinator the problem not occur?

Did you already try repairing the router you are trying to pair it via?

maframan, 2022-11-04:
> 0x0000 = always the coordinator. 
> 
> 
> 
> > Why if pair this same device near to coordinator the problem not occur?
> 
> 
> 
> Did you already try repairing the router you are trying to pair it via?

Problem persist unfortunately
I found other issue that i think to be correlated at main problem in particular new device join with tradri router only and not join with zbdongle-e router.

maframan, 2022-11-05:
I did below actions:
- plugged an usb 2.0 hub to Intel NUC usb 3.0 port
- replaced extension usb 2.0 cable connected to the usb 2.0 hub (before the cable was 3mt now is 1,5 mt) 
and I don't have the problem, stranger behavior, is it possible?

kirovilya, 2022-11-05:
@maframan yes, it is not recommended to use usb3 and recomended use an extension cable for sticks.

Do you still have problems connecting devices to the Sonoff-E stick?

maframan, 2022-11-05:
Now I've problem to connect new device to zbdongle-e with router firmware while the new device join normally with other router like tradri or zbmini devices. I think that this is another problem now, maybe with router firmware on zbdonle-e flashed with this firmware https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/blob/master/Dongle-E/Router/Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl, do you suggest other firmware?

kirovilya, 2022-11-06:
@maframan Sorry, I don't know much about router firmware.

gomble, 2022-11-08:
I have the same problem:

```
	Line  1452: warn  06-11-2022 15:14:41: Device 'Treppe-Bewegungsmelder' left the network
	Line  1455: warn  06-11-2022 15:14:41: Device 'Treppe-Bewegungsmelder' left the network
	Line  2144: warn  06-11-2022 17:00:21: Device 'Dachgeschoss-6Tasten' left the network
	Line  2147: warn  06-11-2022 17:00:21: Device 'Dachgeschoss-6Tasten' left the network
	Line  9636: warn  07-11-2022 06:16:55: Device 'Bewegungsmelder Dachgeschoss' left the network
	Line 13254: warn  07-11-2022 12:03:55: Device 'Bewegungsmelder Dachgeschoss' left the network
	Line 22247: warn  08-11-2022 08:38:50: Device 'Thermostat-Esszimmer' left the network
```

```
	Line    11: info  06-11-2022 13:39:31: Currently 27 devices are joined:
	Line  2150: info  06-11-2022 17:00:23: Device 'Dachgeschoss-6Tasten' joined
	Line  2151: info  06-11-2022 17:00:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Dachgeschoss-6Tasten","ieee_address":"0x04cf8cdf3c7dce26"},"type":"device_joined"}'
	Line 10490: info  07-11-2022 07:47:19: Device 'Bewegungsmelder Dachgeschoss' joined
	Line 10491: info  07-11-2022 07:47:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bewegungsmelder Dachgeschoss","ieee_address":"0x00124b002500279f"},"type":"device_joined"}'
	Line 22250: info  08-11-2022 08:38:50: Device 'Thermostat-Esszimmer' joined
	Line 22251: info  08-11-2022 08:38:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Thermostat-Esszimmer","ieee_address":"0x2c1165fffecdecb7"},"type":"device_joined"}'
```

maframan, 2022-11-08:
@gomble what is this tool?
[image]

gomble, 2022-11-08:
> @gomble what is this tool?
> 
> [image]
> 
> 

Network map on zigbee2mqtt

maframan, 2022-11-08:
> > @gomble what is this tool?
> > 
> > [image]
> > 
> > 
> 
> Network map on zigbee2mqtt

This is network map 
[image]

I am referring to other screenshot with info about usb device

MattWestb, 2022-11-08:
@maframan Your mesh network is not working OK and is boicked most of the time.
Sequence 1749 to 1772 is not getting any 15.4 ack = no frames is going thru.
Also source routing is failing completely.  
Do you have some very bad routing devices or have radio interference from WiFi or other devices ?

maframan, 2022-11-09:
> @maframan Your mesh network is not working OK and is boicked most of the time. Sequence 1749 to 1772 is not getting any 15.4 ack = no frames is going thru. Also source routing is failing completely. Do you have some very bad routing devices or have radio interference from WiFi or other devices ?

Thanks for your info, now i've resolved with extension cable usb 2.0 and usb 2.0 hub connected to NUC usb 3.0 port. The problem that now i've is the battery drain on IKEA Tradri shortcut button

MattWestb, 2022-11-09:
Great @maframan !!
The IKEA and some more brands controlling devices is draining the battery in 48 hours is know.
Its one or more bugs in the device firmware that is making the device not going in sleep and always have the radio on. Its being triggered then the sleeping end device is jumping to one new parent and oft is it done then its pared to the coordinator and its down then updating the system or restart of the system.
Best is paring the device to one good router near it end position so its not jumping around and triggering the bug in the firmware.

maframan, 2022-11-21:
the problem came back, how do i check the interference? The Wi-Fi network in the area are only 1 and 6 channel and zigbee Channel is 20.

maframan, 2022-11-23:
For test, I replaced two zbdongle-e router with two tradfri repeater but the problem persist. Please suggest me ama action plan For resolve or find the root cause, thanks a lot!

maframan, 2022-11-23:
Other test, i've installed other instance of zigbee2mqtt with cc2531 (last firmware zstack 3.0) and devices works fine. The New zigbee2mqtt instance have channel 25 while the first instance  that has problems with channel 20. My point of view is that the channel 20 has interference, right?

closed 2022-12-31 by github-actions[bot]

maframan, 2022-12-31:
Any news about this problem?