Koenkk/zigbee2mqtt #14714: [Livarno Home LED ceiling light HG08008] flickers when turning on, after it being off for "some time"

GsakuL, 2022-10-29:
### What happened?

I bought the light in store and mounted it. Interviewing was not problematic, and worked the first time.
I'm using HomeAssistant and mostly Amazon Alexa, for control.
Come late evening and i want to use the light, so tell Alexa to do it. It turns on, but a few Milliseconds later turns off. Tell Alexa again to turn it on, it stays on.
I've tested this manually within the Z2M Webinterface. It is not a HA/Alexa Problem.
If I've used the lamp "recently enough", it will react "normally".

### What did you expect to happen?

Stays on the "first" time. So no "flicker". As with any other light/device

### How to reproduce it (minimal and precise)

"Enough time" seems to be something like 5-20minutes. I have not timed it yet. But cutting mains is faster and works every time - I just tried it for this very Issue.

- cut off mains power to the Lamp
- wait 20s (or maybe less)
- apply mains
- (Lamp turns on because mains)
- (needs 30s-1min to re-connect and listen for commands - this is kinda slow compared to other devices, but not a major issue in this case)
- turn it off (it should really turn off, maybe "on" then "off" command) 
- turn it on
- it "flickers": stays on for a few ms and turns off
- turn on again (without hitting "off" in between) , and it stays on

---
Additional info:
- on the Packaging/Manual:
  - IAN: 388125_2104
  - Model Version: 10/2021
- on the About Tab:
  - Zigbee Model: TS0505B
  - Zigbee Manufacturer: _TZ3210_p9ao60da
- Settings:
  - Transition: 0
  - "brightness": 41
  - "color_mode": "color_temp",
  - "color_temp": 370 
---

I still have my "old" CC2531 lying around. But I don't own any other Hub/Gateway, nor can I borrow one from/visit someone, to use for Sniffing their traffic.

### Zigbee2MQTT version

1.28.0 commit: 03ba647

### Adapter firmware version

20211217

### Adapter

Slaesh's CC2652RB stick

### Debug log

This is the following:
- lamp was off "long enough" (as i was writing all of this) to not need "mains reset" described above
- i click "on"
- lamp flickers
- i click "on"
- lamp stays on
- i click "off"
- lamp turns off
```
info 2022-10-29 19:55:21Succesfully changed options
info 2022-10-29 19:55:21MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"fadma-1"}'
debug 2022-10-29 19:55:40Received MQTT message on 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0/set' with data '{"state":"ON"}'
debug 2022-10-29 19:55:40Publishing 'set' 'state' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0'
debug 2022-10-29 19:55:40Received Zigbee message from 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
info 2022-10-29 19:55:40MQTT publish: topic 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0', payload '{"brightness":41,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":54,"state":"ON"}'
debug 2022-10-29 19:55:50Received MQTT message on 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0/set' with data '{"state":"ON"}'
debug 2022-10-29 19:55:50Publishing 'set' 'state' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0'
debug 2022-10-29 19:55:50Received Zigbee message from 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
info 2022-10-29 19:55:50MQTT publish: topic 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0', payload '{"brightness":41,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":25,"state":"ON"}'
debug 2022-10-29 19:55:53Received MQTT message on 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0/set' with data '{"state":"OFF"}'
debug 2022-10-29 19:55:53Publishing 'set' 'state' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0'
info 2022-10-29 19:55:53MQTT publish: topic 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0', payload '{"brightness":41,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":51,"state":"OFF"}'
```

tryptophan0, 2022-11-12:
I have the same problem with this lamp. I am sending messages to mqtt via a script. The light would make the transition from off to full brightness, which takes about 1 second and then turn off, instead of staying lit. One thing that popped up in the zigbee2mqtt log is an error about color_mode:
`info  2022-11-11 20:13:32: MQTT publish: topic 'zigbee2mqtt/livarno1', payload '{"brightness":99,"color":{"x":0.448,"y":0.4076},"color_mode":"color_temp","color_temp":350,"last_seen":"2022-11-11T19:13:32.486Z","linkquality":51,"state":"ON"}'`
`error 2022-11-11 20:13:32: No converter available for 'set' 'color_mode' (color_temp)`

After tinkering with different attributes in the mqtt message, I finally found a case, that would keep the light on after the transition. The message content is:
`{'brightness_move': 100, 'state': 'ON'}`

Version: Zigbee2MQTT version 1.28.2 (commit #360a777) and zigbee-herdsman (0.14.68)

GsakuL, 2022-12-13:
using `1.28.4 commit: 52e545f` the issue still exists

flosch-dev, 2022-12-18:
Hello,
I'm having the same issue but with different Livarno Ceiling Lamp: https://www.zigbee2mqtt.io/devices/14153806L.html
In my case, I have 4 of those ceiling lamps some controlled via motion sensor (via  HomeAssistant automation) and some other with Philips wall switch module. 
The strange thing is, that I only hit the problem when triggered by motion sensor, but not when triggered by hue wall switch. In both cases, HomeAssistant will control.
I have also different motion sensors, all with the same problem. (IKEA, Aqara, Ali Express Tuya).

Is there anything I can bring in to investigate into the problem?

GsakuL, 2022-12-18:
@flosch-dev 
I'm out of town atm, so I'm not completely sure.
But I think the Ikea Sensor at least has a setting for brightness (button under the cover) that it will set the light to. 
Which in this case seems to be the cause/solution why they work together.

Sending brightness and "on" together (eg via mqtt) works. So I assume it's doing that.

I think what you can do to "investigate" is to sniff traffic, if you have a USB router left over: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

I have fooled around least week a bit with the dev console and some device regulations/documentations and pretty found out sending any command that alters brightness, work's (eg "move to _ and on". I also noticed "toggle" command will turn on the lamp, but it's not "very predictable" since you would have to know the current state

flosch-dev, 2022-12-18:
just need to correct my statement, it also occours when trigger of automation is philips wall switch. However I have one lamp I'm only switching via wall-switch and this one seems to work fine.

flosch-dev, 2022-12-18:
unfortunately I don't have such sniffer device, but maybe I need to buy one in future.
Currently I'm using aqara motion sensor and I just run into the problem again. 

I'm fearing the only option to nail down what's going on is to sniff the traffic?

flosch-dev, 2022-12-18:
one more observation I want to share...
I can trigger as well as recover the problem by frequently switching on and off (either via HA gui or via automation with ext. trigger). In that case I switch on/off faster than the transition itself.
I also set transition time to 5 sek (just for testing) and noticed that transition usually works. When the problem exists, the transition seems to be less than 1 sek and behaves like in the initial descriptions. It turns on fast and switches off immediately (like less 1 sec).

tomiisp, 2023-01-07:
Any news about this problem ? I have the same problem with this light and also there is a timeout on "power_on_behavior" option.
(Btw, I have also original Lidl Gateway with linked TuYa dev account, another Lidl Gateway flashed with "hacked" FW and also some free USB CC2531 sticks... if someone point me to how to sniff traffic with original GW or via TuYa account I can do it, I only use CC2531 as sniffer for GW with CC2652P which I normally use with Z2M)

GsakuL, 2023-01-07:
@tomiisp 
sadly no news

I've been trying to "at least" make it "somethat work" (="maybe flicker, but only one voice/button command") in HomeAssistant via an "Template: Light" and a number Helper ("last known 'on'-brightness") Entities with a script, but i can't seem get it to "turn on at x%". I've tried to hack it via the script and send multiple custom MQTT Commands in order, but it doesn't work as expected. Either it flickers and stays off (as it would currently) or it would step/move up to 100%, regardless of canceling (with MQTT `"*move/step": "off"` as listed on the device page). I've also tried different QoS Level, but with no luck. Either my script is "too fast" (which could very well be the case) or it does something i don't know.

I've never sniffed traffic, so all I can do is link you to the Z2M page again: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html 🙁

I've "poked around a bit", and found the defintion for the converter: https://github.com/Koenkk/zigbee-herdsman-converters/blob/64e6e5e6be8e35d47291ea3b4f1bf082347c8efd/devices/lidl.js#L585
Which has been initially added via this PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4128
by @pcedik

The only other thing I could think of is asking @Koenkk (I'm sorry to ping you) with regards to the device/Issue itself and the sniffing

tomiisp, 2023-01-07:
@GsakuL have you tried to update to latest version 1.29.1 ? Today I mess around that light and also no luck but afternoon I spot new release, update it and problem is gone but I tested this only few hours but it never happen again after update, maybe it's just a coincidence and that light have only "good day" today but it is worth to try, of course it still need more time to test.

GsakuL, 2023-01-07:
@tomiisp 
i was running 1.29.0, but just updated to 1.29.1 to report. Updated, "About" is showing the new version, yes. But the issue persists. :(

As said above, you can also "force" this "wrong/flashy behaviour", if you set/send the "off" command twice or more in a row. Or you could "hard reset" (like I wrote in the opening message), or "just wait some time".
Remember: if you turn the light on once, and get past "the flickering", it works as any light/lamp should for ""some time"", until it falls back into its "wacky state".

tomiisp, 2023-01-08:
Ok, I setup original TuYa gateway + SmartLife app, I set sniffer and grab network key and pair that light. What exactly need to be sniffed ? 
For example I just made this steps and all traffic is in attached file.
- turn light on
- change color temperature
- change brightness 
- turn light off
- turn light on
- change to RGB Red
- change brightness
- change contrast
- turn light off
- turn light on
(hope I remember these steps correctly)

[livarno-388125_2104_1_08012023.zip](https://github.com/Koenkk/zigbee2mqtt/files/10368194/livarno-388125_2104_1_08012023.zip)

Koenkk, 2023-01-09:
@tomiisp  Can you also provide a sniff when pairing the device? I need the network key of your tuya hub.

Koenkk, 2023-01-10:
@tomiisp thanks, it seems the normal ON command is used. Can you provide the herdsman debug logging when turning the light ON from zigbee2mqtt (at this point it should immediately turn OFF).

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

tomiisp, 2023-01-12:
@Koenkk sorry had not time for this kind of logging in past days (maybe tomorrow I try log it) but maybe I found a source of this problem, for me that light work fine for last few days, no flash at all and now I check what I change and seems it is related to transition value, I have set 1 second at brightness 20 = works without problem, yesterday I changed this value to 3 second and light just blink, everytime, I changed value (in HA) back to 1sec and it start working properly again. I will try to do more test and that log tomorrow.

GsakuL, 2023-01-19:
@tomiisp I had some time, and tried setting transition=1, but this does not solve the flickering with my lamp. If its in "deep sleep"(?) aka the "long off"/"borked" state, it will just flicker, and i still have to send a second 'on' command :(

tomiisp, 2023-01-20:
@GsakuL I start have similar problem with another Livarno light  ( _TZ3210_c0s1xloa, TS0505B, big one 34W - 399629_2110), no motion sensor just scene switch. Light turn on then after 1sec turn off, that On/Off I see also on Z2M web frontend, same if I turn on that light via frontend switch BUT in this case it can be "fixed" by hitting blue refresh button on State, after this light start working again normal, odd. @Koenkk this "refresh state fix" remember me sniffed communication with original TuYa in log (must look again) where request for light state was sent after every action and also it repeats in log every ~3 minutes not matter if that light was on or off.

tomiisp, 2023-01-20:
@Koenkk just my little investigation, two same lights, one connected to Z2M, second to TuYa, I see from log some difference in communication, maybe it is not important but maybe evil is hiding in details.
Z2M:
- light is turned on/off with command "Move to Level with OnOff (0x04)"
- brightness same as on/off - "Move to Level with OnOff (0x04)"
- temperature - Move to Color Temperature (0x0a)
- color - Move to Color (0x07)
command "Read Attributes (0x00)" seems is send only after hitting blue refresh button in frontend

Tuya:
- light is turned on by "On (0x01)" followed by "Read Attributes (0x00)"
- light is turned off by "Off (0x00)" followed by "Read attributes (0x00)"
- brightness  "Move to level" (without "with OnOff") followed by "Read Attributes"
- temperature/color same as Z2M but again followed by "Read Attributes"
every ~3minutes light send to coordinator "Report Attributes", coordinator respond and then coordinator send to light "Read Attributes" and this repeat every 3minutes not matter if light is on or off (I attached TuYa log)

Just thinking, maybe this kind of lights doesn't like some type of commands or sequences ? Or maybe that missing periodical Read/Report attributes commands send that light to unknown state or some "sleep" mode ?
Maybe I am absolute out of topic but I see this difference between Z2M and TuYa communication and same problem appears also on another Livarno light (as I posted above).

[livarno-388125_2104_2_20012023.zip](https://github.com/Koenkk/zigbee2mqtt/files/10470243/livarno-388125_2104_2_20012023.zip)

Koenkk, 2023-01-21:
@tomiisp can you try the following and see if this issue does not appear when:
- turn light off
- z2m frontend -> device -> dev console
- in the bottom box: fill in endpoint `1`, cluster `6`, command: `1`, payload `{}` -> execute: light should turn ON now
- in the bottom box: fill in endpoint `1`, cluster `8`, command: `0`, payload `{"level": 20, "transtime": 1}` -> execute: light changes to brightness 20 and does not turn off (issue fixed 🎉 )

tomiisp, 2023-01-22:
@Koenkk I tested that and it works fine BUT I was able confuse that light in odd way, this steps I repeat few times and its maybe 60/40 ratio between fine/odd response from light :

- light off
- dev 1/6/1/{}
- dev 1/8/0/{"level": 200, "transtime": 50}
until here it was ok, in sniffer I see standard "ZCL On/Off" and "ZCL Level control: Move to Level" cmds and light was ON
- now I tap scene switch, HA sent cmd to /set {"state":"OFF","transition":"5.0}, sniffer show "ZCL Level Control: Move to Level with OnOff" with payload "Level: 0, Transition: 5.0s"
- and here is that odd thing (60/40), sometimes light goes down and turn OFF and sometimes light goes down BUT it not turn OFF and stay ON at level 1, Z2M frontend reporting state OFF until I hit blue refresh button > it change to ON (set reporting for OnOff help a little but it takes a time)
- if light stay ON at level 1 and I send via Dev 1/8/0/{"level":200, "transtime": 50} light goes up normally BUT after ~2sec it turn OFF (very similar effect to that "blink" problem)
- and what is important also, Z2M reporting in frontend/MQTT different status about ON/OFF which can confuse HA well (for example service Light:Toggle do odd things if light is ON but Z2M thinks it is OFF)

From my point of view I still think it have something to do with cmds, just "Move to level with OnOff" (Dev 1/8/4) vs. "Move to level" (Dev 1/8/0) & "OnOff" (Dev 1/6/0-1).

Koenkk, 2023-01-23:
The transition is probably the problem, what if you just send `{"state":"OFF","transition":"5.0}` ?

Barsonax, 2023-01-27:
> The transition is probably the problem, what if you just send `{"state":"OFF"}` ?

For me this does not fix the problem. Only sending a On twice seems to atleast end up with a lamp thats actually on (with some flicker :().

tomiisp, 2023-01-27:
@Barsonax just a question, have you also that problem after some time of light inactivity and after "wakeup" it works for some time normally ?

Barsonax, 2023-01-27:
> @Barsonax just a question, have you also that problem after some time of light inactivity and after "wakeup" it works for some time normally ?

I suspect so but it's hard to really verify.

One way to force this behaviour is sending 2 off's. Then at the next on it will transition to the on state and then turn off by itself. 

Sending another on (without a off in between) will fix this buggy state. That's actually my workaround for now, my code uses 2 on's to turn the light on. Hope that will at least make it reliable.

tomiisp, 2023-01-27:
@Barsonax just for test, can you try to setup reporting for that light (also @GsakuL ) like on this screen ? original TuYa do this every 3 minutes and maybe it help prevent odd state of light... its worth to try this too, at least we eliminate one thing what original TuYa do

EDIT: I am not able upload any image, always get a "something went really wrong"..hmm

here is the list:
Endpoint - Cluster - Attribute - Min rep interval - Max rep interval
- 1 - OnOff - OnOff - 60 - 900
- 1 - LevelCtrl - currentLevel - 60 - 900
- 1 - LColorCtrl - currentHue - 60 - 900
- 1 - LColorCtrl - currentSaturation - 60 - 900
- 1 - LColorCtrl - colorTemperature - 60 - 900

Barsonax, 2023-02-18:
In the meantime the light broke as in the warm leds stopped working so only cold light.

I have had nothing but problems with the lidl lights. I bought several of them and they all have issues. This is not a zigbee2mqtt issue I believe but a issue with the lights themselves.

Iam going to replace the lights with philips as I never have issues with those.

GsakuL, 2023-03-03:
@tomiisp I'm sorry. I completely forgot about doing this. Done it now, I'll keep an eye on it.
Though it still does blink/flicker, if I "force double turn off and once turn on" the lamp. I'll wait and try, when "sleep mode kicks in" (or would).

I undersand @Barsonax 's reasoning. I'd consider returning myself, while I still can. Z2M is really great, but "bad hardware" can make so much work/problems. Since I still have warranty, and I know LIDL is (at least here) pretty chill with returns in that period, I'll keep it for a bit longer.

I don't want to "put pressure" on anyone. But at the end of the day it's "just a generic ceiling light", so "nothing special". So maybe, if we couldn't solve this at all, perhaps consider putting a warning/note onto the Website/Index referencing this issue.

Barsonax, 2023-03-03:
I agree. If it's not trivial to fix then at least a warning these lights have some issues in de docs would help ppl avoid this

Yakie996, 2023-03-16:
I also have this issue with my lidl lamps HG08008 and 14153806L, turning on and it turns off within a couple of milliseconds this behaviour keeps happening until I change the brightness or color of the lamp, sadly this is a temporary fix as the issue returns a couple of minutes later.

petrows, 2023-04-06:
Have the same issue, and i did not found workaround for it yet :-(

In my setup i have 10 commands in loop with 100ms delay, lamps have annoying flickering, but at least they are return to ON back. 

Having same feeling as @GsakuL : i also thinking about to return devices to LIDL, but i like lamps itself, they are really bright and nice looking in my setup.

petrows, 2023-04-07:
Found workaround working for me - i have added special proxy item in OpenHAB, which connects to groups instead of original item, and updates original item only if state is changed. Without repeating OFF events (via groups) my devices now works much better. Now i have only problem with quick ON/OFF and hardcoded transition in lamps, but normal usage now does not flicker or turning off.

Commit: [ef59891](https://github.com/petrows/smarthome-openhab/commit/ef59891bc9e8f09d712a67675a0163a1fb50fa2a)

tomiisp, 2023-04-07:
Just a basic question to all : why this lights works fine with original TuYa gateway and app but doing strange things with Z2M ?... it is FW problem or implementation ? I wrote above (Jan20) some differences in commands (on zigbee level) how that lights is turned On/Off via TuYa and via Z2M.

GsakuL, 2023-04-07:
@tomiisp 
Like I've mentioned earlier, I set up your reportings. I used the light now from time to time, but using this without any "proxy device" still has the issues of turning itself off. Using it with my non-optimized proxy it wont turn off (maybe because the delay is now longer, until the second command from the proxy arrives). But this is still bad, since my proxy is buggy an sometimes slowly ramps up brightness (due to my setup).
Also i have one of those Multi-Button-Switches, which can send events like "brightness start up" and "brightness stop" (which i like, since it's just two events for a continuous change), but for that, i use MQTT directly anyways (for the required payload to the lamp)

To your question:
Could it be, that the TuYa gateway uses some unknown/undocumented(?) setup-commands? Or is that ruled out by the traffic sniff?.
Like the [TuYa TS004F](https://www.zigbee2mqtt.io/devices/TS004F.html) 4-Button-Switch was once changed in Z2M, to put it in "switch mode", so it reports all 12 events (single, double, long click for each button), instead of it's "dimmer mode". I have this switch, and have noticed the change. (but now the documentation says, this can be done in hardware, which is even never)

Regarding different commands:
I've found this site: https://csa-iot.org/csa_product/smart-led-ceiling-light/ there you can "Download Compliance Document", which is a ZIP file with several long docx files. In there are function descriptions and Test Logs. I don't understand all the terminology. But if I take a look at `15-0309-006-0x0006-OnOff-Cluster-Test-Specification-Editors.docx` at section `5.3.2	OO-TC-02S: Primary functionality with server as DUT	26` (just search `OO-TC-02S`) it shows a test sequence (table) where I've seen the following:
[image]
Take a look at step 2 and 3. It seems to be that a on/off command is immediately followed by a read command - as listed in your differences. Also the fact, that these two steps are `*a` and `*b` somewhat makes me think, that this is necessary/intended/required, even though the wording in the last column does not explicitly mention this.

tomiisp, 2023-04-07:
@GsakuL thanks for that link, I have a look deeper on that but I quickly look at that section 5.3.2 and I not see there command what is used by Z2M, I reporting above main differences how is On/Off sent to device (except Read attributes):
Tuya: sent single ON or OFF command and then followed by "Move to level" command
Z2M: sent "combined" single command "Move to level with OnOff" (0x04) for both ON and OFF

petrows, 2023-04-08:
Hey @tomiisp ,

> I start have similar problem with another Livarno light ( _TZ3210_c0s1xloa, TS0505B, big one 34W - 399629_2110), no motion sensor just scene switch. Light turn on then after 1sec turn off, that [...]

I just observed exactly the same behavior for another [Livarno light: E14 RGB bulb (HG07834B, _TZ3000_th6zqqy6, TS0505B)](https://www.zigbee2mqtt.io/devices/HG07834B.html#lidl-hg07834b). It seems to be this problem is at least common for this Zigbee device implementation.

I can also add, that problem is also "fixed" by requesting `state` property via `<device>/get` endpoint.

tomiisp, 2023-04-08:
@petrows 
hm, seems this is common problem for most TS0505B based devices
and about that requesting state, that is what Z2M not do after sending commands to device but TuYa issue "request attributes" command after each command sent to device and also it is mentioned in that docs what was posted by GsakuL, maybe this is real root of this problem.

Yakie996, 2023-05-01:
this happens to me with [this](https://www.zigbee2mqtt.io/devices/HG08008.html#lidl-hg08008)  and [this](https://www.zigbee2mqtt.io/devices/14153806L.html#lidl-14153806l)

tomiisp, 2023-05-01:
@Yakie996 tag this response to @Koenkk too, this problem with "Lidl lights based on TS0505B" must be solved somehow, this Lidl lights have CSA certificate and works fine with Tuya/Lidl gw/apps but Z2M have problem with it and it is clearly problem of Z2M implementation and not Lights itself which follow Tuya protocols and pass CSA certification...

Yakie996, 2023-05-01:
> this happens to me with [this](https://www.zigbee2mqtt.io/devices/HG08008.html#lidl-hg08008) and [this](https://www.zigbee2mqtt.io/devices/14153806L.html#lidl-14153806l) light. turn it on and turn itself off, this is happening for me for more than 4 months. it makes the lights not usable, is a fix coming soon?

> @Yakie996 tag this response to @Koenkk too, this problem with "Lidl lights based on TS0505B" must be solved somehow, this Lidl lights have CSA certificate and works fine with Tuya/Lidl gw/apps but [...]

@Koenkk

Koenkk, 2023-05-04:
@tomiisp from your findings in https://github.com/Koenkk/zigbee2mqtt/issues/14714#issuecomment-1399446243 I think this has something to do with the transition. Did you already try without a transition?

petrows, 2023-06-04:
Dear @Koenkk , i have tested a lot, with/without transition, seems to be it does not affect this issue. Transition has another issue there, for example, lamp may have delayed "off" command on transition, for example, it may switch off after transition delay second time, but this looks like another issue and disabling it does not helps with original problem.

GsakuL, 2023-07-10:
I'm sorry for responding so late, but I had critical personal matters to attend to.

Currently running Z2M v `1.31.2`, just wanted to test this real quick, so didn't want to "risk" an update and letting this go stale.

@Koenkk I've tried your converter. On first impression, turning the lamp on works without flashing. However other functions fail, e.g. set/get brightness:
```txt
debug 2023-07-10 20:48:46Publishing 'set' 'brightness' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0'
error 2023-07-10 20:48:46Publish 'set' 'brightness' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0' failed: 'ReferenceError: colorModeLookup is not defined'
debug 2023-07-10 20:48:46ReferenceError: colorModeLookup is not defined at Object.convertSet (/app/data/extension/externally-loaded.js:25:47) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:52)
info 2023-07-10 20:48:46MQTT publish: topic 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0', payload '{"brightness":1,"color":{"h":227,"hue":227,"s":53,"saturation":53,"x":0.2607,"y":0.2633},"color_mode":"color_temp","color_temp":60,"level_config":{"execute_if_off":true},"linkquality":142,"state":"ON"}'
```
```txt
debug 2023-07-10 20:51:00Received MQTT message on 'zigbee2mqtt/home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0/get' with data '{"brightness":""}'
debug 2023-07-10 20:51:00Publishing get 'get' 'brightness' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0'
error 2023-07-10 20:51:00Publish 'get' 'brightness' to 'home/device/ground_floor/office/ceiling/light/Lidl_Fixture_0' failed: 'Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel''
debug 2023-07-10 20:51:00Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel' at Object.getAttribute (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:93:19) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:546:87) at Object.convertGet (/app/data/extension/externally-loaded.js:110:26) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:274:37) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:126:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1184:20)
```
@tomiisp i had to "reconfigure" (yellow double arrow button on device "About" page) the lamp, because otherwise the external converter seemed not to be used.

closed 2023-08-18 by github-actions[bot]

Yakie996, 2023-12-09:
The problem is still present with my lidl ceiling lights, when i turn the light on it briefly turns on for 0.5 sec turns off again (still displays as on in ha) when I change the brightness the light turns on again and stays on.
 (https://www.zigbee2mqtt.io/devices/HG08008.html#lidl-hg08008 and https://www.zigbee2mqtt.io/devices/14153806L.html#lidl-14153806l)

How can I help solving this problem? 

I have home assistant on my synology nas using a vm and the skyconnect stick.

bruvv, 2024-01-25:
@Koenkk any update ? This is still very annoying https://github.com/Koenkk/zigbee2mqtt/issues/17193

Yakie996, 2024-01-25:
> @Koenkk any update ? This is still very annoying #17193

I've switched tot zha because z2m was to buggy the last two months and there is no problem with lidl lights there

bruvv, 2024-01-25:
Good to know although i am not really feeling it to switch totally over since I am also running custom zigbee devices. and I do not feel like running 2 networks.

sstreng, 2024-02-04:
I'm experiencing the exact same issue with multiple Lidl lamps and it is pretty annoying. However, if I pair the Lamps with a Sonoff ZBbridge running Tasmota they work perfectly fine. I digged into Tasmotas source and found that it has a dedicated "LidlPower" definition, but I was unable to figure out what exactly the difference is. But maybe @s-hadinger would be so nice to shed some light into this issue?

pomeloy, 2024-08-22:
Still an issue sadly. TS0505B-based [Lidl HG07834C](https://www.zigbee2mqtt.io/devices/HG07834C.html) light bulbs are also affected.

RubenKelevra, 2024-10-19:
Just for everyone subscribed here, we figured out what is causing this issue in https://github.com/Koenkk/zigbee2mqtt/issues/21286. Follow along as we try to fix the firmware bug with the Lidl support (or implement a workaround in z2m).