Koenkk/zigbee2mqtt #14724: `No converter available` error regularly reported for several properties

BlackLight, 2022-10-30:
### What happened?

Re-report of https://github.com/Koenkk/zigbee2mqtt/issues/11449.

This issue has been occurring regularly on my network for months.

It regularly applies to properties such as `battery`, `linkquality` and `voltage`, regardless of the device, and it happens whenever I try to retrieve these properties on a device through the `get` API.

[The API documentation](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-friendly-name-get) provides an example on how to retrieve the `state` of a property (push to `zigbee2mqtt/FRIENDLY_NAME/get` with `{"state": ""}` as a payload). But it doesn't tell much about how to retrieve properties that are _not_ strings.

I have tried to pass both `{"battery": ""}` and `{"battery": 0}` to the payload (and, in general, pass a zero if the property is numeric), as well as `{"battery": null}`, but none of them works. The `No converter available` error is always there.

### What did you expect to happen?

`battery`, `linkquality`, `voltage` etc. are correctly retrieved on `zigbee2mqtt/<FRIENDLY_NAME>/get` when passing a payload that expresses them with values consistent with their type (e.g. `""` for strings and `0` for `type=numeric`).

The values on that payload should actually be completely ignored, for they are redundant (it's already known both to the client and the server what's the type of the property they are retrieving/returning), so I don't expect converter errors there. And, if we want to stick to strict type consistency instead, at least we should be consistent with it - if I pass `{"battery": 0}` on the payload of `zigbee2mqtt/MyDevice/get`, and `battery` is a valid numeric property, then no errors should be raised.

### How to reproduce it (minimal and precise)

`zigbee2mqtt/<FRIENDLY_NAME>/get` with e.g. `{"battery": 0}` or `{"battery": ""}` as a payload.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20211115

### Adapter

CC2531

sjorge, 2022-10-30:
Not every attribute has a convertGet function in there toZigbee converter.

Especially for battery powered devices, you're seeing this error because the attribute is missing such a converter. You can easily check if one is support when using the frontend:
[image]

Only the 2  with the blue refresh button support get requests for this random device I used as an example. In all likely hoed if you check the device you are using they will lack get support.

sjorge, 2022-10-30:
@Koenkk I suppose since we now have better support for battery devices with genPollCtl... we could add a convertGet for batteryVoltage and batterPercentRemaining converters ?

Koenkk, 2022-10-30:
@sjorge we can add a converter but due to the `genPollCtl` the response can take a few minutes so not really user friendly. 

@BlackLight `{"battery": 0}` or `{"battery": ""}` makes no difference (the value is ignored).

sjorge, 2022-10-30:
> @sjorge we can add a converter but due to the `genPollCtl` the response can take a few minutes so not really user friendly.

Yes, the response is not going to be instant. With the illuminance one i added for a different issue it indeed sometimes takes multiple minutes to respond. Definitely not something a normal user might expect.

BlackLight, 2022-10-30:
Thanks @sjorge for the hint, it put me on the right track.

I've taken a look at how the frontend renders the "sync" button, and I've realized that I've been wrong about values access this whole time.

The button is rendered only [if the value has `ACCESS_READ`](https://github.com/nurikk/zigbee2mqtt-frontend/blob/d9e6020f5c3bcb80d02a387430c40de00da0383a/src/components/features/composite/FeatureWrapper.tsx#L23), and that is true only if [the 3rd bit, not the 1st](https://github.com/nurikk/zigbee2mqtt-frontend/blob/c82df115bd8e49120eb823d33e54e6c46030fe51/src/types.ts#L108), of `access` is set (that's probably a bit counter-intuitive for somebody used to the UNIX access convention).

So in my case I can fix it by simply doing a `get` if the right access bit is set I guess, and if it's a state-only property it'll be updated at some point later.

closed 2022-10-30 by BlackLight