Koenkk/zigbee2mqtt #14725: Zigbee2mqtt stops working

chersoph, 2022-10-30:
### What happened?

Zigbee2mqtt in Home assistanat stopps working

### What did you expect to happen?

I expect that z2m will send device data to HA but it stops sending data.

### How to reproduce it (minimal and precise)

I do not know how to reproduce it. I tjust happens.
In the log file the problem starts after 00:35:39.

### Zigbee2MQTT version

 1.28.0

### Adapter firmware version

0x26690700

### Adapter

Raspbee ii

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9896068/log.txt)

jimmi1982, 2022-11-02:
Hi,
I appear to be having similar issues lately. Zigbee devices not responding however when I restart zigbee2mqtt all works correctly again. 

If you need me to add any logs etc then please let me know.

wifiuk, 2022-11-07:
i got similar issues with a conbee II, having to keep restarting HA to get Z2M to work again and even then I'm getting lots of timeouts, wonder if its the stick?

closed 2022-11-07 by chersoph

wifiuk, 2022-11-08:
yeah im not using a rpi, have a nuc sort of thing, power isn't the issue for me