Koenkk/zigbee2mqtt #14726: [New device support]:

TheModin, 2022-10-30:
### Link

https://nimly.io/sv/accessoarer/nimly-module

### Database entry

{"id":60,"type":"EndDevice","ieeeAddr":"0xf4ce36410e0b7724","nwkAddr":43458,"manufId":4660,"manufName":"Onesti Products AS","powerSource":"DC Source","modelId":"NimlyPRO","epList":[11],"endpoints":{"11":{"profId":260,"epId":11,"devId":10,"inClusterList":[0,1,3,5,4,257,65186],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"NimlyPRO","manufacturerName":"Onesti Products AS","powerSource":4,"zclVersion":2,"appVersion":7,"stackVersion":10,"hwVersion":11,"dateCode":"20220614","swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":7,"stackVersion":10,"hwVersion":11,"dateCode":"20220614","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1667146587897,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new device page but got stuck at restarting the zigbee2mqtt edge addon in homeassistant.
This is the lock: [Nimly Touch Pro](https://nimly.io/sv/produkter/nimly-touch-pro-ultimate-black)
This is the zigbee module inside: [Connect Module](https://nimly.io/sv/accessoarer/nimly-module)
After adding the file NimlyPRO.js next to configuration.yaml i get the following error when restarting zigbee2mqtt.

`[17:43:47] INFO: Preparing to start...
[17:43:48] INFO: Socat not enabled
[17:43:48] INFO: Starting Zigbee2MQTT...
node:internal/modules/cjs/loader:959
  throw err;
  ^
Error: Cannot find module '../lib/exposes'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:956:15)
    at Function.Module._load (node:internal/modules/cjs/loader:804:27)
    at Module.require (node:internal/modules/cjs/loader:1028:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /app/dist/util/externally-loaded.js:1:17
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)`

I tried modifying the following file, since it looks to be the same kind of lock: [onesti.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/onesti.js)
So the file looks like this: 
`const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const reporting = require('../lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const constants = require('../lib/constants');

module.exports = [
    {
        zigbeeModel: ['NimlyPRO', 'easyCodeTouch_v1', 'EasyCodeTouch', 'EasyFingerTouch'],
        model: 'NimlyPRO',
        vendor: 'Onesti Products AS',
        description: 'Zigbee module for EasyAccess code touch series',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery, fz.lock_programming_event, fz.easycodetouch_action],
        toZigbee: [tz.lock, tz.easycode_auto_relock, tz.lock_sound_volume, tz.pincode_lock],
        meta: {pinCodeCount: 50},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(11);
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
            await endpoint.read('closuresDoorLock', ['lockState', 'soundVolume']);
            device.powerSource = 'Battery';
            device.save();
        },
        exposes: [e.lock(), e.battery(), e.sound_volume(),
            e.lock_action_source_name(), e.lock_action_user(),
            e.action(Array.from(Object.values(constants.easyCodeTouchActions))),
            exposes.binary('auto_relock', ea.STATE_SET, true, false).withDescription('Auto relock after 7 seconds.'),
            e.pincode(),
        ],
    },
];`

Koenkk, 2022-10-30:
The imports should be like `require('zigbee-herdsman-converters/lib/exposes');` not `require('../lib/exposes');`

TheModin, 2022-10-30:
Thanks @Koenkk that solved it.
Now i'm not sure how to do the next step.
I get:
```
No converter available for 'NimlyPRO' with cluster 'closuresDoorLock' and type 'commandUnlockDoorRsp' and data '{"status":0}'
```
I can not find an existing converter with commandUnlockDoorRsp at https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js

Koenkk, 2022-10-31:
@TheModin I think this one can be ignored, the lock status updates correctly when changing it right?

TheModin, 2022-10-31:
@Koenkk ok cool. Yes everything works fine.
To get this as a supported device would I do a pull request with NimlyPRO in zigbee model in the onesti.js file?

Koenkk, 2022-11-01:
Yes!

closed 2022-11-04 by TheModin

TheQue42, 2023-03-19:
@TheModin : Hi Martin,  

Could you please explain if the current support of the nimly locks, is supposed to be able to report which user opened the lock, and which source (fingerprint, tag, keycode, etc), opened the lock? 
I'm trying to follow the various tickets regarding this lock, but I cant figure out if everyone else got this working, since I havent :-( 
I can just get the lock/unlock events, volume, and configure auto-lock.