Koenkk/zigbee2mqtt #14731: Water consumption working bad on RTX ZVG1

yah996, 2022-10-31:
### What happened?

Water consumption is shown incorrectly. 

In the nativ Tuya App it shows correctly.

On the rtx.js on the DP5 is a strange converter entry, current values divided by 33.88...
I assume this is not correct and will "blow up" this value to an undefined stage and show often as 0.

Please check this, why there is an fl. oz converter. 

// Assume value is reported in fl. oz., converter to litres
return {water_consumed: (value / 33.8140226).toFixed(2)};
}

### Zigbee2MQTT version

 1.28.0

### Adapter firmware version

20210708

### Adapter

Sonoff CC2652P zStack3x0

### Debug log

These values are correct... (Taken from the Tuya Gateway).

{"result":{"devId":"xx","datas":[{"devId":"xx","eventType":"Report","sourceDetail":"","eventTime":1667115020633,"eventName":"water once","eventTimeStr":"2022-10-30 07:30:20:633","eventDetail":"4.30L","requestFrom":"device itself"},{"devId":"xx","eventType":"Report","sourceDetail":"","eventTime":1667115020497,"eventName":"water once","eventTimeStr":"2022-10-30 07:30:20:497","eventDetail":"100.00L","requestFrom":"device itself"},{"devId":"xx","eventType":"Report","sourceDetail":"","eventTime":1667113891661,"eventName":"water once","eventTimeStr":"2022-10-30 07:11:31:661","eventDetail":"15.70L","requestFrom":"device itself"},{"devId":"xx","eventType":"Report","sourceDetail":"","eventTime":1667113785285,"eventName":"water once","eventTimeStr":"2022-10-30 07:09:45:285","eventDetail":"8.20L","requestFrom":"device itself"},{"devId":"xx","eventType":"Report","sourceDetail":"","eventTime":1667113683052,"eventName":"water once","eventTimeStr":"2022-10-30 07:08:03:052","eventDetail":"0.00L","requestFrom":"device itself"}],"docType":1,"hasNext":false,"totalCount":0,"currentPageStartRowKey":"xxx"},"t":1667119219472,"success":true,"status":"ok"}

ZVG1: {
cluster: 'manuSpecificTuya',
type: 'commandDataResponse',
....
case 5: {
// Assume value is reported in fl. oz., converter to litres
return {water_consumed: (value / 33.8140226).toFixed(2)};
}

Koenkk, 2022-10-31:
What is published when changing that line to `return {water_consumed: value};`

yah996, 2022-10-31:
I am using HA, running a ssh session in the 45df7312_zigbee2mqtt  add-on.

That's what I like to test, but I had no success, maybe this is behind my Z2M knowledge. 

I am able to modify the  /app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js (vi). But strange enough to me, after restart the Z2M the modification was lost. It looks to me, restart will overwrite my file modification.

I tried to use an external converter rtx.js, but I had no success, not sure if the naming convention was correct. 

Maybe you can give me a hint, how to change the file correctly. 
Thanks.

Koenkk, 2022-10-31:
You should restart the container using `docker restart CONTAINER_ID`, then changes will be kept (so not via HA ui). 

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#testing-changes-locally

yah996, 2022-10-31:
thanks, i will try

yah996, 2022-11-01:
I changed it to return {water_consumed: (value / 10)};
To me it looks ok, the water values are now in Liter. 

Still problems, the return of a DP6 is not recognised.
Maybe this will be used an "overrun flag" for the 100 Liter marker.

I would like to run it as external converter, but  I had no idea how the filename ???.js should look like. I assume if have to use the part of the fromZigbee file and also I hope the external file will be used first.

Maybe a small hint would be great, thanks.

Koenkk, 2022-11-01:
Since you have the TuYa gateway, can you figure out all the datapoints? https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

closed 2022-12-09 by github-actions[bot]