Koenkk/zigbee2mqtt #14733: [New device support]: HZC Electronics motion sensor

robinpersson, 2022-10-31:
### Link

https://en.m.nu/sensor-zigbee/rorelsevakt-med-zigbee

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0x84fd27fffea5461c","nwkAddr":56887,"manufId":4714,"manufName":"Shyugj","powerSource":"Battery","modelId":"MotionSensor-ZB3.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1024,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"MotionSensor-ZB3.0","manufacturerName":"Shyugj","powerSource":3,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20210715","swBuildId":"4.0"}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":10791}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff06a41b","zoneState":1,"zoneStatus":52}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a41b","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20210715","swBuildId":"4.0","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1667229739387,"defaultSendRequestWhen":"active"}

### Comments

Got stuck hard. Please help

robinpersson, 2022-11-18:
No help?

johthe, 2022-12-26:
Hi
This is a me too...  help, TeLLuS#9412 on zigbee2mqtt Discord, if someone have the time.
Probably same as this https://www.alibaba.com/product-detail/Wireless-Zigbee-Intelligent-Position-Detector-Alarm_1600553824261.html

I use it with zigbeeModel: ['MotionSensor-ZB3.0'], model: 'S902M-ZG', vendor: 'Shyugj'

Also got it in, but not decoding correctly.  Think bit 3 for tamper.. last two bits for detect.

debug 2022-12-26 23:00:11Received Zigbee message from '...', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":55}' from endpoint 1 with groupID null
debug 2022-12-26 23:02:01Received Zigbee message from '...', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":51}' from endpoint 1 with groupID null
debug 2022-12-26 23:02:31Received Zigbee message from '...', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":48}' from endpoint 1 with groupID null
debug 2022-12-26 23:02:31Received Zigbee message from '...', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":52}' from endpoint 1 with groupID null
debug 2022-12-26 23:02:35Received Zigbee message from '...', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":55}' from endpoint 1 with groupID null

Adding comment to not close this.
/Johan

johthe, 2022-12-27:
Trying now with this setting, appear to work so far.
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.illuminance],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.battery(), e.illuminance(), e.tamper()],

debug 2022-12-27 23:34:46: Received Zigbee message from '...', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":55}' from endpoint 1 with groupID null
info  2022-12-27 23:34:46: MQTT publish: topic '...', payload {"battery":null,"battery_low":false,"illuminance":31571,"illuminance_lux":1435,"last_seen":"2022-12-27T23:34:46+01:00","linkquality":255,"occupancy":true,"tamper":true}'

Now, waiting for battery update, 3V two AAA, and may also try to include OTA values later.
May need hardware modification on the back for it to reach the tamper-switch to make it work as intended.

I hope it helps you also Robin.
/Johan

johthe, 2023-01-02:
[image]

150x150 pixel HZC-Shyugj-S902M-ZG.jpg image as requested

Battery show up as full 100% and green under Power.
But it is the only device that show up as green and I have some IKEA and Xiaomi that show full 100% battery but they are not in green color.

johthe, 2023-02-03:
We still waiting for this to be added to git and included in release, at-least it is working with this manual external script.

closed 2023-03-14 by github-actions[bot]

johthe, 2023-05-23:
Open again.. bad bot close.

Koenkk, 2023-05-24:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

theepicsaxguy, 2023-08-29:
I still have this issue. With both Edge version and the standard one. Using Homeassistant OS. 

```
Device '0xa4c138a1359d39ba' with Zigbee model 'MotionSensor-ZB3.0' and manufacturer name 'Shyugj' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

johthe, 2023-09-01:
I suspect we have been confused by robinpersson and now theepicsaxguy that use zigbeeModel: ['HZC Electric motion sensor']  Instead of zigbeeModel: ['MotionSensor-ZB3.0'],   That it identifies as in the log so that it still show up unsuported for us.   Unless I use my own definition.  Maybe should use fingerprint instead and add Shyugj?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['MotionSensor-ZB3.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'S902M-ZG', // Vendor model number, look on the device for a model number
    vendor: 'Shyugj', // Vendor of the device (only used for documentation and startup logging)
    description: 'Motion sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.illuminance],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.occupancy(), e.battery_low(), e.battery(), e.illuminance(), e.tamper()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};
module.exports = definition;

```