Koenkk/zigbee2mqtt #14735: Paulman 371000001 light flashing

cantfindanam, 2022-10-31:
### What happened?

Hi,
I noticed some strange behaviour of the Paulmann 371000001.
A few days ago it started to randomly turn on or off for just a second and it starts to happen more and more often.

The debug logs show that the device is requesting an OTA update every minute but otherwise nothing suspicious (it is definitely not flashing every time the request is send).
I attached a snippet of the request that was send around the time the light was flashing. However the message alway the same.
Any idea how to fix that?

### Zigbee2MQTT version

1.28.0 commit: 03ba647

### Adapter firmware version

20220219

### Adapter

ZigStar stick V4

### Debug log

 2022-10-31 20:09:07 Received Zigbee message from 'Kitchen_main_light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType": 4364," manufacturerCode":4624}' from endpoint 1 with groupID 0

Debug 2022-10-31 20:09:07Device 'Kitchen_main_light' requested OTA

Koenkk, 2022-11-09:
@mh166 can you try with z2m 1.28.2?

cantfindanam, 2022-11-09:
@Koenkk I already updated to "1.28.2 commit: 360a777". Same behaviour.

```
Debug 2022-11-09 18:25:52Received Zigbee message from 'Kitchen_main_light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":4364,"manufacturerCode":4624}' from endpoint 1 with groupID 0
Debug 2022-11-09 18:25:52Device 'Kitchen_main_light' requested OTA
Debug 2022-11-09 18:26:53Received Zigbee message from 'Kitchen_main_light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":4364,"manufacturerCode":4624}' from endpoint 1 with groupID 0
Debug 2022-11-09 18:26:53Device 'Kitchen_main_light' requested OTA
```

cantfindanam, 2022-11-09:
@mh166 Hmm that's weird. Why should only Paulmann devices show this quirk?
Thanks for the input! I just deactivated permit join and observe what happens.

timdj, 2022-11-17:
I suddenly also have this issue with my Paulmann lights since recently updating my home assistant and zigbee2mqtt. Joining for new devices is disabled.

I see no messages in my logs except the requested OTA

timdj, 2022-11-18:
> @mh166 can you try with z2m 1.28.2? (something has been changed in the OTA handling)

@Koenkk 
I think I started to see the problem after updating to 1.28.2 (not sure what I was running before). What changed?  I'll try downgrading soon and see if that solves the problem. Any suggestion what version I should go back to?

TMV-NL, 2022-11-18:
I have issues with the [Paulmann 929.66](https://www.zigbee2mqtt.io/devices/929.66.html) flashing at will or switching on at will. However the [Paulmann 500.44](https://www.zigbee2mqtt.io/devices/500.45.html) has no issues. As all my other lights from other vendors.
I'd like to verify that the issue is linked to Z2MQTT by connecting these lights via ZHA, but unfortunately ZHA does not recognize my Conbee II stick.

Koenkk, 2022-11-20:
@timdj 1.28.0

DenisLug, 2022-11-20:
I'm experiencing the same issue with two Paulmann 371000002 with many OTA update requests at least one in every minute. It happened after an update from 1.18.3 to 1.25.0.

mh166, 2022-11-27:
> @mh166 can you try with z2m 1.28.2? (something has been changed in the OTA handling)

@Koenkk Sorry for the late reply. Took me while to get to it. Here are the logs with v1.28.2:

<details>
<summary>Logs, grouped by request</summary>

```
2022-11-27T20:56:19.858Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,239,224,1,13,0,193,0,18,158,225,0,0,12,1,28,1,0,16,18,12,17,1,0,0,0,239,224,29,79]
2022-11-27T20:56:19.859Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,239,224,1,13,0,193,0,18,158,225,0,0,12,1,28,1,0,16,18,12,17,1,0,0,0,239,224,29,79]
2022-11-27T20:56:19.859Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,239,224,1,13,0,193,0,18,158,225,0,0,12,1,28,1,0,16,18,12,17,1,0,0,0,239,224,29] - 79
2022-11-27T20:56:19.860Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":57583,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":193,"securityuse":0,"timestamp":14786066,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,28,1,0,16,18,12,17,1,0,0,0]}}
2022-11-27T20:56:19.865Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":28,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4624,"imageType":4364,"fileVersion":1},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":57583,"endpoint":1,"linkquality":193,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
Zigbee2MQTT:debug 2022-11-27 21:56:19: Received Zigbee message from 'LED-Strahler 3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":4364,"manufacturerCode":4624}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-11-27 21:56:19: MQTT publish: topic 'zigbee2mqtt/LED-Strahler 3', payload '{"brightness":28,"color_mode":"color_temp","color_temp":370,"last_seen":"2022-11-27T21:56:19+01:00","linkquality":193,"state":"OFF"}'
Zigbee2MQTT:debug 2022-11-27 21:56:19: Device 'LED-Strahler 3' requested OTA
2022-11-27T20:56:19.892Z zigbee-herdsman:controller:endpoint CommandResponse 0x00158d00082038dd/1 genOta.queryNextImageResponse({"status":152}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-11-27T20:56:19.893Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00158d00082038dd:57583/1 (0,0,1)
2022-11-27T20:56:19.894Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":57583,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":45,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,90,2,152]}}
2022-11-27T20:56:19.895Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,14,36,1,239,224,1,1,25,0,45,0,30,4,25,90,2,152,211]
2022-11-27T20:56:19.897Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T20:56:19.910Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-11-27T20:56:19.911Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-11-27T20:56:19.911Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-11-27T20:56:19.911Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-11-27T20:56:19.912Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-11-27T20:56:19.914Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,45,235]
2022-11-27T20:56:19.915Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,45,235]
2022-11-27T20:56:19.916Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,45] - 235
2022-11-27T20:56:19.917Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":45}
2022-11-27T20:56:19.918Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

2022-11-27T20:56:42.337Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,145,152,1,13,0,40,0,9,14,7,0,0,12,1,54,1,0,16,18,12,17,1,0,0,0,145,152,29,225]
2022-11-27T20:56:42.339Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,145,152,1,13,0,40,0,9,14,7,0,0,12,1,54,1,0,16,18,12,17,1,0,0,0,145,152,29,225]
2022-11-27T20:56:42.339Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,145,152,1,13,0,40,0,9,14,7,0,0,12,1,54,1,0,16,18,12,17,1,0,0,0,145,152,29] - 225
2022-11-27T20:56:42.340Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":39057,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":40,"securityuse":0,"timestamp":462345,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,54,1,0,16,18,12,17,1,0,0,0]}}
2022-11-27T20:56:42.345Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"tra [... cut ...]
2022-11-27T20:59:04.501Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,58] - 252
2022-11-27T20:59:04.501Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":58}
2022-11-27T20:59:04.502Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

</details>

I tried to get only the parts relevant to each LED light's communication. I grouped them with an empty line between each request. Should I've missed something or you need any more information, please let me know. :)

Koenkk, 2022-11-28:
@mh166 we can try disabling the OTA response completely, how are you running z2m then I can provide instructions?

mh166, 2022-11-28:
I'm running it using docker:

```
docker run \
        --name zigbee2mqtt \
        --device=/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_50f7ab909345ec11ae0994fd6f14af06-if00-port0:/dev/ttyACM0 \
        -e DEBUG=zigbee-herdsman* \
        -p 8080:8080 \
        -v $(pwd)/data:/app/data \
        -v /run/udev:/run/udev:ro \
        -e TZ=Europe/Berlin \
        --restart=unless-stopped \
        koenkk/zigbee2mqtt
```

timdj, 2022-11-29:
My attempt to downgrade zigbee2mqtt within Home assistant failed. Reinstalled 1.28 while maintaining my configuration. Now the problem is gone.

mh166, 2022-12-01:
@Koenkk

I tried it, but it does make no difference. Even with this line commented out, I still get these OTA requests from the lights:

<details>
<summary>[Logfiles again]</summary>

```
2022-12-01T21:09:20.472Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,239,224,1,13,0,189,0,165,65,79,0,0,12,1,80,1,0,16,18,12,17,1,0,0,0,239,224,29,185]
2022-12-01T21:09:20.473Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,239,224,1,13,0,189,0,165,65,79,0,0,12,1,80,1,0,16,18,12,17,1,0,0,0,239,224,29,185]
2022-12-01T21:09:20.474Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,239,224,1,13,0,189,0,165,65,79,0,0,12,1,80,1,0,16,18,12,17,1,0,0,0,239,224,29] - 185
2022-12-01T21:09:20.476Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":57583,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":189,"securityuse":0,"timestamp":5194149,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,80,1,0,16,18,12,17,1,0,0,0]}}
2022-12-01T21:09:20.483Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":80,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4624,"imageType":4364,"fileVersion":1},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":57583,"endpoint":1,"linkquality":189,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
Zigbee2MQTT:debug 2022-12-01 22:09:20: Received Zigbee message from 'LED-Strahler 3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":4364,"manufacturerCode":4624}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-12-01 22:09:20: MQTT publish: topic 'zigbee2mqtt/LED-Strahler 3', payload '{"brightness":28,"color_mode":"color_temp","color_temp":370,"last_seen":"2022-12-01T22:09:20+01:00","linkquality":189,"state":"OFF"}'
Zigbee2MQTT:debug 2022-12-01 22:09:20: Device 'LED-Strahler 3' requested OTA
2022-12-01T21:09:20.528Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

2022-12-01T21:09:25.010Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,145,152,1,13,0,40,0,71,149,83,0,0,12,1,111,1,0,16,18,12,17,1,0,0,0,145,152,29,57]
2022-12-01T21:09:25.011Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,145,152,1,13,0,40,0,71,149,83,0,0,12,1,111,1,0,16,18,12,17,1,0,0,0,145,152,29,57]
2022-12-01T21:09:25.011Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,145,152,1,13,0,40,0,71,149,83,0,0,12,1,111,1,0,16,18,12,17,1,0,0,0,145,152,29] - 57
2022-12-01T21:09:25.013Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":39057,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":40,"securityuse":0,"timestamp":5477703,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,111,1,0,16,18,12,17,1,0,0,0]}}
2022-12-01T21:09:25.021Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":111,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4624,"imageType":4364,"fileVersion":1},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":39057,"endpoint":1,"linkquality":40,"groupID":0,"wasBroadcast":false,"destinationEndpoint":13}'
Zigbee2MQTT:debug 2022-12-01 22:09:25: Received Zigbee message from 'LED-Strahler 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":4364,"manufacturerCode":4624}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-12-01 22:09:25: MQTT publish: topic 'zigbee2mqtt/LED-Strahler 2', payload '{"brightness":51,"color_mode":"color_temp","color_temp":370,"last_seen":"2022-12-01T22:09:25+01:00","linkquality":40,"state":"OFF"}'
Zigbee2MQTT:debug 2022-12-01 22:09:25: Device 'LED-Strahler 2' requested OTA
2022-12-01T21:09:25.057Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

2022-12-01T21:09:28.046Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,198,225,1,13,0,43,0,210,122,86,0,0,12,1,198,1,0,16,18,12,17,1,0,0,0,145,152,28,195]
2022-12-01T21:09:28.048Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,198,225,1,13,0,43,0,210,122,86,0,0,12,1,198,1,0,16,18,12,17,1,0,0,0,145,152,28,195]
2022-12-01T21:09:28.049Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,198,225,1,13,0,43,0,210,122,86,0,0,12,1,198,1,0,16,18,12,17,1,0,0,0,145,152,28] - 195
2022-12-01T21:09:28.051Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":57798,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":43,"securityuse":0,"timestamp":5667538,"transseqnumber":0,"len":12," [... cut ...] r power state.

Whenever the flashing occurs, _no messages_ are sent over the Zigbee protocol. At least I don't see any in the herdsman debug log. So I don't think the flashing is related to the OTA messages. Rather, I would assume, that this might be a "safety feature" of the lights, perhaps? As if they wanted to tell you "Hey, be careful. You're still allowing any device to join your network!" This would go along with the fact that there is an initial delay before it starts flashing regularly.

Koenkk, 2022-12-01:
> Whenever the flashing occurs, no messages are sent over the Zigbee protocol. At least I don't see any in the herdsman debug log. 

This does not have to be the case. Not all messages are received by the coordinator and not all messages send to the coordinator are forwarded to zigbee-herdsman. To continue debugging this:

- Sniff the traffic of the moment it starts flashing (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)
- If we find nothing in this sniff, sniff the traffic when these bulbs are used with the original gateway (assuming bulbs do not flash when paired with the original gateway)

mh166, 2022-12-04:
I'm afraid I won't be able to help out here: neither have I got a CC2531 nor an original gateway. I'm only using [this Sonoff Zigbee 3.0 USB dongle](https://www.amazon.de/dp/B09KXTCMSC) (ID `10c4:ea60` according to `lsusb`).

I do however have an old QIVICON ZigBee Stick lying around. `lsusb` shows it with the ID `10c4:89fb`. I don't know if this stick might be of any help. It came together with a smart home solution of my former ISP. Never have used it actually...

timdj, 2022-12-12:
After not having this problem for 2 weeks mine started flashing again after a new device joined the network [Osram plug](https://www.zigbee2mqtt.io/devices/AB3257001NJ.html).

Ricox1975, 2023-01-07:
Gibts hier schon Neuigkeiten?

Crumar, 2023-01-07:
Still happening, nothing changed

ballwax, 2023-01-26:
Hello everyone,

I do not know what causes the flashing technically, but I can confirm that after setting permit_join on "false" and reboot Openhab system, the flashing disappears. Setting back on "true" and reboot, flashing behaviour reappears.

jaapio, 2023-02-12:
I have the same issue with paulmann [50064](https://www.zigbee2mqtt.io/devices/50064.html#paulmann-50064) in my case the device is reported as EcoDim after an update. This seems to be introduced in: https://github.com/Koenkk/zigbee2mqtt/issues/12412

Those device id's are matching. I'm not sure this is causing the issue.

elronzo, 2023-02-22:
I do definitely have this particular problem. As far as I can tell at the moment Paulmann devices show this behaviour. (371000001, 371000002)

As I am using zigbee2mqtt since yesterday I am not familiar with the debugging process. How can I get into it quickly?

I switched off "Permit join" at 3:00 AM and will observe if the problem reaccurs. (But I fear it does as I have watched a LEDVANCE 4058075208339 behaving just like that 5 minutes ago. At exactly the same moment I saw the zigbee2mqtt GUI showing a message "Starting network scan". This may be a coincidence but also maybe not.)

cantfindanam, 2023-02-22:
FYI. I turned off "permit join" after mh66 suggested it in November and haven't had this issue since then.

closed 2023-02-22 by cantfindanam

GrohmS2977, 2023-03-05:
> 

hi i got the same spots (Paulmann 92962) and the same problem as @Crumar above:
1.) SOLVED: randomly flashing lights --> solved by turning off Permit Join -> ok so far, can live with that
2.) NOT SOLVED: all of my 5 spots are randomly turning on at low brightness, without apparently being triggered by Home Assistant.

I got the Sonoff Coordinator Stick stick as well. Would be thankful for any idea.

Already thinking about switching to a different hub for these spots (maybe Hue or sth else, but actually I would like to avoid that) :-(

dubtec, 2023-04-14:
I also see this issue with two Paulmann controllers of type 500.46. When off, the connected light flashes at full brightness every now and then (seemingly no pattern). When on, the connected light would go off, then full brightness and then back go the previous setting.

Also, I observe that when the lights are switched off (=0% brightness) and I turn them on via Aqara Zigbee switch, they blink at full brightness and then return to the original state.

This started occuring in the last three weeks. I always try to be on the latest Z2M version. Controller is CC2352, its firmware may not be the latest.

I will review the permit join, but know for certain that it’s disabled in the UI.

byterazor, 2025-02-16:
Has there been any new information concerning this issue? I have the same behavior with two Paulman LED Panels. Permit join has always been disabled in my setup.