Koenkk/zigbee2mqtt #14736: [New device support]: Samotech SM309-S

sigtel, 2022-10-31:
### Link

https://www.samotech.co.uk/products/zigbee-dimmer-sm309-s/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x9035eafffe641d45","nwkAddr":65037,"manufId":4107,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"SM309-S","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,1794,2820,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SM309-S","manufacturerName":"Samotech","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r54"}},"genOta":{"attributes":{"currentFileVersion":55}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b694af","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b694af","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r54","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1461352984},"lastSeen":1667238891397,"defaultSendRequestWhen":"immediate"}

### Comments

I just received a Samotech SM309-S (as opposed to SM309) which comes up as unknown. Having duplicated the samotech SM309 code block in ./node_modules/zigbee-herdsman-converters/devices/samotech.js and renamed this to SM309-S it appears to function correctly. I am new to this (as of today) so not sure if this is simply the new name for the old(er) device or has new functionality - but it's a dimmer and it dims and turns off and on and reports it status just fine via the frontend and via homebridge / Siri etc.

closed 2022-11-01 by Koenkk

alexjfisher, 2023-01-11:
I _think_ this variant might also support energy monitoring.  I've got one on order.  Might not arrive for a few days, but if nobody beats me to it, I might see if I can figure out how to add the functionality to the converter.

alexjfisher, 2023-01-22:
@Koenkk Ta-dah! https://github.com/Koenkk/zigbee-herdsman-converters/pull/5384

Caveat.  I don't really know what I'm doing or how I made this seemingly work perfectly on my first attempt! :)