Koenkk/zigbee2mqtt #14740: Frontend empty

jagjordi, 2022-10-31:
### What happened?

The front-end is empty. Clicing "permit join" doesnt work. Devices does not show in the list. I am able to add devices by sending a mqtt command "zigbee2mqtt/bridge/request/permit_join". The device is successfully addded (I see it in the logs and in home assistant) but it still does not show in the frontend

### What did you expect to happen?

Front end works

### How to reproduce it (minimal and precise)

Install, open front end

### Zigbee2MQTT version

not visible, see https://imgur.com/NTA7JtF

### Adapter firmware version

also empty

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Debug log

```
[00:14:51] INFO: Preparing to start...
[00:14:52] INFO: Socat not enabled
[00:14:53] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-11-01 00:14:55: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-01.00-14-55' filename: log.txt
Zigbee2MQTT:info  2022-11-01 00:14:55: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-11-01 00:14:55: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:info  2022-11-01 00:14:56: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-11-01 00:14:56: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-11-01 00:14:56: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-11-01 00:14:56: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-11-01 00:14:56: Connecting to MQTT server at mqtt://192.168.1.110:1883
Zigbee2MQTT:info  2022-11-01 00:14:56: Connected to MQTT server
Zigbee2MQTT:info  2022-11-01 00:14:56: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-11-01 00:14:57: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info  2022-11-01 00:14:57: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0x00124b0024c2cf78","panID":6754},"permit_join":false,"version":"1.28.0"}'
Zigbee2MQTT:info  2022-11-01 00:14:57: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:warn  2022-11-01 00:17:50: Device '0x00158d0004840119' left the network
Zigbee2MQTT:info  2022-11-01 00:17:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004840119","ieee_address":"0x00158d0004840119"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-11-01 00:17:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00158d0004840119"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-11-01 00:24:28: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-11-01 00:24:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":20,"value":true},"status":"ok"}'
Zigbee2MQTT:info  2022-11-01 00:25:03: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-11-01 00:25:03: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":200,"value":true},"status":"ok"}'
Zigbee2MQTT:info  2022-11-01 00:25:16: Device '0x00158d0004840119' joined
Zigbee2MQTT:info  2022-11-01 00:25:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004840119","ieee_address":"0x00158d0004840119"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-11-01 00:25:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0004840119"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-11-01 00:25:16: Starting interview of '0x00158d0004840119'
Zigbee2MQTT:info  2022-11-01 00:25:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004840119","ieee_address":"0x00158d0004840119","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-11-01 00:25:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0004840119"},"type":"pairing"}'
Zigbee2MQTT:info  2022-11-01 00:25:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004840119","ieee_address":"0x00158d0004840119"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-11-01 00:25:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d0004840119"},"type":"device_announced"}'
Zigbee2MQTT:info  2022-11-01 00:25:39: Successfully interviewed '0x00158d0004840119', device has successfully been paired
Zigbee2MQTT:info  2022-11-01 00:25:39: Device '0x00158d0004840119' is supported, identified as: Xiaomi MiJia door & window contact sensor (MCCGQ01LM)
Zigbee2MQTT:info  2022-11-01 00:25:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"MiJia door & window contact sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Number of power outages","name":"power_outage_count","property":"power_outage_count","type":"numeric"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"MCCGQ01LM","options":[],"supports_ota":false,"vendor":"Xiaomi"},"friendly_name":"0x00158d0004840119","ieee_address":"0x00158d0004840119","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2022-11-01 00:25:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"MiJia door & window contact sensor","friendly_name":"0x00158d0004840119","model":"MCCGQ01LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
Zigbee2MQTT:info  2022-11-01 00:25:39: MQTT publish: topic 'homeassistant/sensor/0x00158d0004840119/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0004840119"],"manufacturer":"Xiaomi","model":"MiJia door & window contact sensor (MCCGQ01LM)","name":"0x00158d0004840119"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00158d0004840119","name":"0x00158d0004840119 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0004840119","unique_id":"0x00158d0004840119_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-11-01 00:25:39: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0004840119/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0004840119"],"manufacturer":"Xiaomi","model":"MiJia door & window contact sensor (MCCGQ01LM)","name":"0x00158d0004840119"},"device_class":"door","json_attributes_topic":"zigbee2mqtt/0x00158d0004840119","name":"0x00158d0004840119 contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x00158d0004840119","unique_id":"0x00158d0004840119_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
Zigbee2MQTT:info  2022-11-01 00:25:39: MQTT publish: topic 'homeassistant/sensor/0x00158d0004840119/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0004840119"],"manufacturer":"Xiaomi","model":"MiJia door & window contact sensor (MCCGQ01LM)","name":"0x00158d0004840119"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x00158d0004840119","name":"0x00158d0004840119 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0004840119","unique_id":"0x00158d0004840119_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
[... cut ...] ],"manufacturer":"Xiaomi","model":"MiJia door & window contact sensor (MCCGQ01LM)","name":"0x00158d0004840119"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d0004840119","name":"0x00158d0004840119 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0004840119","unique_id":"0x00158d0004840119_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
```

sushifrick, 2022-11-02:
I can confirm the problem, but it happened for me after upgrading to 1.28.1

Here some additional information:
- Devices is empty
- Dashboard is empty
- Map seems to work after loading, although the devices do not have an icon
- Settings
  - Settings is mostly empty
  - Tools Download state does work!
  - About is empty or "unknown"
  - Raw is nearly empty
- Groups is empty
- Ota is empty
- Logs do actually work
- Permit do join does not work

jagjordi, 2022-11-02:
n my case it turned out it was some issue with NGINX reverse proxy. If I access directly with the server IP it works

sushifrick, 2022-11-03:
Hi @jagjordi, 

thank you for the suggestion. In my case it seems to be the same problem. I am using zigbee2mqtt as a Plugin in Home Assistant and had to expose the port 8099 for the frontend in the configuration, which was not necessary before, as the docker containers run in a separate docker network. Not sure, what made these changes necessary, but I am glad I got access to the Gui for now.

Kind regards

closed 2022-11-03 by Koenkk

jagjordi, 2022-11-04:
Glad it helped @sushifrick .In my case, this error coincided with a server upgrade, so I was sure I had done something wrong when setting up the USB connection or the HA addon.

OptimusGREEN, 2022-11-27:
Anyone have a solution to get this working behind nginx proxy manager? Mine is empty using it.