Koenkk/zigbee2mqtt #14742: TuYa TS0601_human_presence_sensor is flooding the zb network

MattL0, 2022-11-01:
### What happened?

This device is flooding the zb network. 

### What did you expect to happen?

Being less chatty( alot lot less) on the zigbee network

### How to reproduce it (minimal and precise)

Install the device

### Zigbee2MQTT version

dev

### Adapter firmware version

dev most recent

### Adapter

sonoff cc2652p

### Debug log

[TuYa_TS0601_human_presence_sensor_Flood.log](https://github.com/Koenkk/zigbee2mqtt/files/9906256/TuYa_TS0601_human_presence_sensor_Flood.log)

danieledwardgeorgehitchcock, 2022-11-01:
This is coming from the device itself so, is not a Z2M issue - You may want to tweak some of the reporting periods / thresholds (if any) for the device to reduce the frequency of data

MattL0, 2022-11-01:
yes i figured this out, sorry if this was not clear, but I was wondering if someone know how to limit reports/thresholds for this device?

danieledwardgeorgehitchcock, 2022-11-01:
Do you have access to the GUI / Fronted? It can be done in the Reporting tab for the device

Koenkk, 2022-11-01:
@danieledwardgeorgehitchcock these TuYa devices don't support "reporting" (we don't even set up reporting on them). AFAIK spamming of this device cannot be reduced.

MattL0, 2022-11-09:
Just a thought. Is there a way to make this a endpoint/child node? It is now acting as a router (unlike the xiaomi aqara fp1 that is also plugged into the wall) I wonder if this is contributing to the spam.

Koenkk, 2022-11-10:
Z2M cannot chnage the type of a device (router/enddevice), this is determined by the firmware on the device.

closed 2022-11-10 by MattL0