Koenkk/zigbee2mqtt #14745: IKEA E2001/E2002 actions change on/off state of connected zigbee router power outlets

nweiser94, 2022-11-01:
### What happened?

I extended my zigbee network recently with a few power outlets from Aqara and IKEA Traedfri acting as router devices. When i added an IKEA E2001/E2002 remote to the network through one of the zigbee routers, i noticed that triggering the on/off actions from the remote do affect the on/off state of the zigbee power outlets as well. When connecting the remote directly to my main zigbee coordinator, the problem does not occur at all.

### What did you expect to happen?

I expected that the on/off actions from the IKEA E2001/E2002 remote do not affect the state of the power outlets at all, unless explicitly configured through an automation.

### How to reproduce it (minimal and precise)

- Add a zigbee power outlet router device to the network
- Add the IKEA E2001/E2002 remote to the network through the router device
- Trigger the on/off actions from the remote
- The power outlet that acts as the router between coordinator and E2001/E2002 remote will be turned on/off accordingly

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[debug_2022_11_01.log](https://github.com/Koenkk/zigbee2mqtt/files/9908615/debug_2022_11_01.log)

vetinari, 2022-11-05:
same as #13903?

nweiser94, 2022-11-05:
Thanks for the hint. Seems to be a similar issue. Though in my case i experienced it not only with the aqara plug but also with the Traedfri plugs. Gonna check that later in detail.

closed 2022-12-14 by github-actions[bot]