Koenkk/zigbee2mqtt #14748: [Feature request]: Use MQTT update feature in HA

andriej, 2022-11-01:
### Is your feature request related to a problem? Please describe

Since 2022.11 in HA there will be possibility to use 'update' platform for firmware updates
That could be used for OTA-available firmwares in z2m

https://github.com/home-assistant/core/pull/81131

### Describe the solution you'd like

Show 'update' possibility in HA when new firmware available for device

### Describe alternatives you've considered

n/a

### Additional context

n/a

bkbilly, 2022-11-09:
I've just created a new python package which works as a service that you might be interested on trying out.
It reads the current version and the latest version of zigbee2mqtt and other services and creates a new update sensor with the ability to install the newest version.
You can find it here: https://pypi.org/project/service-updater/

It would be great if we could have this feature directly on zigbee2mqtt, but for now I hope this helps.

Koenkk, 2022-11-09:
Added in https://github.com/Koenkk/zigbee2mqtt/pull/14905

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

PR #14905 (2022-11-09): Support Home Assistant MQTT update
This PR adds support for the Home Assistant [MQTT update](https://www.home-assistant.io/integrations/update.mqtt/). 

Fixes #14748

A limitation is that Zigbee firmwares don't have proper version numbering. Each firmware has a `fileVersion` number (which differs from the device `swBuildId`). Therefore I chose to map the `installed_version` to the `swBuildId`. In case a new firmware is available `latest_version` becomes `newer` (if not it will be equal to the `installed_version`).
closed 2022-11-09 by Koenkk