Koenkk/zigbee2mqtt #14749: Zigbee2MQTT not accepting Install Codes

joaquinvacas, 2022-11-01:
### What happened?

Tried to install a Schneider Electric CCT5010-0002W micro dimmer module with Zigbee2MQTT

Actually it joins, shows all the data like it was joined correctly.

[image]

But after that, I can't change anything or use the device at all.

I noticed the device has a Install Code printed on it's side.

[image]

The problem here is when I try to add the install code, this shows instead.

[image]

How can I get Z2M to support the shorter install code? Or maybe Schneider is hiding some part of the install code that the Wiser app assumes?

Anyway, I can't test it, don't know how to get the "full 95 or 91" install code.

Also, Zigbee2MQTT shows it like a CCT5010-0001 device and not CCT5010-0002W one.

The 2W variant is the updated one, same functions but with bluetooth (didn't need it for joining)

So, here it's joined but after a short time, it shows "Unavailable" like if it was turned off.

### What did you expect to happen?

Accept install code to let control the joined device, 

### How to reproduce it (minimal and precise)

1. Try to add install code
2. You get the error showing it's not 95 or 91 char long.
3. Try to pair anyway
4. Try to control it, it won't work

### Zigbee2MQTT version

1.28

### Adapter firmware version

 20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus by ITead CC2652P

Koenkk, 2022-11-01:
The install code is (AFAIK) only required for joining, and since the device joins fine there is no need to add support for this alternative format.

> But after that, I can't change anything or use the device at all.

What happens if you turn on/off the device?

joaquinvacas, 2022-11-02:
> The install code is (AFAIK) only required for joining, and since the device joins fine there is no need to add support for this alternative format.
> 
> > But after that, I can't change anything or use the device at all.
> 
> What happens if you turn on/off the device?

Actually, it does nothing. Keeps trying like if the device were offline.

danieledwardgeorgehitchcock, 2022-11-02:
@Koenkk - I do think that there may be an issue with the Install Code logic - the code above is valid (as per the Z-Stack specs) with an a hex encoded 8 character IEEE (`000D6FFFFEC6A8A1`) and an 18 character (`1C2817051E362613411F7F676AB2E4531379`) Install Code - This follows the same format as the Bosch Outdoor Alarms that originally highlighted the need for the install codes

joaquinvacas, 2022-11-02:
> @Koenkk - I do think that there may be an issue with the Install Code logic - the code above is valid (as per the Z-Stack specs) with an a hex encoded 8 byte IEEE (`000D6FFFFEC6A8A1`) and an 18 byte (`1C2817051E362613411F7F676AB2E4531379`) Install Code - This follows the same format as the Bosch Outdoor Alarms that originally highlighted the need for the install codes.
> 
> [image]

How can I pass my install code thenm

danieledwardgeorgehitchcock, 2022-11-02:
I have adapted an extension (not external converter) that Koenkk created when diagnosing the Bosch Alarms.

If you add this extension, enable debug logging, I have hard-coded the install code into it. Restart Z2M and hopefully you will see some good log messages [Install Code Extension](https://gist.github.com/danieledwardgeorgehitchcock/744950f11aa50428396ec88301a62963)

joaquinvacas, 2022-11-12:
> The install code is (AFAIK) only required for joining, and since the device joins fine there is no need to add support for this alternative format.
> 
> > But after that, I can't change anything or use the device at all.
> 
> What happens if you turn on/off the device?

Okay, somehow that device started working that same night by its own, it was when 1.28.1 arrived so I thought it was the update that fixed this.

Just installed another two ones today, check this.

[image]
[image]
[image]

Also, joined without install code too.

Don't know what changes but I suppose it will work in no time, but feels weird.

Don't know how to debug what happens, why it takes hours to work as expected.

Also, the dimmer stays in red like there were an error, but after it begins to work again, it turns green forever.

joaquinvacas, 2022-11-12:
> I have adapted an extension (not external converter) that Koenkk created when diagnosing the Bosch Alarms.
> 
> If you add this extension, enable debug logging - I have hard-coded the install code into it. Restart Z2M and hopefully you will see some good log messages [Install Code Extension](https://gist.github.com/danieledwardgeorgehitchcock/744950f11aa50428396ec88301a62963)
> 
> Just to temper expectations though, as Koenkk said above, if the device joins your network without the need for the install code, I don't expect the addition of it to add any additional functionality to what you have already reported...

But this is hardcoded, not able to test it.

I'm going to install a new one right now, I can test it.

How do I change the install code? I'll check in a minute.

joaquinvacas, 2022-11-12:
[image]

That's what I get when I try to use the .js

joaquinvacas, 2022-11-13:
Also, noticed some values that are exposed on this device aren't working (like previous state)

[image]

I think there are some minor differences between CCT5010-0002W and the CCT5010-0001 that Z2M recognizes.

joaquinvacas, 2022-11-15:
@Koenkk So, I didn't report it, but the way to get those working is by stopping Zigbee2Mqtt for a while and then start it again.

Something changes in the device and begins to work again.

joaquinvacas, 2022-12-06:
@Koenkk Some of the settings doesn't work well with this dimmer. 

I think this issue may have to be moved to "New Device support"?

joaquinvacas, 2022-12-21:
> @joaquinvacas can you provide all the settings that don't work. Likely the device doesn't support (simply remove them is the solution)

The "current level startup". But the device does accept it.
Though, I don't have any master device from Schneider to read it's values.

I think there may be some differences between the first hw version (0001) and this (0002W) and the way it changes the previous value.

joaquinvacas, 2022-12-23:
@Koenkk Sorry, it's in Spanish but as you can see, the manual of the newer device shows "Último nivel (B)" that means last/previous used level.

[image]

Koenkk, 2022-12-23:
Can you sniff the traffic when setting this? maybe they use a special command. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

danieledwardgeorgehitchcock, 2022-12-23:
@Koenkk - I think the install code functionality is generally broken - I tried using it for my bosch siren the other day, which is what originally drove the requirement and the testing for install codes, and couldn't get Z2M to accept the string..

Koenkk, 2022-12-24:
@danieledwardgeorgehitchcock but this was working before right?

danieledwardgeorgehitchcock, 2022-12-24:
Yes it was - I think it stopped when the new code type was introduced..?

Koenkk, 2022-12-24:
Can you try with z2m 1.27.1, this includes the install code support but not support for the new type.

volodymyrpavlenko, 2023-01-16:
I've just bought a schneider Micro Module, CCT5010-0002, and it does not seem to work at all in Zigbee2MQTT
It seems to be correctly discovered, but none of the controls work. 
If I go to dev console, I can menually control some of the values, like on/off, brightness, min/max brightness.

Just to add context, the status led on the device itself shortly becomes green after restart, and then it's red all the time when the lamp is on. I assume this might be normal though.

joaquinvacas, 2023-01-17:
> I've just bought a schneider Micro Module, CCT5010-0002, and it does not seem to work at all in Zigbee2MQTT It seems to be correctly discovered, but none of the controls work. If I go to dev [...]

Shut down Z2M, wait 5min and then start it up again.

joaquinvacas, 2023-01-17:
> Can you sniff the traffic when setting this? maybe they use a special command. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Don't have a sniffer! 😢

Maybe somebody from here could do?

volodymyrpavlenko, 2023-01-17:
> Shut down Z2M, wait 5min and then start it up again.

Hm, yes, this is the case

What I found, when the requests were timing out, they were being sent to endpoint 242 instead of endpoint 3.

After restart, requests are sent to endpoint 3.

Maybe some initialization bug?

joaquinvacas, 2023-02-17:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 7 days

Not staled.

Actually this should be changed to "new device support" instead of the actual name. @Koenkk

joaquinvacas, 2023-03-21:
Changed title

joaquinvacas, 2023-04-21:
Not stale. Just not hardware available for debugging.

Hedda, 2023-05-09:
FYI, if you are using ConBee or RaspBee (deconz Zigbee Coordinator adapter from dresden elektronik) then see related issue -> https://github.com/Koenkk/zigbee2mqtt/issues/17492

joaquinvacas, 2023-06-12:
@Koenkk how can I fix this? I mean, is there any way for debugging it without hardware?

joaquinvacas, 2023-07-13:
Not stale.

patapovich, 2023-08-20:
Not stale

closed 2023-09-27 by github-actions[bot]

slashbv, 2023-09-29:
Any updates on this ? I also bought one and it doesn't work well.

fariouche, 2023-11-17:
is this device still not working? I was planning to buy one.
Inhave a sniffer if we can make it work

joaquinvacas, 2023-11-17:
It works almost perfect, there are some minor settings that don't seem to
be "refined" but OK, completely usable.