Koenkk/zigbee2mqtt #14750: TS0012 turn on and turn off both gangs at the same time

quinzico, 2022-11-01:
### What happened?

I've installed 1 TS0012  2 Gang without neutral but every time i turn on on Homeassistant/Z2M one switch (left) the other one is also turned on. I can turn on independently on the switch and works fine. 

### What did you expect to happen?

when i ask to turn on the left gang, it should turn on only the left gang.

On MQTT:

Topic: zigbee2mqtt/ZG-SW-L-Garagen/left/set
Payload: ON

Only the left gang should turn on. The same for OFF or independently to the right gang.

### How to reproduce it (minimal and precise)

Topic: zigbee2mqtt/ZG-SW-L-Garagen/left/set
Payload: ON

where ZG-SW-L-Garagen is the friendly name of the device

### Zigbee2MQTT version

1.28

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee Dongle zStack3x0

### Debug log

debug 2022-11-01 14:51:59Received MQTT message on 'zigbee2mqtt/ZG-SW-L-Garagen/left/set' with data 'ON'
debug 2022-11-01 14:51:59Publishing 'set' 'state' to 'ZG-SW-L-Garagen'
debug 2022-11-01 14:51:59Received Zigbee message from 'ZG-SW-L-Garagen', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2022-11-01 14:51:59MQTT publish: topic 'zigbee2mqtt/ZG-SW-L-Garagen', payload '{"linkquality":18,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"ON","state_right":"OFF","switch_type":"toggle"}'
debug 2022-11-01 14:52:00Received Zigbee message from 'ZG-SW-L-Garagen', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
info 2022-11-01 14:52:00MQTT publish: topic 'zigbee2mqtt/ZG-SW-L-Garagen', payload '{"linkquality":18,"power_on_behavior":"off","power_on_behavior_left":"off","state_left":"ON","state_right":"ON","switch_type":"toggle"}'
debug 2022-11-01 14:52:03Received MQTT message on 'zigbee2mqtt/ZG-SW-L-Garagen/left/set' with data 'OFF'
debug 2022-11-01 14:52:03Publishing 'set' 'state' to 'ZG-SW-L-Garagen'

Koenkk, 2022-11-01:
Please check if this is solved in 1.28.1, if not provide the exact model of this device

quinzico, 2022-11-01:
Hi,

I’ve updated to 1.28.1 and the issue still persists.

Zigbee dongle details:

DEVLINKS: >-
  /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_66788ea5d312ec11818823c7bd930c07-if00-port0
  /dev/serial/by-path/pci-0000:00:06.0-usb-0:2:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: /devices/pci0000:00/0000:00:06.0/usb2/2-2/2-2:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: Sonoff_Zigbee_3.0_USB_Dongle_Plus
ID_MODEL_ENC: Sonoff\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus
ID_MODEL_ID: ea60
ID_PATH: pci-0000:00:06.0-usb-0:2:1.0
ID_PATH_TAG: pci-0000_00_06_0-usb-0_2_1_0
ID_REVISION: '0100'
ID_SERIAL: ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_66788ea5d312ec11818823c7bd930c07
ID_SERIAL_SHORT: 66788ea5d312ec11818823c7bd930c07
ID_TYPE: generic
ID_USB_DRIVER: cp210x
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_VENDOR: ITead
ID_VENDOR_ENC: ITead
ID_VENDOR_ID: 10c4D: '12788642'

Koenkk, 2022-11-02:
I need the model details of your TS0012 (z2m frontend -> device -> about tab)

quinzico, 2022-11-02:
Friendly name*ZG-SW-L-Garagen*Description

Last seenN/AAvailabilityOnlineDevice typeEndDeviceZigbee ModelTS0012Zigbee
Manufacturer_TZ3000_jl7qyupfDescription2 gang switch module - (without
neutral)Support status

Supported
IEEE Address0xa4c138c4efadf6dcNetwork address0xDFC5ManufacturerTuYa
<https://www.zigbee2mqtt.io/supported-devices/#v=TuYa>Model
TS0012_switch_module
<https://www.zigbee2mqtt.io/devices/TS0012_switch_module.html#tuya-ts0012_switch_module>
Power
Interview completedTrue

à(s) 16:54:

Koenkk, 2022-11-02:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-02 by Koenkk

quinzico, 2022-11-04:
Thanks @Koenkk, when this will be available on the Master branch?

Koenkk, 2022-11-05:
With the next release on 1 December.