Koenkk/zigbee2mqtt #14763: IKEA E1744 SYMFONISK sound controller Flooding Network

j6s33m, 2022-11-01:
### What happened?

As of today, my zigbee2mqtt network is being flooded with actions from my 2 IKEA E1744 Symfonisk sound remotes. I'm not even moving the dial or clicking on the buttons and they are sending thousands upon thousands of commands at breakneck speed.

### What did you expect to happen?

I expect them to only show actions when I physically take actions on the remotes.

### How to reproduce it (minimal and precise)

I removed and readded and am still having same issue.

### Zigbee2MQTT version

1.28.1-dev commit: c3f6e79

### Adapter firmware version

0x26780700

### Adapter

Conbee II

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9914007/log.txt)

j6s33m, 2022-11-01:
I also tried going backwards off of the "dev" version and readding the devices ... same issue.

marazmarci, 2022-11-02:
I did something, I don't remember exactly, but it eventually stopped. Probably I just rotated it a bit and pushed the button.

I also have an interesting observation: When I pushed the button, the event got published only once to MQTT. After restarting Z2M, again, only one button push event got delivered to MQTT, subsequent events were not. However, I see the Zigbee message received logs for every button press.

j6s33m, 2022-11-02:
Ya this isn't working for me. Something must have triggered this in the recent release. I have pulled these devices from my zigbee network until there is a solve.

daranto, 2022-11-02:
I have the same issue. Only one event. I thougt my remote is defect but after the docker restart it will send one event again.

wokkeltje13, 2022-11-02:
I probably have the same issue after the update from 1.28.0 to 1.28.1.
My E1744 is no longer working, revert to 1.28.0 and it is working again.
In the log I see
```
Info 2022-11-02 19:03:49MQTT publish: topic 'zigbee2mqtt/Ikea-RotateButton-1', payload '{"action":null,"battery":34,"brightness":255,"last_seen":"2022-11-02T19:03:10+01:00","linkquality":83,"update":{"state":"idle"},"update_available":false}'
Info 2022-11-02 19:03:49MQTT publish: topic 'zigbee2mqtt/Ikea-RotateButton-1', payload '{"action":null,"battery":34,"brightness":255,"last_seen":"2022-11-02T19:03:10+01:00","linkquality":83,"update":{"state":"idle"},"update_available":false}'
Info 2022-11-02 19:03:49MQTT publish: topic 'zigbee2mqtt/Ikea-RotateButton-1', payload '{"action":null,"battery":34,"brightness":255,"last_seen":"2022-11-02T19:03:10+01:00","linkquality":83,"update":{"state":"idle"},"update_available":false}'
```
There is no reaction if I rotate or push action remains null

Koenkk, 2022-11-02:
@j6s33m can you try setting `legacy: false` for this device in the configuration.yaml? https://www.zigbee2mqtt.io/devices/E1744.html#legacy-integration

wokkeltje13, 2022-11-02:
For me legacy is off without any difference

kokoon, 2022-11-02:
Same here, E1744 was working beautifully in legacy mode, now it's flooding whenever turned after restarting the addon in hassio. If first even is click, the flooding doesn't happen but it still crashes and doesn't report any more events. I nuked everything I know about this device and it's always the same after adding it back.

Was sure it's a hardware issue, changed the battery too. Now to find a way to revert the version.

j6s33m, 2022-11-02:
> @j6s33m can you try setting `legacy: false` for this device in the configuration.yaml? https://www.zigbee2mqtt.io/devices/E1744.html#legacy-integration

They were in this mode before. Now I've readded them and they seem to stable, but the actions are only showing up in "debug" not in "info" logs so nothing comes through MQTT

Koenkk, 2022-11-02:
@j6s33m 

> They were in this mode before. Now I've readded them and they seem to be stable, but the actions are only showing up in "debug" not in "info" logs so nothing comes through MQTT

Can you provide the debug log of that? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

kokoon, 2022-11-02:
Can confirm reverting to 1.28.0-1 solves the issue.

j6s33m, 2022-11-02:
> @j6s33m
>
>
> Can you provide the debug log of that?
>
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

[example logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/9924235/example.logs.txt)

olympia, 2022-11-03:
I have the same issue. Only one event passed to mqtt per Z2M boot.

Ionshard, 2022-11-03:
Same issue with my SYMFONISK device after updating to 1.28.1, the device no longer sent messages over MQTT and enabling the debug logs shows the device absolutely flooding the logs. Downgraded my docker image back to 1.28.0 and it works fine again. No other changes made but a downgrade.

habitoti, 2022-11-03:
Same for me, but unfortunately I can‘t go back to 1.28.0…after restoring that backup, the UI fails to start with a „502 Bad Gateway“.

Koenkk, 2022-11-03:
Published a possible fix.

Changes will be available in the dev branch in 30 mins from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Cadsters, 2022-11-03:
Thanks for the fix! 
Works for me now with latest-dev.

j6s33m, 2022-11-03:
@Koenkk Confirming it's working in the new dev.  Thank you!

Koenkk, 2022-11-04:
Great, I've published a hotfix release: 1.28.2, enjoy!

closed 2022-11-04 by Koenkk

nipowers, 2023-03-12:
Hi @Koenkk 

This issue seems to have resurfaced in 1.30.2-1 or I missed a step to correct the issue. The dial is in legacy mode. As soon as I rotate the dial with an automation set to increase brightness until stop is received the system gets flooded with messages until I restart Z2M even is I disable the automations within HA.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10952189/log.txt)

patrex0304, 2023-05-13:
Do you use the Light Hook of Awesome HA Blueprints? I had since some updates the same Problem with the Media Hook of Awesome HA Blueprints.

quenthal, 2024-01-09:
I also started to have this old flooding with E1744. Using latest firmware no E1744 and latest z2m. What is strange is that log was flooding even if I removed the battery. Maybe the problem was some powered device relaying ghost device..?