Koenkk/zigbee2mqtt #14764: [New device support]: EGLO 31703

Richardmarkink, 2022-11-01:
### Link

https://www.eglo.com/en/ceiling-light-rovito-z-900091.html

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c1389eafc22aa4","nwkAddr":46112,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EBF_RGB_Zm","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EBF_RGB_Zm","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.12_250"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.3.12_250","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1667337110477,"defaultSendRequestWhen":"immediate"}/app #

### Comments

I am a newbie adding unsuported devices, did not manage to do it myself.
The Light contans also an RGB Ring and the normal light that is not rgb but 2700-6500K

### Supported color modes

Color Temperature, RGB

Koenkk, 2022-11-02:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Richardmarkink, 2022-11-02:
@Koenkk 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

Koenkk, 2022-11-02:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-02 by Koenkk