Koenkk/zigbee2mqtt #14768: Philips Hue tap dial switch

yltsa, 2022-11-02:
### What happened?

Hi, the pairing or other functioning of Philips Huea tap dial switches are not OK.

I have the newest firmware from Philips which updated in Hue's own hub before trying to pair to Z2M.

BR, Ilkka

### What did you expect to happen?

The pairing process goes through but the tap dial switch does not stop led’s blinking after the pairing. So something is not OK.

The pairing is OK when using Hue’s own hub.

### How to reproduce it (minimal and precise)

Put tap dial switch to pairing mode. It needs usually, that you press the switch under the small hole 5-10 sec before Z2H detects the tap dial. After that the pairing goes thought in Z2M but the tap dial continues blinking.. (pairing mode still..?)

### Zigbee2MQTT version

1.28.1-1

### Adapter firmware version

I cannot find this ..?

### Adapter

Slaesh CC2652RB

chairstacker, 2022-11-12:
I'm having the same issue; this is what I see in the logs:
```
Debug Received Zigbee message from '0x001788010d12d123a', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":33569555,"imageType":289,"manufacturerCode":4107}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'RDM002' and manufacturer name 'Signify Netherlands B.V.'
```

chairstacker, 2022-11-16:
I had to add manually upgrade the Home Assistant Add-On from V1.18.1-1 to V1.28.2-1 - followed the steps at [this link](https://community.home-assistant.io/t/zigbee2mqtt-add-on-upgrade-instructions-requested/286697/9) at it's working great for me now.

closed 2022-12-25 by github-actions[bot]