Koenkk/zigbee2mqtt #14770: Zigbee2MQTT fails to start

AmigoDen, 2022-11-02:
### What happened?

After Update of zigbee2mqtt 1.28,1-1 home assistant addon the logfile is swamped with following errors:
_[12:16:02] INFO: Preparing to start...
[12:16:02] INFO: Socat not enabled
[12:16:04] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:394
				.withSensor(['IN', 'OU', 'AL'], ea.STATE_SET)
				 ^
TypeError: exposes.climate(...).withLocalTemperature(...).withSetpoint(...).withLocalTemperatureCalibration(...).withRunningState(...).withPreset(...).withSensor is not a function
    at /app/dist/util/externally-loaded.js:394:6
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)_

This prevents Zigbee2Mqtt start and work. I can go back to a previous version for now though.

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20220928

### Adapter

zStack3x0

arjenanker, 2022-11-02:
Just updated to 1.28.1-1 in Homeassistant too, and my addon refuses to start too. Different errors though... Already uninstalled, entered the configuration from scratch, switched to Edge to see if that made a difference, nothing. Unfortunately there's no backup to roll back to and no easy way to install an older version in HA...

This is what my logs says:
`[12:18:38] INFO: Preparing to start...
[12:18:38] INFO: Socat not enabled
[12:18:38] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2022-11-02 12:18:40: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'ID')
    at /app/lib/extension/bridge.ts:700:23
    at Array.map (<anonymous>)
    at Bridge.publishGroups (/app/lib/extension/bridge.ts:698:45)
    at Bridge.start (/app/lib/extension/bridge.ts:111:20)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.callExtensions (/app/lib/controller.ts:314:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5))
Zigbee2MQTT:error 2022-11-02 12:18:40: Failed to call 'Groups' 'start' (TypeError: Cannot read properties of undefined (reading 'ID')
    at /app/lib/extension/groups.ts:93:66
    at Array.find (<anonymous>)
    at Groups.syncGroupsWithSettings (/app/lib/extension/groups.ts:93:33)
    at Groups.start (/app/lib/extension/groups.ts:44:20)
    at Controller.callExtensions (/app/lib/controller.ts:314:40)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5))
Zigbee2MQTT:error 2022-11-02 12:18:40: Failed to call 'Availability' 'start' (TypeError: Cannot read properties of undefined (reading 'isGroup')
    at Object.isAvailabilityEnabledForEntity (/app/lib/util/utils.ts:286:16)
    at Availability.publishAvailabilityForAllEntities (/app/lib/extension/availability.ts:124:23)
    at Availability.start (/app/lib/extension/availability.ts:119:14)
    at Controller.callExtensions (/app/lib/controller.ts:314:40)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5))
Zigbee2MQTT:error 2022-11-02 12:18:40: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'isDevice')
    at HomeAssistant.getDiscoverKey (/app/lib/extension/homeassistant.ts:1111:23)
    at HomeAssistant.discover (/app/lib/extension/homeassistant.ts:1116:34)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:106:18)
    at Controller.callExtensions (/app/lib/controller.ts:314:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5))
Zigbee2MQTT:error 2022-11-02 12:18:40: Error: Cannot read properties of undefined (reading 'isDevice')
Zigbee2MQTT:error 2022-11-02 12:18:40: Error: Cannot read properties of undefined (reading 'isDevice')
TypeError: Cannot read properties of undefined (reading 'ID')
    at State.exists (/app/lib/state.ts:68:49)
    at Controller.start (/app/lib/controller.ts:158:32)
    at start (/app/index.js:109:5)`

AmigoDen, 2022-11-03:
I restored from backup version 1.28.0-1 and everything works again.
In my configuration there is an external converter for the AVATTO ME81H. 
I think it's the external zigbee converter, but I don't understand what the problem is and how to fix it.

Koenkk, 2022-11-03:
Remove the `withSensor` part from the external converter.

AmigoDen, 2022-11-03:
Thanks [Koenkk](https://github.com/Koenkk)
It's working.
Maybe you can add support Floor heating thermostat AVATTO ME81H to the release?

external converter for the AVATTO ME81H:
[avatto_me81h-2.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/9927009/avatto_me81h-2.js.txt)

Koenkk, 2022-11-03:
I can only accept it when the new tuya format is used (with `tuyaDatapoints` mapped in `meta). Maybe you can ask the author to refactor it. An example can be found here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/ec62e465308a6f0051bfcb00adefe7d38f5e37b3/devices/tuya.js#L2100

closed 2022-12-11 by github-actions[bot]