Koenkk/zigbee2mqtt #14771: BRT-100-TRV null system_mode

marcosimeone, 2022-11-02:
### What happened?

Since last update (**1.28.1**) all **BRT-100-TRV** devices report **null** **system_mode**, making home assistant integration not properly functioning.

### Zigbee2MQTT version

1.28.1 commit: 48117fd

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9919146/log.txt)

callacomp, 2022-11-02:
the change came from this pull request. https://github.com/Koenkk/zigbee-herdsman-converters/pull/4832

This has broken my TRVs also. They now say unknown. In the operation dropdown I can now only see heat (which is part of the update) but if I select it it changes back to unknown again, error in HA mqtt:
[image]
[image]
doesn't reflect in history or logbook
[image]

HA log
[image]

nothing in Z2m logs. altered heating, then temp, only logged temp change
`Debug 2022-11-02 16:07:29Received MQTT message on 'zigbee2mqtt/Dining room TRV/set/current_heating_setpoint' with data '8.0'
Debug 2022-11-02 16:07:29Publishing 'set' 'current_heating_setpoint' to 'Dining room TRV'
Debug 2022-11-02 16:07:29Received Zigbee message from 'Dining room TRV', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,8],"type":"Buffer"},"datatype":2,"dp":2}],"seq":14}' from endpoint 1 with groupID 0
Info 2022-11-02 16:07:29MQTT publish: topic 'zigbee2mqtt/Dining room TRV', payload '{"battery":88,"boost_heating":"OFF","boost_heating_countdown":0,"boost_heating_countdown_time_set":300,"child_lock":"LOCK","current_heating_setpoint":8,"eco_mode":"OFF","eco_temperature":17,"last_seen":"2022-11-02T16:07:29+00:00","linkquality":116,"local_temperature":17,"local_temperature_calibration":-1,"max_temperature":29,"min_temperature":6,"position":75,"preset":"manual","programming_mode":"11:06/18.5°C 14:32/21.5°C 11:30/22°C 17:30/23.5°C 07:02/22.5°C 12:01/23.5°C 12:31/22.5°C 16:31/21°C 04:01/20.5°C 17:28/20°C 13:30/21°C 18:30/20°C","running_state":null,"update":{"state":null},"update_available":null,"valve_state":"CLOSED","window":"CLOSED","window_detection":"OFF"}'
`

There's also no info in Z2M screen for new fields:
[image]

leroyloren, 2022-11-02:
try to force a change on the valve, e.g. by switching programming/manual

FredWst, 2022-11-02:
Hi,

If u put in programming mode and back to manual, u'll get back to normal status. (manual & heat)

Fred

xworld21, 2022-11-02:
I suspect this is working as intended... sorry about the confusion! As soon as `preset` changes, the valve will report the `system_mode`, and it should work smoothly from then onward. In hindsight, I could have added `system_mode` to the `valve_state` so that it updates more frequently, but after all this kind of glitch can only happen during an update – joining a new valve should return all properties right away.

Re the mqtt.climate error, I thought Home Assistant would work in optimistic mode about `system_mode`, i.e. if there is no set command, it thinks the change has gone ahead. But now I think I misunderstood the documentation. I'll look into this more carefully and see if there is a proper fix for this (either here on in Home Assistant).

closed 2022-11-07 by Koenkk