Koenkk/zigbee2mqtt #14772: New Tuya TS0601_thermostat (_TZE200_cpmgn2cf) not showing battery_low value

X4LD1M0, 2022-11-02:
### What happened?

After installing a new TS0601 (_TZE200_cpmgn2cf) I noticed that the battery_low field is missing.
```
    "auto_lock": "MANUAL",
    "away_mode": "OFF",
    "away_preset_days": 1,
    "away_preset_temperature": 16,
    "boost_time": 300,
    "child_lock": "LOCK",
    "comfort_temperature": 21,
    "current_heating_setpoint": 15,
    "eco_temperature": 16,
    "force": "normal",
```
I also notice that the new one is behaving different, as it uses 5% steps for the valve motor, while the old one does 25% steps. But I can't see anywhere any h/w or s/w versions to compare.

### What did you expect to happen?

The same fields as per the older TS0601 (_TZE200_2atgpdho):

```
    "auto_lock": "MANUAL",
    "away_mode": "OFF",
    "away_preset_days": 1,
    "away_preset_temperature": 15,
    "battery_low": false,
    "boost_time": 300,
    "child_lock": "UNLOCK",
    "comfort_temperature": 21,
    "current_heating_setpoint": 18,
    "eco_temperature": 16,
    "force": "normal",
```

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20220219

### Adapter

Electrolama zzh!

### Debug log

[2022-11-02.09-41-08.log](https://github.com/Koenkk/zigbee2mqtt/files/9919579/2022-11-02.09-41-08.log)

danieledwardgeorgehitchcock, 2022-11-02:
These both use the same converter so it may be that one is not reporting the `batteryLow` data point whereas the other is. The state attributes are only populated / cached when data is received for a particular data point so, it may be that your device just hasn't reported a low battery yet

X4LD1M0, 2022-11-02:
But shouldn't it then, by default, provide the value 'false'? I would expect it to send a false when I put the new batteries in. So you say have to wait for these to run out, then hope it sets it to true, then when I put in new batteries it would change it back to false?

danieledwardgeorgehitchcock, 2022-11-02:
Or the device doesn't report it's battery status

X4LD1M0, 2022-12-03:
I have now notice that it now showing as enabled (low battery), so I guess the initial state is unknown until the first change, so will close it now.

closed 2022-12-03 by X4LD1M0