Koenkk/zigbee2mqtt #14778: Zigbee2mqtt 1.28.1 all devices off line

frabar55, 2022-11-02:
### What happened?

after the upgrade all devices go off line
[config_entry-mqtt-161e9aac3932f59023a2721348dd1d48.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/9923032/config_entry-mqtt-161e9aac3932f59023a2721348dd1d48.json.txt)

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8

HelpITru, 2022-11-02:
the same 

Zigbee2MQTT:warn  2022-11-02 19:33:39: Failed to ping 'розетка в сарай' (attempt 1/1, Read 0x00124b0009fe0b10/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Zigbee2MQTT:debug 2022-11-02 19:33:42: Active device 'розетка в сарай' was last seen '40.67' minutes ago.
Zigbee2MQTT:warn  2022-11-02 19:33:53: Failed to ping 'kabinet' (attempt 1/1, Read 0x0c4314fffe9adf00/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) **failed (sendZclFrameToEndpointInternal error))**
Zigbee2MQTT:debug 2022-11-02 19:33:56: Active device 'kabinet' was last seen '40.65' minutes ago.
Zigbee2MQTT:debug 2022-11-02 19:34:55: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:warn  2022-11-02 19:37:03: Failed to ping 'rele1' (attempt 1/1, Read 0xa4c138ed9b2fed6d/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) **failed (sendZclFrameToEndpointInternal error))**
Zigbee2MQTT:debug 2022-11-02 19:37:06: Active device 'rele1' was last seen '50.61' minutes ago.
Zigbee2MQTT:debug 2022-11-02 19:39:55: Saving state to file /config/zigbee2mqtt/state.json

glinkin55, 2022-11-02:
the same

HelpITru, 2022-11-02:
[image]

oystermon, 2022-11-03:
Hi I think I have the same issue but not 100% sure:

I recently updated to 1.28.1 and after a period of time (almost a day?), suddenly all my devices were not responding in HA. After restarting HA/Z2M it worked briefly before breaking again. The Z2M logs look fine (devices were connected and responding) but it looked like the link from MQTT to HA was broken.

I have just rolled back to 1.28.0 and it seems to be working again.

closed 2022-12-11 by github-actions[bot]