Koenkk/zigbee2mqtt #14783: OWON PIR313-E partial communication

Uhmoja, 2022-11-03:
### What happened?

Hi !
I'm trying to use the OWON PIR313-E sensor.
The sensor is well recognized and noted as supported, but the problem is that I don't have all the data normally available.

The sensor transmits :

Battery status.
The quality of the signal.
The presence detection.
The detection of manipulation.
However, it does not transmit (null value):

The temperature measurement.
Humidity measurement.
Luminosity measurement.
Here is an example of a communication received:
_
nov. 02 14:05:13 raspberrypi npm[3251]: Zigbee2MQTT:info 2022-11-02 14:05:13: MQTT publish: topic 'zigbee2mqtt/Multi sensor OWON', payload '{"battery_low":false,"humidity":null,"illuminance":null,"illuminance_lux":null,"linkquality":255,"occupancy":true,"tamper":true,"temperature":null}'
_

Can I do something to resolve it ?

Thanks.

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDONGLE-E

danieledwardgeorgehitchcock, 2022-11-03:
It looks like everything that is needed for temperature / humidity and luminosity is in the converter.

How long have you waited with the sensor connected to Z2M? Sometimes, it may take a sensor a while to transmit the values..

Uhmoja, 2022-11-03:
Thank you for your help.
I have had the sensor on for several days without receiving any data.
A little less than two weeks.

Here is the database.db concerning the sensor.
`{
    "id": 6,
    "type": "EndDevice",
    "ieeeAddr": "0x3c6a2cfffed1b1ad",
    "nwkAddr": 46059,
    "manufId": 4412,
    "manufName": "OWON",
    "powerSource": "Battery",
    "modelId": "PIR313",
    "epList": [
        1,
        2,
        3
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "inClusterList": [
                0,
                1,
                3,
                1280
            ],
            "outClusterList": [
                3
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "PIR313",
                        "manufacturerName": "OWON",
                        "powerSource": 3,
                        "zclVersion": 4,
                        "appVersion": 10,
                        "stackVersion": 0,
                        "hwVersion": 8,
                        "dateCode": "20211221"
                    }
                },
                "ssIasZone": {
                    "attributes": {
                        "iasCieAddr": "0xe0798dfffe719824",
                        "zoneState": 1
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "2": {
            "profId": 260,
            "epId": 2,
            "inClusterList": [
                0,
                3,
                1024
            ],
            "outClusterList": [
                3
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "stackVersion": 0
                    }
                }
            },
            "binds": [
                {
                    "cluster": 1026,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0xe0798dfffe719824",
                    "endpointID": 1
                },
                {
                    "cluster": 1029,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0xe0798dfffe719824",
                    "endpointID": 1
                }
            ],
            "configuredReportings": [],
            "meta": {}
        },
        "3": {
            "profId": 260,
            "epId": 3,
            "inClusterList": [
                0,
                3,
                1026,
                1029
            ],
            "outClusterList": [
                3
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "stackVersion": 0
                    }
                }
            },
            "binds": [
                {
                    "cluster": 1024,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0xe0798dfffe719824",
                    "endpointID": 1
                }
            ],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 10,
    "stackVersion": 0,
    "hwVersion": 8,
    "dateCode": "20211221",
    "zclVersion": 4,
    "interviewCompleted": true,
    "meta": {
        "configured": 1
    },
    "lastSeen": 1667470098587,
    "defaultSendRequestWhen": "immediate"
}`

danieledwardgeorgehitchcock, 2022-11-03:
Ahh! I think I have spotted something!

Looks like (from your database.db entry), the reporting endpoints for temperature and humidity, and luminance are the wrong way round.

Could remove your devices from Z2M, add this [external converter](https://gist.github.com/danieledwardgeorgehitchcock/7a2b0dc696b12a88907cb77911e7cfdd), restart Z2M and re-add your devices?

Uhmoja, 2022-11-03:
Ah ! I hope it will work !
I have done the steps you told me, but now Z2M don't want to start, it return this error : 

nov. 03 16:55:34 raspberrypi systemd[1]: Started zigbee2mqtt.
nov. 03 16:55:37 raspberrypi npm[2583]: > zigbee2mqtt@1.28.1 start
nov. 03 16:55:37 raspberrypi npm[2583]: > node index.js
nov. 03 16:55:45 raspberrypi npm[2595]: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:91
nov. 03 16:55:45 raspberrypi npm[2595]:         if (converter.options) {
nov. 03 16:55:45 raspberrypi npm[2595]:                       ^
nov. 03 16:55:45 raspberrypi npm[2595]: TypeError: Cannot read properties of undefined (reading 'options')
nov. 03 16:55:45 raspberrypi npm[2595]:     at Object.addDefinition [as addDeviceDefinition] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:91:23)
nov. 03 16:55:45 raspberrypi npm[2595]:     at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:15:17)
nov. 03 16:55:45 raspberrypi npm[2595]:     at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
nov. 03 16:55:45 raspberrypi npm[2595]:     at start (/opt/zigbee2mqtt/index.js:108:18)
nov. 03 16:55:45 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
nov. 03 16:55:45 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

danieledwardgeorgehitchcock, 2022-11-03:
I've just updated the external converter in the Gist.

Could you re-download and try again?

Uhmoja, 2022-11-03:
Haha ! It's working !
I receive all data that I need, thanks a lot for your help and your time !
I suppose it will help for other too ! :)

closed 2022-11-03 by Uhmoja

Krinou3000, 2022-11-11:
Hi Uhmoja,
Can you share me the owon_pir313_e.js ? Please
I don't find it, the link is dead.

Thanks you !

Uhmoja, 2022-11-14:
Hi @Krinou3000,
Sorry for the delay, I share it to you tonight ! :)

Uhmoja, 2022-11-14:
Good evening,
I'm a bit confused... I had planned to deliver the solution tonight, but it turns out that my solution is running on a raspberry pi, and that it can't start anymore... (error "end Kernel panic").
I don't know yet if my raspberry is dead or if it's fixable, but for the moment I don't have access to the file you are interested in.
I'll try to keep you informed as soon as possible.

Maybe try to update Zigbee2MQTT to the dev version, the patch fixing the problem has obviously been freshly accepted:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4882/files