Koenkk/zigbee2mqtt #14786: [New device support]:

MeisterAdebar, 2022-11-03:
### Link

https://www.obi.de/eglo-connect-leuchten/eglo-fernbedienung-zigbee-remote-2-0-schwarz-1-5-v/p/9358508?wt_mc=gs.pla.Wohnen.LampenLeuchten.SmartHomeLampen&wt_cc1=661774050&wt_cc2=9358508&wt_cc4=c&wt_cc5=138830890790&wt_cc8=pla-online&wt_cc9=41959169468&storeId=&gclid=CjwKCAjwzY2bBhB6EiwAPpUpZifVf78rBsn2vkeHoPK-8I7WIxqcgVRGgguK0LRT5_EZmZqKIhkHtxoCtqUQAvD_BwE&et_uk=b73c649be45745a48f914be2ef7f1500ps://www.obi.de/eglo-connect-leuchten/eglo-fernbedienung-zigbee-remote-2-0-schwarz-1-5-v/p/9358508?wt_mc=gs.pla.Wohnen.LampenLeuchten.SmartHomeLampen&wt_cc1=661774050&wt_cc2=9358508&wt_cc4=c&wt_cc5=138830890790&wt_cc8=pla-online&wt_cc9=41959169468&storeId=&gclid=CjwKCAjwzY2bBhB6EiwAPpUpZifVf78rBsn2vkeHoPK-8I7WIxqcgVRGgguK0LRT5_EZmZqKIhkHtxoCtqUQAvD_BwE&et_uk=b73c649be45745a48f914be2ef7f1500

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0xa4c1386c5a0939dc","nwkAddr":58315,"manufId":4417,"manufName":"AwoX","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,3,4,4096],"outClusterList":[0,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1667474038591,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,

I bought this remote control yesterday, an employee was kind enough to look for the remote control out of a large box that was still on a pallet, today the supplier comes and puts the goods on the shelf.

in the configuration.yaml I added:
`external_converters:
  - WSDCGQ01LM.js
`

I'm only failing at the on/off buttons.

I've already tried other converters, but I think fz.command_on/fz.command_on is the right choice.
`

debug 2022-11-03 12:38:24Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32778
debug 2022-11-03 12:38:24No converter available for '33951/33948' with cluster 'genOnOff' and type 'commandOff' and data '{}'
debug 2022-11-03 12:38:25Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32778
debug 2022-11-03 12:38:25No converter available for '33951/33948' with cluster 'genOnOff' and type 'commandOn' and data '{}'
`

Can anyone explain what I haven't implemented correctly yet ?

### External converter

```shell
const [ konst]  fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TLSR82xx'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '99099', // Vendor model number, look on the device for a model number
    vendor: 'Eglo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Eglo Remote Connect.Z', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fz.command_on],
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity()],
};

module.exports = definition;
```

MeisterAdebar, 2022-11-03:
I think the problem is the same modelId (TLSR82xx), like this one in awox.js ?

```
{
        zigbeeModel: ['TLSR82xx'],
        model: '33951/33948',
        vendor: 'AwoX',
        description: 'LED white',
        extend: extend.light_onoff_brightness(),
        whiteLabel: [{vendor: 'EGLO', model: '12229'}],
    }
```

That's probably the reason why I don't get the ...is NOT supported... .

The device will appear as this
 recognized.

Koenkk, 2022-11-03:
What z2m version are you using?

MeisterAdebar, 2022-11-03:
1.28.1

Koenkk, 2022-11-03:
External converters override the z2m ones. Make sure you add it correctly to the z2m `configuration.yaml`. There is one syntax error btw: `const [ konst]  fz = require('zigbee-herdsman-converters/converters/fromZigbee');` -> `const  fz = require('zigbee-herdsman-converters/converters/fromZigbee');`

MeisterAdebar, 2022-11-03:
Ok, syntax error fixed, now when I press a key I get:

```
debug 2022-11-03 18:27:27Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32778
warning 2022-11-03 18:27:27Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
warning 2022-11-03 18:27:27Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
  user: moi
  password: bliblablub
serial:
  adapter: deconz
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2439900-if00
advanced:
  network_key:
    - 179
    - 30
    - 86
    - 35
    - 163
    - 160
    - 181
    - 10
    - 34
    - 190
    - 238
    - 93
    - 66
    - 249
    - 179
    - 131
  log_level: debug
  external_converters:
    - TLSR82xx.js
frontend:
  port: 8080
  host: 192.168.x.x
  auth_token: bliblablub
devices:
  '0x00158d000360ba24':
    friendly_name: '0x00158d000360ba24'
  '0x00158d0003274a6c':
    friendly_name: '0x00158d0003274a6c'
  '0x00158d000340eac3':
    friendly_name: '0x00158d000340eac3'
  '0x00158d0003609724':
    friendly_name: '0x00158d0003609724'
  '0x00158d0003602b8a':
    friendly_name: '0x00158d0003602b8a'
  '0x00158d000302cc1e':
    friendly_name: '0x00158d000302cc1e'
  '0x00158d0003278378':
    friendly_name: '0x00158d0003278378'
  '0x00158d0002fdc5d7':
    friendly_name: '0x00158d0002fdc5d7'
  '0x00158d00028aedf7':
    friendly_name: '0x00158d00028aedf7'
  '0x00158d0003e5ee0c':
    friendly_name: '0x00158d0003e5ee0c'
  '0x00158d0002fd8bb4':
    friendly_name: '0x00158d0002fd8bb4'
  '0x00158d00047d3f22':
    friendly_name: '0x00158d00047d3f22'
  '0x00158d0004038f6b':
    friendly_name: '0x00158d0004038f6b'
  '0x00158d000411c41c':
    friendly_name: '0x00158d000411c41c'
  '0x00158d00031c22fa':
    friendly_name: '0x00158d00031c22fa'
  '0x2c1165fffedecf76':
    friendly_name: '0x2c1165fffedecf76'
  '0x00158d000411b050':
    friendly_name: '0x00158d000411b050'
  '0x00158d00032c6d54':
    friendly_name: '0x00158d00032c6d54'
  '0x04cd15fffea4c6ad':
    friendly_name: '0x04cd15fffea4c6ad'
  '0xa4c1386c5a0939dc':
    friendly_name: '0xa4c1386c5a0939dc'
\n\nadvanced:\n    network_key: GENERATE
blocklist: []
```

```
-rw-r--r--  1 pi pi  1926 Nov  3 17:52 configuration.yaml
-rw-r--r--  1 pi pi 19117 Nov  3 17:52 database.db
drwxr-xr-x 12 pi pi  4096 Nov  3 15:15 log
-rw-r--r--  1 pi pi    19 Nov  3 10:34 secret.yaml
-rw-r--r--  1 pi pi  3785 Nov  3 17:52 state.json
-rw-r--r--  1 pi pi  1309 Nov  3 14:39 TLSR82xx.js
```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

```
const definition = {
    zigbeeModel: ['TLSR82xx'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '99099', // Vendor model number, look on the device for a model number
    vendor: 'Eglo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Eglo Remote Connect.Z', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fzLocal.colors, fzLocal.refresh, fzLocal.refreshColored, fz.command_off,
            fz.command_step, fz.command_move, fz.command_stop, fz.command_recall, fz.command_step_color_temperature],
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity()],
};

module.exports = definition;
```

Koenkk, 2022-11-03:
`external_converters` should not be under `advanced` (move it one tab to the left)

MeisterAdebar, 2022-11-03:
Thank you so much, everything should be the same then  

```
model: '33952',
        vendor: 'AwoX',
        description: 'Remote controller',
        fromZigbee: [fz.command_on, fzLocal.colors, fzLocal.refresh, fzLocal.refreshColored, fz.command_off,
            fz.command_step, fz.command_move, fz.command_stop, fz.command_recall, fz.command_step_color_temperature],
        toZigbee: [],
        exposes: [e.action(['on', 'off', 'red', 'refresh', 'refresh_colored', 'blue', 'yellow',
            'green', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down', 'brightness_stop',
            'recall_1', 'color_temperature_step_up', 'color_temperature_step_down'])],
    },
```
I'll deal with that later, I mean, only the two favorite keys would have to be added.

Maybe the instructions should be adjusted:

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt

Koenkk, 2022-11-03:
The instructions are correct, `external_converters` is not indented:

[image]

MeisterAdebar, 2022-11-04:
I've now found the following, I'm dealing with it for the first time, and so far, even after a very long test, I haven't been able to determine exactly how to pair it:

I copied this template 1:1:

```
        model: '33952',
        vendor: 'AwoX',
        description: 'Remote controller',
```

with externals converters:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const awoxRemoteHelper = {
    convertToColorName: (buffer) => {
        const commonForColors = buffer[0] === 17 && buffer[2] === 48 && buffer[3] === 0 && buffer[5] === 8 && buffer[6] === 0;

        if (commonForColors && buffer[4] === 255) {
            return 'red';
        } else if (commonForColors && buffer[4] === 42) {
            return 'yellow';
        } else if (commonForColors && buffer[4] === 85) {
            return 'green';
        } else if (commonForColors && buffer[4] === 170) {
            return 'blue';
        }
        return null;
    },
    isRefresh: (buffer) => {
        return buffer[0] === 17 && buffer[2] === 16 && buffer[3] === 1 && buffer[4] === 1;
    },
};

const fzLocal = {
    colors: {
        cluster: 'lightingColorCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const color = awoxRemoteHelper.convertToColorName(msg.data);
            if (color != null) {
                return {
                    action: color,
                };
            }
        },
    },
    refreshColored: {
        cluster: 'lightingColorCtrl',
        type: ['commandMoveHue'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.movemode === 1 && msg.data.rate === 12) {
                return {
                    action: 'refresh_colored',
                };
            }
        },
    },
    refresh: {
        cluster: 'genLevelCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            if (awoxRemoteHelper.isRefresh(msg.data)) {
                return {
                    action: 'refresh',
                };
            }
        },
    },
};

const definition = {
        fingerprint: [
            {
                type: 'EndDevice', manufacturerName: 'AwoX', modelID: 'TLSR82xx', powerSource: 'Battery', endpoints: [
                    {ID: 1, profileID: 260, deviceID: 2048, inputClusters: [0, 3, 4, 4096], outputClusters: [0,3,4,5,6,8,768,4096]},
                ],
            },
        ],
    zigbeeModel: ['TLSR82xx'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '99099', // Vendor model number, look on the device for a model number
    vendor: 'Eglo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Eglo Remote Connect.Z', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fz.command_off, fzLocal.colors, fzLocal.refresh, fzLocal.refreshColored, fz.command_step, fz.command_move, fz.command_stop, fz.command_recall, fz.command_step_color_temperature],
    toZigbee: [],
    exposes: [e.action(['on', 'off', 'red', 'refresh', 'refresh_colored', 'blue', 'yellow',
            'green', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down', 'brightness_stop',
            'recall_1', 'color_temperature_step_up', 'color_temperature_step_down'])],
};

module.exports = definition;
```
Everything works, except for the keys 1,2,3, nothing is recognized there?
How can that be, they only send via Bluetooth?

If I remove external_converters from the configuration.yaml, the remote control is displayed as:

 ```
zigbeeModel: ['TLSR82xx'],
        model: '33951/33948',
        vendor: 'AwoX',
        description: 'LED white',

```
Can you do something with this information and see where the problem lies?
If the buttons 1,2,3 really don't send anything, the remote control should be called

```
        model: '33952',
        vendor: 'AwoX',
        description: 'Remote controller',
```

be recognized in the future and simply add 99099 to model

MeisterAdebar, 2022-11-05:
The two remotes are different.

I'm ready now:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const awoxRemoteHelper = {
    convertToColorName: (buffer) => {
        const commonForColors = buffer[0] === 17 && buffer[2] === 48 && buffer[3] === 0 && buffer[5] === 8 && buffer[6] === 0;

        if (commonForColors && buffer[4] === 255) {
            return 'red';
        } else if (commonForColors && buffer[4] === 85) {
            return 'green';
        } else if (commonForColors && buffer[4] === 170) {
            return 'blue';
        }
        return null;
    },
    isRefresh: (buffer) => {
        return buffer[0] === 17 && buffer[2] === 16 && buffer[3] === 1 && buffer[4] === 1;
    },
};

const fzLocal = {
    colors: {
        cluster: 'lightingColorCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const color = awoxRemoteHelper.convertToColorName(msg.data);
            if (color != null) {
                return {
                    action: color,
                };
            }
        },
    },
    refreshColored: {
        cluster: 'lightingColorCtrl',
        type: ['commandMoveHue'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.movemode === 1 && msg.data.rate === 12) {
                return {
                    action: 'refresh_colored',
                };
            }
        },
    },
    refresh: {
        cluster: 'genLevelCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            if (awoxRemoteHelper.isRefresh(msg.data)) {
                return {
                    action: 'refresh',
                };
            }
        },
    },
};

const definition = {
        fingerprint: [
            {
                type: 'EndDevice', manufacturerName: 'AwoX', modelID: 'TLSR82xx', powerSource: 'Battery', endpoints: [
                    {ID: 1, profileID: 260, deviceID: 2048, inputClusters: [0, 3, 4, 4096], outputClusters: [0,3,4,5,6,8,768,4096]},
                ],
            },
        ],
    zigbeeModel: ['TLSR82xx'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '99099', // Vendor model number, look on the device for a model number
    vendor: 'Eglo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Eglo Remote Connect.Z', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fz.command_off,
                fzLocal.colors, fzLocal.refresh,
                fz.command_step,
                fz.command_recall,
                fz.command_move_to_color_temp, fz.command_step_color_temperature],
    toZigbee: [],
    exposes: [e.action(['on', 'off',
            'red', 'green', 'blue', 'refresh',
            'brightness_step_up', 'brightness_step_down',
            'recall_1','recall_2',
            'color_temperature_step_up', 'color_temperature_step_down', 'color_temperature_move'])],
};

module.exports = definition;
```

What's missing are the long button presses, I haven't quite understood that yet.

```
debug 2022-11-05 13:38:52Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":4}' from endpoint 1 with groupID 32779
debug 2022-11-05 13:38:52No converter available for '99099' with cluster 'genLevelCtrl' and type 'commandMoveToLevelWithOnOff' and data '{"level":254,"transtime":4}'
debug 2022-11-05 13:38:58Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":1,"transtime":4}' from endpoint 1 with groupID 32779
debug 2022-11-05 13:38:58No converter available for '99099' with cluster 'genLevelCtrl' and type 'commandMoveToLevelWithOnOff' and data '{"level":1,"transtime":4}'
```

Koenkk, 2022-11-06:
Add `fz.command_move_to_level` to `fromZigbee`

MeisterAdebar, 2022-11-06:
Many Thanks.

now all buttons work, there are no more "No converter available for ..." messages.

If something needs to be added now, I just don't know what and you would have to point it out to me.

With point 4 of the instructions I don't know what I have to do at first, I've never done it before, I have to deal with it, I would prefer you to check it in.

I have one more question, because I have no idea about it (of javascript I don't know at all).

Does it have to be that these keys are always followed by "action":"on"?

```
fzLocal.colors, fzLocal.refresh,
fz.command_move_to_color_temp, fz.command_step_color_temperature
```

Koenkk, 2022-11-06:
> Does it have to be that these keys are always followed by "action":"on"?

Can you explain this more? I don't know what you mean.

MeisterAdebar, 2022-11-06:
```
debug 2022-11-06 14:47:54Received Zigbee message from '0xa4c1386c5a0939dc', type 'raw', cluster 'lightingColorCtrl', data '{"data":[17,27,48,0,85,8,0],"type":"Buffer"}' from endpoint 1 with groupID 32779
info 2022-11-06 14:47:54MQTT publish: topic 'zigbee2mqtt/0xa4c1386c5a0939dc', payload '{"action":"green","linkquality":95}'
debug 2022-11-06 14:47:54Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32779
info 2022-11-06 14:47:54MQTT publish: topic 'zigbee2mqtt/0xa4c1386c5a0939dc', payload '{"action":"on","action_group":32779,"linkquality":95}'
debug 2022-11-06 14:48:07Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":0,"minimum":0,"stepmode":1,"stepsize":32,"transtime":4}' from endpoint 1 with groupID 32779
info 2022-11-06 14:48:07MQTT publish: topic 'zigbee2mqtt/0xa4c1386c5a0939dc', payload '{"action":"color_temperature_step_up","action_group":32779,"action_step_size":32,"action_transition_time":0.04,"linkquality":95}'
debug 2022-11-06 14:48:07Received Zigbee message from '0xa4c1386c5a0939dc', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32779
info 2022-11-06 14:48:07MQTT publish: topic 'zigbee2mqtt/0xa4c1386c5a0939dc', payload '{"action":"on","action_group":32779,"linkquality":103}'
```

MeisterAdebar, 2022-11-06:
and one more question, I only bought the remote control because of the three channels (1,2,3). Nothing arrives, why will that be the case?

Koenkk, 2022-11-06:
> Does it have to be that these keys are always followed by "action":"on"?

This matches with what the device sends so yes.

> I only bought the remote control because of the three channels (1,2,3). Nothing arrives, why will that be the case?

Are these separate buttons?

MeisterAdebar, 2022-11-06:
> Are these separate buttons?

look at the link above

Koenkk, 2022-11-06:
Is anything logged to the herdsman debug log when pressing these buttons?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

MeisterAdebar, 2022-11-06:
nothing

MeisterAdebar, 2022-11-06:
funny, if I press the buttons (1,2,3) for 10 seconds, I can pair and control them with one of my Müller Tint GU10.

With
`DEBUG=zigbee-herdsman* npm start`
I still see nothing

MeisterAdebar, 2022-11-07:
> try resetting the bulb

Thank you.

to all who read here.

(In the case of lights paired directly with the remote control, the switching status is not transmitted to z2m.)

To unpair (reset) the remote control with the lamp, you have to switch the lamp on and off 5 times.

It is not necessary to repair the light in z2m.

> To figure out how this work it needs to be reverse engineered when used via the original hub (sniff).

That's why I won't buy a gateway from Eglo Connect.Z.

Stegi02345, 2023-04-22:
will this remote be in the next version of z2m that i dont have to use un external converter?

m1k3f15h, 2023-07-11:
Hello,

How is the Status here?
Did the remote work?

Mike :)

GeoffNT, 2023-07-14:
I've just added one of these to Z2M. Initially it wasn't recognised as the firmware on mine was giving the manufacturer as AwoX rather than EGLO, but after editing the herdsman file to AwoX it paired successfully.  

The 1,2 & 3 buttons send nothing. I reckon they just switch between scene modes within the remote itself so it works on the 3 different scenes? I couldn't see anything in the output that identified which scene mode it was currently in though, so who knows. 

There is no battery reporting so far, but the remote works well for the other buttons. It's also got a flat front so it was easy to make a custom label to stick over it. I'm using it as a media player remote.

errrrata, 2023-09-18:
If anybody managed to get this remote working, can you please share the pairing method? I have tried 'off' + 'blue' and z2m doesn't even discover the remote

GeoffNT, 2023-09-19:
IIRC To pair it's press and hold the Power On button (top right) and the blue rather than the off button and blue.

You could also try press and hold the Power On and the "circular back green/red/blue button" in the centre which resets it.

Pirupuku, 2024-01-21:
Pairing the remot → Hold Top Right (On - button) and blue circle button! Didn't work for me either at first, but then I stood right next to my Zigbee dongle and it recognized it.

@Koenkk I have the newest Z2M version and all. The remote works so far, but a lot of buttons don't! E. g. when I hit the color buttons it is recognized as if I would press the On-button. Group buttons (1, 2, 3) don't work at all. Brightness up and down works and Color Temp up and down also works.
Any news on that?!
@MeisterAdebar Could you explain how to do it with external converters?!

MeisterAdebar, 2024-01-21:
Hello,

> E. g. when I hit the color buttons it is recognized as if I would press the On-button

Something wasn't right for me with the latest version, so I'm still using my converter file.

With the Concerter file I always get an "on" but also the color:
```
zigbee2mqtt/0xa4c1386c5a0939dc {"action":"blue","linkquality":239}
zigbee2mqtt/0xa4c1386c5a0939dc {"action":"on","action_group":32779,"linkquality":239}
```

>  Group buttons (1, 2, 3) don't work at all. 

(https://github.com/Koenkk/zigbee2mqtt/issues/14786#issuecomment-1305871764)

> @MeisterAdebar Could you explain how to do it with external converters?!

I created the following file  in the data folder and specified it in the UI under Settings/External converters:

```
const fz =  require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const awoxRemoteHelper = {
    convertToColorName: (buffer) => {
        const commonForColors = buffer[0] === 17 && buffer[2] === 48 && buffer[3] === 0 && buffer[5] === 8 && buffer[6] === 0;

        if (commonForColors && buffer[4] === 255) {
            return 'red';
        } else if (commonForColors && buffer[4] === 85) {
            return 'green';
        } else if (commonForColors && buffer[4] === 170) {
            return 'blue';
        }
        return null;
    },
    isRefresh: (buffer) => {
        return buffer[0] === 17 && buffer[2] === 16 && buffer[3] === 1 && buffer[4] === 1;
    },
};

const fzLocal = {
    colors: {
        cluster: 'lightingColorCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const color = awoxRemoteHelper.convertToColorName(msg.data);
            if (color != null) {
                return {
                    action: color,
                };
            }
        },
    },
    refreshColored: {
        cluster: 'lightingColorCtrl',
        type: ['commandMoveHue'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.movemode === 1 && msg.data.rate === 12) {
                return {
                    action: 'refresh_colored',
                };
            }
        },
    },
    refresh: {
        cluster: 'genLevelCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            if (awoxRemoteHelper.isRefresh(msg.data)) {
                return {
                    action: 'refresh',
                };
            }
        },
    },
};

const definition = {
        fingerprint: [
            {
                type: 'EndDevice', manufacturerName: 'AwoX', modelID: 'TLSR82xx', powerSource: 'Battery', endpoints: [
                    {ID: 1, profileID: 260, deviceID: 2048, inputClusters: [0, 3, 4, 4096], outputClusters: [0,3,4,5,6,8,768,4096]},
                ],
            },
        ],
    zigbeeModel: ['TLSR82xx'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '99099', // Vendor model number, look on the device for a model number
    vendor: 'Eglo', // Vendor of the device (only used for documentation and startup logging)
    description: 'Eglo Remote Connect.Z', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fz.command_off,
                fzLocal.colors, fzLocal.refresh,
                fz.command_step, fz.command_move_to_level,
                fz.command_recall,
                fz.command_move_to_color_temp, fz.command_step_color_temperature],
    toZigbee: [],
    exposes: [e.action(['on', 'off',
            'red', 'green', 'blue', 'refresh',
            'brightness_step_up', 'brightness_step_down',
            'recall_1','recall_2',
            'color_temperature_step_up', 'color_temperature_step_down', 'color_temperature_move'])],
};

module.exports = definition;
```

Pirupuku, 2024-01-22:
> I created the following file in the data folder and specified it in the UI under Settings/External converters:

I am quite new to HA... and more to Z2M and custom stuff

Creating a file means in the Zigbee2MQTT subfolder of my HA and the file has to be a .js file, as far as I understood.
Would it be possible to give me a as-short-as-possible step-by-step guide?

Pirupuku, 2024-01-23:
```
zigbee2mqtt/0xa4c1386c5a0939dc {"action":"blue","linkquality":239}
zigbee2mqtt/0xa4c1386c5a0939dc {"action":"on","action_group":32779,"linkquality":239}
```
Oh on that. As the behaviour seems to be the same....you just read that in the logs right? I enabled the log_level: debug in the `configuration.yaml` in my `Zigbee2MQTT` folder. Is that all or can I log that in a more direct window?!

Suschibomber, 2024-01-28:
@MeisterAdebar Ich habe mir auch gestern eine Eglo Zigbee 2.0 gestern im Obi gekauft. Habe sie dann in der App erstmal auf die neuste Firmware gebracht und sie dann in Zigbee2mqtt angelernt. Mein Poblem ist jetzt, dass ich außer der Link Quality nichts sehe. Es werden keine Entitäten in HomeAssistant für die einzelnen Buttons erzeugt. Wie bekomme ich es jetzt hin das ich die Buttons in HomeAssistant bekomme? Könntest du mir helfen? :) VG

MeisterAdebar, 2024-01-28:
Hallo,

ich hab mich noch nie mit irgendeiner App von Eglo beschäftigt, wegen der Update-Möglichkeit werde ich mich aber mal dazu einlesen.
Hab keine Erfahrung mit HA, mein trautes Heim ist FHEM.
Hast du das Problem mit meiner hier gezeigten Converter-Datei oder mit der Version die mit z2m kommt ?
Ich hatte die ausprobiert aber irgendwas passte nicht und hatte keine Lust mich weiter mit zu beschäftigen und blieb bisher bei meiner "Interpretation".

Gruß

Thomas

Suschibomber, 2024-01-29:
@MeisterAdebar 
Also ich habe halt auf der Seite von Zigbee2Mqtt nachgeschaut und im Internet gelesen , dass man das Gerät zunächst auf die neuste Firmware per hauseigener App bringen soll und dann in Zigbee2Mqtt anlernen soll.

Zum einen weiß ich leider nicht wie ich eine von dir angesprochene converter Datei einbinden soll oder kann und ja also das pure anlernen um dann die Buttons über mein homeassistent steuern zu können klappt nicht.

Also mit anderen Worten wäre es echt Klasse, wenn du mir sagen könntest was ich machen muss um deine Variante zu verwenden. Bitte möglichst Anfänger freundlich erklären 🙈

Suschibomber, 2024-02-01:
@MeisterAdebar 
Hast du meine Antwort gesehen? :)

Pirupuku, 2024-02-01:
Also ich musste die von @MeisterAdebar gezeigte Converter-Datei etwas abändern und die Fernbedienung funktioniert. Mit Außnahme allerdings der Gruppentasten, also 1, 2 und 3.

Ein weitere Problem bei mir ist aber eher die Leuchte! Meine Deckenleuchte Eglo Rovito.Z wird als AwoX 33943/33944/33946 erkannt. Das ist glaube ich nicht weiter schlimm, da auch diese eine RGB Hintergrundbeleuchtung hat und ich sehe diese auch als Attribute in den Entwicklerwerkzeugen. Jedoch kann ich bei meinen Automationen nicht auf diese zugreifen. Das heißt meine Fernbedienung kann die Lichter nur an-/ausschalten und die Helligkeit verändern. Farbtemperatur und RGB geht nicht.

Gibt es für diese Deckeleuchte auch eine External-Converter Datei?

Suschibomber, 2024-02-01:
> Also ich musste die von @MeisterAdebar gezeigte Converter-Datei etwas abändern und die Fernbedienung funktioniert. Mit Außnahme allerdings der Gruppentasten, also 1, 2 und 3.
>
> Ein weitere Problem [...]

OK und wie bekomme die Buttons dann in Homeassistent? Möchte damit keine Eglo Lampen nutzen sondern die Buttons auf der Fernbedienung einfach frei mit irgendwelchen Aktionen programmieren.

Und das mit der Converter Datei verstehe ich auch nicht. Was muss man damit machen? Sorry für die dummen Fragen.

Pirupuku, 2024-02-01:
Durch das drücken der Buttons, dadurch werden die erkannt!

Die Konverter-Datei genauso anlegen wie geschrieben, dann sollte er diese Datei nehmen anstatt der "hinterlegten". Ach und du musst die dann nochmal neu pairen.

Suschibomber, 2024-02-01:
@Pirupuku 
Vielen Dank für deine Anleitung. Wann sollte ich denn den Z2M Dienst stoppen. Bevor ich den Eintrag in die configuration.yaml mache?
Und wenn ich das jetzt richtig verstanden habe bei @MeisterAdebar funktionierte in seiner Converter Datei alles bis auf "Long-Button-Press" Funktionen korrekt?
Und warum funktionieren bei dir die Gruppen-Tasten dann nicht? Beim MeisterAdebar funktionierten die laut ihm aber doch oder sehe ich das falsch? Muss ich dann auch noch was abändern in der Converter Datei?

Suschibomber, 2024-02-01:
> external_converters:

Also habe jetzt alles so gemacht wie du gesagt hast aber jedes mal wenn ich eine Taste drücke, kommt unter Z2M unter dem Gerät -> Details folgendes:

Action
Ausgelöstes Ereignis (z.B. ein Aktivieren eines Tasters)
String ohne Inhalt("")

Unter dem Gerät in MQTT erscheinen auch keine Buttons oder Entitäten. Wie kann ich denn dann jetzt die einzelnen Buttons zuweisen? 

Und die Gruppen-Tasten so wie die rot,grün und blau funktionieren bei mir auch nicht. Da passiert gar nichts wenn ich drauf drücke.

Pirupuku, 2024-02-02:
Hast du mal versucht eine Automatisierung zu machen?

Suschibomber, 2024-02-02:
@Pirupuku 

Nein habe ich nicht. Dafür müssten mir doch unter MQTT oder in den Entwicklerwerkzeugen Entitäten angezeigt werden, damit ich diese triggern kann oder nicht?

Pirupuku, 2024-02-02:
Ich bin gerade nur am Handy online, aber ziemlich sicher, dass ich auch keine Attribute dort habe.

Suschibomber, 2024-02-02:
@Pirupuku 

OK dann müsstest du mir vielleicht noch Mal sagen wie eine Automatisierung aussehen muss, um eine bestimmte Taste zu triggern.

Suschibomber, 2024-02-03:
[image]
Das geht zu Glück schon mal. So siehts aus. Scheint auch bei mir einiges zu fehlen :(

ustix, 2024-02-04:
You are really a life saver, i was considering throwing away this remote, but now it is working almost as intended.
Do you have any idea how to make 1,2,3 buttons work as an action like all the others? (I havent read all those German comments, so sorry if it is already explained)

MeisterAdebar, 2024-02-04:
> Do you have any idea how to make 1,2,3 buttons work as an action like all the others? (I havent read all those German comments, so sorry if it is already explained)

https://github.com/Koenkk/zigbee2mqtt/issues/14786#issuecomment-1304806489

https://github.com/Koenkk/zigbee2mqtt/issues/14786#issuecomment-1305871764

Pirupuku, 2024-02-04:
> ![eglo](https://private-user-images.githubusercontent.com/51985160/302061398-70dd8610-5b6a-43f9-9683-6713e7e6bf6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNTA0ODYsIm5iZiI6MTcwNzA1MDE4NiwicGF0aCI6Ii81MTk4NTE2MC8zMDIwNjEzOTgtNzBkZDg2MTAtNWI2YS00M2Y5LTk2ODMtNjcxM2U3ZTZiZjZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDEyMzYyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM1ZTBiODA3M2U5YmUxMzMyZjEyZGVmZmY1NTFmZGE4ZjNmMGZkY2U1MzFjMDIwYTQwMmNhMDBjNDU1YzhhODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.utDW4QjrEm1TSXnaAzYYGjQjx92jnVG_VwxTw__CSWA) Das geht zu Glück schon mal. So siehts aus. Scheint auch bei mir einiges zu fehlen :(

Für die anderen buttons red, green und blue musst du debuggen und nach dem payload ausschau halten. Ich musste die Werte aus MeisterAdebars external converter verändern. Bei mir waren alle um 1 kleiner für die jeweilige Farbe

Pirupuku, 2024-02-04:
> You are really a life saver, i was considering throwing away this remote, but now it is working almost as intended. Do you have any idea how to make 1,2,3 buttons work as an action like all the others? (I havent read all those German comments, so sorry if it is already explained)

Yeah sorry for switching to german. Well if you check the screenshot above, yours should be somehow the same. What's missing there and was the same for me are the color bottons (red, green, blue).
To get those I had to adapt the values from the external converter file provided by @MeisterAdebar to my needs! Basically all my values for each color were 1 smaller than those in the external converter

ustix, 2024-02-04:
> Yeah sorry for switching to german. Well if you check the screenshot above, yours should be somehow the same. What's missing there and was the same for me are the color bottons (red, green, blue). [...]

I have the color buttons working (red, green, blue) i would like to use the group buttons on the bottom of the remote labeled with 1 and 2 and 3. Are they any different then the other buttons or why arent they implented?

Pirupuku, 2024-02-04:
I don't know in which sense they are different but they definitely seems so. They are not recognized at all.

msaphoto06, 2024-02-05:
Good evening,
Sorry for my English, I'm French.
The remote control works quite well and I understood the 3 groups with the keys 1,2,3 -> Group 32778 32779 32780.
1- how to add the blue red and green keys
2- Is it possible to change the group number when purchasing a second remote control?

THANKS

ustix, 2024-02-05:
> Good evening, Sorry for my English, I'm French. The remote control works quite well and I understood the 3 groups with the keys 1,2,3 -> Group 32778 32779 32780. 1- how to add the blue red and green keys 2- Is it possible to change the group number when purchasing a second remote control?
> 
> THANKS
Do you mean you can use the 1,2,3 buttons somehow? The color buttons are working with the external converter as intended.

msaphoto06, 2024-02-05:
YES! The buttons are associated with the groups mentioned.
Just create a group with id 32778 for button 1
By pressing 1, only the group reacts to the command.
ditto 2 and 3..

msaphoto06, 2024-02-05:
In z2m, create a group with id code 32778 (1) 32779 (2) 32780 (3)
in each group you add the devices (one or more)
when pressing 1, the remote control sends the command to group 32778, same as 2 and 3...
In addition, I discovered that the remote control memorized the state of each group by remaining pressed on heart 1 and heart 2...
which creates 2 stored scenes for each group.

how to proceed in debugging.. I'm fairly new to z2m.
I just see "on" instead of "red" or "green"

Pirupuku, 2024-02-06:
Can you show me an example with the groups and group ids? I am fairly new myself :D

For debugging:
Go to -> File Editor -> homeassistant/zigbee2mqtt/configuration.yaml and add these lines
> advanced:
>   - log_level: debug
I am pretty sure you already have "advanced" written somewhere so you only add "  - log_level:debug"
Than you can see in Zigbee2MQTT -> log -> filter through "debug" and "your device". That way you see the payload and with the example of MeisterAdebar change the values to your color codes

ustix, 2024-02-06:
> Can you show me an example with the groups and group ids? I am fairly new myself :D
>
> For debugging: Go to -> File Editor -> homeassistant/zigbee2mqtt/configuration.yaml and add these lines
>
> ``` [...]

I think thats the way he thought:
[image]
I haven't tried it, because there is no other device in my z2m now (test config) but i think that could work. I will try this week.

Pirupuku, 2024-02-06:
@ustix 
Yeah, that worked. Great! Thanks for explaining!

Only thing I need now is the color buttons! The problem is not the remote but the light bulb!

msaphoto06, 2024-02-06:
I don't use the converter, so I don't have the colors.
Can someone explain to me how to set it up to get the colors?
I will always have the groups with the converters?
Possible to change the group IDs with a converter to have several remote controls?
THANKS

msaphoto06, 2024-02-06:
Ps:
No addon Homeassistant but Z2m dockers

Pirupuku, 2024-02-06:
> > @ustix Yeah, that worked. Great! Thanks for explaining!
> > Only thing I need now is the color buttons! The problem is not the remote but the light bulb!
> 
> @Pirupuku Please, explain it further, what's wrong with what kind of light bulb? If I understand, you can use the color buttons, HA recognize them in the automation menu, and if you push them they are sent to the Action history too, but you want to use them to change the bulbs to the actual color of the button? So you need help figuring out how to automate that in HA?

Yes! In Developer Tools of HA I can see all attributes of the light bulb (e.g. brightness, color temp, rgb color)  but I don't see rgb color as an action in my automation for the light bulb. Hence I have to use a data template or mqtt as action and send the payload or something. For that I don't know how to increment the rgb color.

[21 more replies]