Koenkk/zigbee2mqtt #14790: MatSeePlus PC321-Z-TY 3-phase Clamp Power Meter

dokkomp, 2022-11-03:
### Link

https://www.aliexpress.com/item/1005002683417480.html?spm=a2g0o.store_pc_groupList.8148356.1.2a6e682eTFsesZ&pdp_npi=2%40dis%21USD%21US%20%2431.98%21US%20%2415.99%21%21%21%21%21%402100bdd716674884869692815e2f2b%2112000021722081118%21sh

### Database entry

'{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":122}],"seq":44032}

### Comments

Please add the Tuya Smart PC321-Z-TY 3-phase Clamp Power Meter

Description: Tuya Smart 3 Phase Single Phase ZigBee Energy Meter KWh Power Monitor 500A 300A 200A 120A 80A with Clamp Current Transformer

Model: PC321-Z-TY
Zigbee model 'TS0601' 
Manufacturer name '_TZE200_nslr42tt'
Vendor: MatSeePlus
ZigBee Profile: ZigBee HA1.2
Exposes: Irms, Vrms, Active Power & Energy, Reactive Power & Energy

Vendors product description:

I have the latest version now 1.28.1-1

Here is my log:
Zigbee2MQTT:debug 2022-11-02 14:47:22: Received Zigbee message from 'Energy', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":122}],"seq":44032}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn 2022-11-02 14:47:22: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_nslr42tt'
Zigbee2MQTT:warn 2022-11-02 14:47:22: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

I have added a file ( 321-z-ty.js ) into the config dir, with the content from issues 9525:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    PC321_metering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const payload = {};
            if (msg.data.hasOwnProperty('owonL1Energy')) {
                payload.energy_l1 = msg.data['owonL1Energy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL2Energy')) {
                payload.energy_l2 = msg.data['owonL2Energy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL3Energy')) {
                payload.energy_l3 = msg.data['owonL3Energy'][1] / 1000.0;
            }
            
            if (msg.data.hasOwnProperty('owonL1ReactiveEnergy')) {
                payload.reactive_energy_l1 = msg.data['owonL1ReactiveEnergy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL2ReactiveEnergy')) {
                payload.reactive_energy_l2 = msg.data['owonL2ReactiveEnergy'][1] / 1000.0;
            }
            if (msg.data.hasOwnProperty('owonL3ReactiveEnergy')) {
                payload.reactive_energy_l3 = msg.data['owonL3ReactiveEnergy'][1] / 1000.0;
            }
            
            if (msg.data.hasOwnProperty('owonL1PhasePower')) {
                payload.power_l1 = msg.data['owonL1PhasePower'];
            }
            if (msg.data.hasOwnProperty('owonL2PhasePower')) {
                payload.power_l2 = msg.data['owonL2PhasePower'];
            }
            if (msg.data.hasOwnProperty('owonL3PhasePower')) {
                payload.power_l3 = msg.data['owonL3PhasePower'];
            }
            
             if (msg.data.hasOwnProperty('owonL1PhaseReactivePower')) {
                payload.reactive_power_l1 = msg.data['owonL1PhaseReactivePower'];
            }
            if (msg.data.hasOwnProperty('owonL2PhaseReactivePower')) {
                payload.reactive_power_l2 = msg.data['owonL2PhaseReactivePower'];
            }
            if (msg.data.hasOwnProperty('owonL3PhaseReactivePower')) {
                payload.reactive_power_l3 = msg.data['owonL3PhaseReactivePower'];
            }
            
            if (msg.data.hasOwnProperty('owonL1PhaseVoltage')) {
                payload.voltage_l1 = msg.data['owonL1PhaseVoltage'] / 10.0;
            }
            if (msg.data.hasOwnProperty('owonL2PhaseVoltage')) {
                payload.voltage_l2 = msg.data['owonL2PhaseVoltage'] / 10.0;
            }
            if (msg.data.hasOwnProperty('owonL3PhaseVoltage')) {
                payload.voltage_l3 = msg.data['owonL3PhaseVoltage'] / 10.0;
            }
            
            if (msg.data.hasOwnProperty('owonL1PhaseCurrent')) {
                payload.current_l1 = msg.data['owonL1PhaseCurrent'] / 1000.0;
            }
            
            if (msg.data.hasOwnProperty('owonL2PhaseCurrent')) {
                payload.current_l2 = msg.data['owonL2PhaseCurrent'] / 1000.0;
            }
            
            if (msg.data.hasOwnProperty('owonL3PhaseCurrent')) {
                payload.current_l3 = msg.data['owonL3PhaseCurrent'] / 1000.0;
            }
            
            if (msg.data.hasOwnProperty('owonFrequency')) {
                payload.frequency = msg.data['owonFrequency'];
            }
            
            if (msg.data.hasOwnProperty('owonReactiveEnergySum')) {
                payload.reactive_energy_sum = msg.data['owonReactiveEnergySum'];
            }
            
            
            return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'TS0601', // Vendor model number, look on the device for a model number
    vendor: 'MatSeePlus', // Vendor of the device (only used for documentation and startup logging)
    description: '3-Phase Clamp Power Meter', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.metering, fzLocal.PC321_metering],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        const payload = [{
                attribute: 'owonFrequency',
                minimumReportInterval: 5,
                maximumReportInterval: 3600,
                reportableChange: 0,
            }];
            await endpoint.configureReporting('seMetering', payload);
    },
	exposes: [e.energy(),
	exposes.numeric('voltage_l1',ea.STATE).withUnit('V').withDescription('Phase 1 Voltage'),
	exposes.numeric('voltage_l2',ea.STATE).withUnit('V').withDescription('Phase 2 Voltage'),
	exposes.numeric('voltage_l3',ea.STATE).withUnit('V').withDescription('Phase 3 Voltage'),
	exposes.numeric('current_l1',ea.STATE).withUnit('A').withDescription('Phase 1 Current'),
	exposes.numeric('current_l2',ea.STATE).withUnit('A').withDescription('Phase 2 Current'),
	exposes.numeric('current_l3',ea.STATE).withUnit('A').withDescription('Phase 3 Current'),
	exposes.numeric('energy_l1',ea.STATE).withUnit('kWh').withDescription('Phase 1 Energy'),
	exposes.numeric('energy_l2',ea.STATE).withUnit('kWh').withDescription('Phase 2 Energy'),
	exposes.numeric('energy_l3',ea.STATE).withUnit('kWh').withDescription('Phase 3 Energy'),
	exposes.numeric('reactive_energy_l1',ea.STATE).withUnit('kVArh').withDescription('Phase 1 Reactive Energy'),
	exposes.numeric('reactive_energy_l2',ea.STATE).withUnit('kVArh').withDescription('Phase 2 Reactive Energy'),
	exposes.numeric('reactive_energy_l3',ea.STATE).withUnit('kVArh').withDescription('Phase 3 Reactive Energy'),
	exposes.numeric('power_l1',ea.STATE).withUnit('W').withDescription('Phase 1 Power'),
	exposes.numeric('power_l2',ea.STATE).withUnit('W').withDescription('Phase 2 Power'),
	exposes.numeric('power_l3',ea.STATE).withUnit('W').withDescription('Phase 3 Power'),
	exposes.numeric('reactive_power_l1',ea.STATE).withUnit('VAr').withDescription('Phase 1 Reactive Power'),
	exposes.numeric('reactive_power_l2',ea.STATE).withUnit('VAr').withDescription('Phase 2 Reactive Power'),
	exposes.numeric('reactive_power_l3',ea.STATE).withUnit('VAr').withDescription('Phase 3 Reactive Power'),
	exposes.numeric('frequency',ea.STATE).withUnit('Hz').withDescription('Frequency'),
	exposes.numeric('reactive_energy_sum',ea.STATE).withUnit('kVArh').withDescription('Reactive Energy Sum'),
	],
	
};

module.exports = definition; 
```

Thanks

Koenkk, 2022-11-03:
In order to add support for this device we need to know all the datapoints. As of now there are no other TuYa TS0601 power meters which supports 3 phases supported.

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

dokkomp, 2022-11-08:
Hi @Koenkk 

Phase A 
```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "101",
  "startTime": 1667735665677,
  "endTime": 1667908465677,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Phase B

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "111",
  "startTime": 1667735865038,
  "endTime": 1667908665038,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Phase C

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "121",
  "startTime": 1667735935284,
  "endTime": 1667908735284,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Temperature

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "133",
  "startTime": 1667736011242,
  "endTime": 1667908811243,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Today

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "4",
  "startTime": 1667736094391,
  "endTime": 1667908894392,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
This Month

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "3",
  "startTime": 1667736181854,
  "endTime": 1667908981854,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```

Power Factor A

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "102",
  "startTime": 1667736299434,
  "endTime": 1667909099434,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Power Factor B

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "112",
  "startTime": 1667736373907,
  "endTime": 1667909173907,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```

Power Factor C

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "122",
  "startTime": 1667736419437,
  "endTime": 1667909219437,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Frequency

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "132",
  "startTime": 1667736696238,
  "endTime": 1667909496238,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Total Current

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "131",
  "startTime": 1667736752738,
  "endTime": 1667909552738,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Total Active Power

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "1",
  "startTime": 1667736880383,
  "endTime": 1667909680383,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Device Status

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "134",
  "startTime": 1667737254763,
  "endTime": 1667910054763,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Total Power

```
{
  "startRowId": "",
  "pageNo": 1,
  "pageSize": 10,
  "code": "9",
  "startTime": 1667737349539,
  "endTime": 1667910149540,
  "projectCode": "p1667838625039mcnhq9",
  "sourceId": "eu16678397778059eFQ2",
  "sourceType": "4",
  "deviceId": "bf9b8f10af87702c9anpmv",
  "pageStartRow": "",
  "region": "EU"
}
```
Three more indicators:
Phase A  "code": "6" Event Details: "CIQABYMAASw="
Phase B  "code": "7" Event Details: "CHQAAOQAAAg="
Phase C  "code": "8" Event Details: "CJgADssAAwk="

Koenkk, 2022-11-08:
Can you try and see if with the following external converter ac_frequency, energy, power and temperature works?

https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d

dokkomp, 2022-11-09:
hi @Koenkk 

current
Instantaneous measured electrical current
51.49A

ac_frequency
Measured electrical AC frequency
0.5Hz

power
Instantaneous measured power
1035W

temperature
Measured temperature value
166°C

linkquality
Link quality (signal strength)
120lqi

```
Zigbee2MQTT:info  2022-11-09 10:42:06: MQTT publish: topic 'zigbee2mqtt/0xa4c138129f4abd2e', payload '{"ac_frequency":0.5,"current":52.76,"linkquality":120,"power":1054,"temperature":173}'
Zigbee2MQTT:warn  2022-11-09 10:42:09: Datapoint 134 not defined for '_TZE200_nslr42tt' with data {"dp":134,"datatype":5,"data":{"type":"Buffer","data":[0,0,0,0]}}
Zigbee2MQTT:warn  2022-11-09 10:42:12: Datapoint 6 not defined for '_TZE200_nslr42tt' with data {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[8,98,0,5,128,0,1,40]}}
Zigbee2MQTT:warn  2022-11-09 10:42:15: Datapoint 7 not defined for '_TZE200_nslr42tt' with data {"dp":7,"datatype":0,"data":{"type":"Buffer","data":[8,118,0,1,14,0,0,15]}}
Zigbee2MQTT:warn  2022-11-09 10:42:19: Datapoint 8 not defined for '_TZE200_nslr42tt' with data {"dp":8,"datatype":0,"data":{"type":"Buffer","data":[8,161,0,14,12,0,2,226]}}
Zigbee2MQTT:info  2022-11-09 10:42:22: MQTT publish: topic 'zigbee2mqtt/0xa4c138129f4abd2e', payload '{"ac_frequency":0.5,"current":52.76,"linkquality":124,"power":1051,"temperature":173}'
Zigbee2MQTT:warn  2022-11-09 10:42:25: Datapoint 1 not defined for '_TZE200_nslr42tt' with data {"dp":1,"datatype":2,"data":{"type":"Buffer","data":[0,0,15,161]}}
Zigbee2MQTT:warn  2022-11-09 10:42:28: Datapoint 101 not defined for '_TZE200_nslr42tt' with data {"dp":101,"datatype":2,"data":{"type":"Buffer","data":[0,0,39,191]}}
Zigbee2MQTT:warn  2022-11-09 10:42:31: Datapoint 111 not defined for '_TZE200_nslr42tt' with data {"dp":111,"datatype":2,"data":{"type":"Buffer","data":[0,0,37,237]}}
Zigbee2MQTT:warn  2022-11-09 10:42:34: Datapoint 121 not defined for '_TZE200_nslr42tt' with data {"dp":121,"datatype":2,"data":{"type":"Buffer","data":[0,0,78,169]}}
```

dokkomp, 2022-11-09:
I would output power in kilowatts 
power divided by 1000 (kW)

Koenkk, 2022-11-09:
> I would output power in kilowatts

Is the the current power or accumulated power? (energy). For all devices we output the current power in W.

I updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d and also added all the other points (values are the raw ones they require conversion). Can you check in the z2m forntend -> device -> state and see what values the attributes get?

dokkomp, 2022-11-10:
> ac_frequency - multiply by 10

Sorry, i was wrong.

ac_frequency - multiply by 100

> Is the the current power or accumulated power? (energy). 

current power

```
Zigbee2MQTT:warn  2022-11-10 08:02:29: Datapoint 6 not defined for '_TZE200_nslr42tt' with data {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[8,129,0,1,86,0,0,33]}}
Zigbee2MQTT:warn  2022-11-10 08:02:32: Datapoint 7 not defined for '_TZE200_nslr42tt' with data {"dp":7,"datatype":0,"data":{"type":"Buffer","data":[8,127,0,19,77,0,3,246]}}
Zigbee2MQTT:warn  2022-11-10 08:02:35: Datapoint 8 not defined for '_TZE200_nslr42tt' with data {"dp":8,"datatype":0,"data":{"type":"Buffer","data":[8,48,0,71,186,0,14,226]}}
```

[image]

dokkomp, 2022-11-11:
> Updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d , can you check the last part of https://github.com/Koenkk/zigbee2mqtt/issues/14790#issuecomment-1309197703 ?

Yes, I have updated the file.
I only get these values (Parameter values are correct):

```
current
Instantaneous measured electrical current
15.745A
ac_frequency
Measured electrical AC frequency
50Hz
power
Instantaneous measured power
2430W
temperature
Measured temperature value
13.7°C
linkquality
Link quality (signal strength)
164lqi
```

Here is the screenshot:
[image]
[image]

Koenkk, 2022-11-11:
What shows up under the "State" tab?

dokkomp, 2022-11-11:
> What shows up under the "State" tab?

{
    "ac_frequency": 50,
    "active_power": 15241,
    "current": 14.763,
    "device_status": 0,
    "linkquality": 168,
    "phase_a": 35404,
    "phase_b": 31417,
    "phase_c": 85599,
    "power": 2794,
    "power_factor_a": 64,
    "power_factor_b": 58,
    "power_factor_c": 98,
    "temperature": 19.5
}

dokkomp, 2022-11-11:
"active_power": 15241 / 100 = 152,4 kW
"phase_a": 35404 + "phase_b": 31417 + "phase_c": 85599 = 152420W = 152,4 kW 
>I'm wondering what the phase_a/b/c values could be

total W for each phase

[image]

[image]

[image]

Koenkk, 2022-11-12:
I updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d more should be under "Exposes" now, please check if it is correct. 

- Are the power_factor values under "state" correct?
- Are there any other datapoints for the voltage, current and power (not total power) per phase?

dokkomp, 2022-11-13:
Current version: 1.28.2-1

> more should be under "Exposes" now, please check if it is correct.

```
ac_frequency
Measured electrical AC frequency
50Hz

temperature
Measured temperature value
23.9°C

current
Instantaneous measured electrical current
2.76A

power
Instantaneous measured power
353W

energy
Sum of consumed energy
0.12kWh

energy
Sum of consumed energy
0.77kWh

energy
Sum of consumed energy
0.29kWh

energy
Sum of consumed energy
0.17kWh

linkquality
Link quality (signal strength)
140lqi
```

 "energy": 0.12 - wrong, should be 0,77+0,29 +0,17=1,23

```
{
    "ac_frequency": 50,
    "current": 2.522,
    "device_status": 0,
    "energy": 0.12,
    "energy_a": 0.77,
    "energy_b": 0.29,
    "energy_c": 0.17,
    "linkquality": 140,
    "power": 268,
    "power_factor_a": 61,
    "power_factor_b": 51,
    "power_factor_c": 25,
    "temperature": 23.9
}
```

> Are the power_factor values under "state" correct?

correct

> Are there any other datapoints for the voltage, current and power (not total power) per phase?

no others

There are these datapoints:

```
Zigbee2MQTT:warn  2022-11-10 08:02:29: Datapoint 6 not defined for '_TZE200_nslr42tt' with data {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[8,129,0,1,86,0,0,33]}}
Zigbee2MQTT:warn  2022-11-10 08:02:32: Datapoint 7 not defined for '_TZE200_nslr42tt' with data {"dp":7,"datatype":0,"data":{"type":"Buffer","data":[8,127,0,19,77,0,3,246]}}
Zigbee2MQTT:warn  2022-11-10 08:02:35: Datapoint 8 not defined for '_TZE200_nslr42tt' with data {"dp":8,"datatype":0,"data":{"type":"Buffer","data":[8,48,0,71,186,0,14,226]}}
```

Koenkk, 2022-11-14:
> "energy": 0.12 - wrong, should be 0,77+0,29 +0,17=1,23

Fixed

> Are the power_factor values under "state" correct? -> correct

Added them to the exposes

> There are these datapoints:

Added these now, are the voltage/current/power for a/b/c in the state correct? (e.g. `voltage_a`)

Updated: https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d

dokkomp, 2022-11-15:
```
{
    "ac_frequency": 50,
    "current": 5.338,
    "device_status": 0,
    "energy": 9292,
    "energy_a": 251.55,
    "energy_b": 188.97,
    "energy_c": 488.66,
    "linkquality": 116,
    "power": 940,
    "power_factor_a": 52,
    "power_factor_b": 51,
    "power_factor_c": 92,
    "temperature": 10.8
}
```
Now displays the total consumption for all each phase
"energy": 9292 - wrong, should be 251.55 +188.97 +488.66=929,2

Last time, the current consumption for each phase was displayed:

{
    "ac_frequency": 50,
    "current": 2.522,
    "device_status": 0,
    "energy": 0.12,
    "energy_a": 0.77,
    "energy_b": 0.29,
    "energy_c": 0.17,
    "linkquality": 140,
    "power": 268,
    "power_factor_a": 61,
    "power_factor_b": 51,
    "power_factor_c": 25,
    "temperature": 23.9
}

> Added these now, are the voltage/current/power for a/b/c in the state correct? (e.g. voltage_a)

Nothing is showing

`Zigbee2MQTT:error 2022-11-15 08:41:10: Exception while calling fromZigbee converter: Buffer is not defined}`

[image]

[image]

[image]

Koenkk, 2022-11-15:
Updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d

dokkomp, 2022-11-16:
Only this data:

```
{
    "ac_frequency": 50,
    "current": 3.554,
    "device_status": 0,
    "energy": 15142,
    "energy_a": 408.42,
    "energy_b": 292.5,
    "energy_c": 813.3,
    "linkquality": 132,
    "power": 533,
    "power_factor_a": 72,
    "power_factor_b": 17,
    "power_factor_c": 86,
    "temperature": 21.7
}
```

    "energy": 15142 - wrong

Koenkk, 2022-11-17:
Can you make sure you copied the converter correctly? Anyway I updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d now, energy should be divided by 10 now.

If it doesn't work change this line: 

```
[1, 'energy', tuya.valueConverter.divideBy100],
```

To one of
- `[1, 'energy', tuya.valueConverter.raw],`
- `[1, 'energy', tuya.valueConverter.divideBy10],`
And see if that fixes. Make sure to wait after updating and restarting z2m to make sure the device reports a new value.

dokkomp, 2022-11-18:
I reconnected the device
Here is the data:
```
{
    "ac_frequency": 50,
    "current": 15.774,
    "device_status": 0,
    "energy": 6.7,
    "energy_a": 0.72,
    "energy_b": 2.24,
    "energy_c": 3.81,
    "linkquality": 144,
    "power": 2896,
    "power_factor_a": 93,
    "power_factor_b": 97,
    "power_factor_c": 92,
    "temperature": 10.8
}
```
"energy": 6.7 - now it shows correctly

`[1, 'energy', tuya.valueConverter.divideBy10],`

Here is the data from the converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const Buffer = require('buffer');

const valueConverter = {
    divideBy1000: tuya.valueConverterBasic.divideBy(1000),
    phaseA: {
        from: (v) => {
            const buf = Buffer.from(v, 'base64');
            return {
                voltage_a: (buf[1] | buf[0] << 8) / 10,
                current_a: (buf[4] | buf[3] << 8) / 1000,
                power_a: (buf[7] | buf[6] << 8)};
        },
    },
    phaseB: {
        from: (v) => {
            const buf = Buffer.from(v, 'base64');
            return {
                voltage_b: (buf[1] | buf[0] << 8) / 10,
                current_b: (buf[4] | buf[3] << 8) / 1000,
                power_b: (buf[7] | buf[6] << 8)};
        },
    },
    phaseC: {
        from: (v) => {
            const buf = Buffer.from(v, 'base64');
            return {
                voltage_c: (buf[1] | buf[0] << 8) / 10,
                current_c: (buf[4] | buf[3] << 8) / 1000,
                power_c: (buf[7] | buf[6] << 8)};
        },
    },
};

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_nslr42tt']),
    model: 'TS0601_3_phase_clamp_meter',
    vendor: 'TuYa',
    description: '3-phase clamp power meter',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    whiteLabel: [{vendor: 'MatSeePlus', model: 'PC321-Z-TY'}],
    exposes: [e.ac_frequency(), e.temperature(), e.current(), e.power(), e.energy(), e.energy().withProperty('energy_a'), 
        e.energy().withProperty('energy_b'), e.energy().withProperty('energy_c'), e.power_factor().withProperty('power_factor_a'), 
        e.power_factor().withProperty('power_factor_b'), e.power_factor().withProperty('power_factor_c')],
    meta: {
        tuyaDatapoints: [
            [132, 'ac_frequency', tuya.valueConverter.raw],
            [133, 'temperature', tuya.valueConverter.divideBy10],
            [1, 'energy', tuya.valueConverter.divideBy10],
            [101, 'energy_a', tuya.valueConverter.divideBy100],
            [111, 'energy_b', tuya.valueConverter.divideBy100],
            [121, 'energy_c', tuya.valueConverter.divideBy100],
            [131, 'current', valueConverter.divideBy1000],
            [9, 'power', tuya.valueConverter.raw],
            [102, 'power_factor_a', tuya.valueConverter.raw],
            [112, 'power_factor_b', tuya.valueConverter.raw],
            [122, 'power_factor_c', tuya.valueConverter.raw],

            [6, null, valueConverter.phaseA],
            [7, null, valueConverter.phaseB],
            [8, null, valueConverter.phaseC],

            [134, 'device_status', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2022-11-20:
Updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d to fix the buffer issue (and correct the energy divisor)

dokkomp, 2022-11-20:
> Added these now, are the voltage/current/power for a/b/c in the state correct? (e.g. voltage_a)

correct

```
{
    "ac_frequency": 50,
    "current": 12.755,
    "current_a": 10.425,
    "current_b": 3.801,
    "current_c": 10.061,
    "device_status": 0,
    "energy": 1002.8,
    "energy_a": 249.83,
    "energy_b": 194.16,
    "energy_c": 558.72,
    "linkquality": 164,
    "power": 4462,
    "power_a": 2100,
    "power_b": 756,
    "power_c": 2017,
    "power_factor_a": 100,
    "power_factor_b": 43,
    "power_factor_c": 93,
    "temperature": 13.7,
    "voltage_a": 203.2,
    "voltage_b": 204.3,
    "voltage_c": 206.8
}
```

Koenkk, 2022-11-20:
Great, so everything seems to work now right? If you confirm then I can add this converter for support out-of-the-box.

dokkomp, 2022-11-20:
I didn't look at these parameters correctly:
   ```
 "energy": 1002.8,
    "energy_a": 249.83,
    "energy_b": 194.16,
    "energy_c": 558.72,
```
it should be

```
    "energy": 107.24,
    "energy_a": 26.981,
    "energy_b": 19.792,
    "energy_c": 60.477,

      [1, 'energy', tuya.valueConverter.divideBy100],
      [101, 'energy_a', valueConverter.divideBy1000],
      [111, 'energy_b', valueConverter.divideBy1000],
      [121, 'energy_c', valueConverter.divideBy1000],
```
there are also these options:

> Today "code": "4"
> This Month "code": "3"

can you show them too?

Values under "EXposes" are just these:
[image]
[image]

Maybe the energy should be renamed for each phase Energy A, Energy B, Energy C?

And I have another question. Why only three sensors on that device?:
 
[image]

Koenkk, 2022-11-20:
- Fixed the energy divisor

> there are also these options:
- These can only be retrieved when polled and we don't know how to do that yet.

> Maybe the energy should be renamed for each phase Energy A, Energy B, Energy C?
- Done, I also added the current/power/voltage per phase

> And I have another question. Why only three sensors on that device?:
The other ones are probably disabled, you can enable them by going to the device in HA

Updated https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d

dokkomp, 2022-11-20:
```
/app/dist/util/externally-loaded.js:27
    energy_phase: (phase) => new exposes.numeric(`energy_${phase}`, access.STATE).withUnit('kWh').withDescription(`Sum of consumed energy (phase ${phase.toUpperCase()})`),
                                                                    ^
ReferenceError: access is not defined
    at Object.energy_phase (/app/dist/util/externally-loaded.js:27:69)
    at /app/dist/util/externally-loaded.js:45:22
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
```

Koenkk, 2022-11-21:
Whoops, fixed!

dokkomp, 2022-11-22:
/app/dist/util/externally-loaded.js:79
module.exports = definition;const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
                                  ^
SyntaxError: Identifier 'fz' has already been declared
    at new Script (node:vm:100:7)
    at createScript (node:vm:265:10)
    at Object.runInNewContext (node:vm:306:10)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)

Koenkk, 2022-11-22:
https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d should be good now.

dokkomp, 2022-11-26:
Everything is displayed now

[image]
[image]
[image]

Maybe add an icon?
[image]

Koenkk, 2022-11-26:
Great, I've added the device, will be supported in the next release.

The icon will also be added with the next release, can you provide a product picture?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-26 by Koenkk

dokkomp, 2022-11-26:
[image]

Dogmeat88, 2023-01-21:
Hi. Thanks for the recent efforts in supporting this device.
I have just started using this device in conjunction with a grid tied solar setup.
It is mostly working well, but there is once significant issue. When exporting power I would expect the power reading to be a negative number, however instead it jumps to a very high number (see attached screenshot)
I have not looked at the code, but I am thinking it has been implemented as an unsigned Integer and is overflowing.
Is there workaround someone could recommend, or will this need a code change?

[image]

steko71, 2023-01-23:
Hello, same problem here. Can you fix this?
Can I contribute?
This device is very hot, would be good if it works 100%!

steko71, 2023-03-19:
Hello, is there any news about the part? Will this part work in the future?

Dogmeat88, 2023-03-19:
It would be great to get this fixed, is there any thing we can help with?

Der-Nax, 2023-03-26:
Hi, I have the same issue.
Power for phase 2, 3 and common power has very strange (fantasy) values. 
[image]
[image]
[image]

trozman, 2023-11-25:
First, thanks for the HA support for [TS0601_3_phase_clamp_meter]. It works great except:
I have a small grid-tied PV and when it exports to grid, it shows enormous positive values (~30000W) instead of the negative number.
(to me it also looks like a data type issue, unsigned int or something)
[image]
How can I help?

pipiche38, 2023-12-30:
> First, thanks for the HA support for [TS0601_3_phase_clamp_meter]. It works great except: I have a small grid-tied PV and when **it exports to grid**, it shows enormous positive values (~30000W) instead of the negative number (e. g. -300W). (to me it also looks like a data type issue, unsigned int or something) <img alt="image" width="410" src="https://private-user-images.githubusercontent.com/26304983/285590987-72b39a74-d7c1-451a-96d9-30e0c93d50dd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM5NzA4MzEsIm5iZiI6MTcwMzk3MDUzMSwicGF0aCI6Ii8yNjMwNDk4My8yODU1OTA5ODctNzJiMzlhNzQtZDdjMS00NTFhLTk2ZDktMzBlMGM5M2Q1MGRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyMzEyMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjMwVDIxMDg1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhNGJkZjQyMWIwNTU2MDc2MDQ3ZjIzN2M0NGRlNDc3ZTJmYTVmM2YyMTJjYjMwODNiNzExZDc2MThkZjM1OWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MKEyRezl9j7I5cz0l1HfQ0FT-0ID5-D150oV9xgvmvs"> How can I help?

I'm currently integrating this device in the Zigbee for Domoticz plugin, and I found the same. I have added a mask on the Power to prevent taking all bits on the Total Power

On the other side, I have some issue to get the line power and current

trozman, 2024-01-05:
I've created a 'patched' sensor for phase B and Total power in Home Assistant:
https://blog.rozman.info/tuya-3-phase-energy-meter-and-home-assistant/

waltervl, 2024-01-05:
> I've created a 'patched' sensor for phase B and Total power in Home Assistant: https://blog.rozman.info/tuya-3-phase-energy-meter-and-home-assistant/

Hi, I am using the same device but then with Domoticz instead of Home Assistant.
I noticed that when my solar panels are producing more power than I consume the power is reported negative but the energy is still positive (see also my github remark below). Do you see the same with zignee2mqtt and Home Assistant?

https://github.com/zigbeefordomoticz/z4d-certified-devices/issues/38#issuecomment-1879031266

waltervl, 2024-01-05:
Addition, to check I changed one of my Phase Energy meters to computed (instead of from device). Domoticz should ignore the Energy values coming from the device and calculate the Energy based on the power values. So lets see what the difference is with the other phase devices in Domoticz.

[image]

trozman, 2025-04-28:
Regarding the bug related to the reporting of phase B power (32xxx W) instead of negative value when export to grid: 
Today I noticed the bug is gone. Now it correctly reports negative power.
I don't know for how long is this fixed and who fixed it, but whoever did it, THANKS!