Koenkk/zigbee2mqtt #14791: Xiaomi ZNCZ04LM can't find update image

SAOPP, 2022-11-03:
### What happened?

Xiaomi ZNCZ04LM plug with firmware build date from `09-06-2019` can't find update image to `01-23-2022`.

The behavior is as follows: there are two outlets, one I updated a couple of weeks ago, the current one about I talking, it had an update available, but I couldn't update it because it's behind the server (I had to unplug it from the server so that during the process update, the socket didn't turn it off), when I did this and clicked on the update, the socket says that it cannot find the image to update, and the socket that has the latest firmware at all swears that the firmware is newer than the one in the repository, what the?

### What did you expect to happen?

Updating...

### How to reproduce it (minimal and precise)

Maybe try to update or check update this socket.

### Zigbee2MQTT version

1.28.1-dev commit: c3f6e79

### Adapter firmware version

20220219

### Adapter

ZigStar Lan POE

### Debug log

`Update of 'corridor_proxmox_socket' failed (No new image available)`

Koenkk, 2022-11-03:
Can you provide the debug log when checking for an update? I need to know the imageType and fileVersion.

SAOPP, 2022-11-03:
Here is on:

```
Debug 2022-11-03 22:54:19Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":16408,"fileVersion":22}'
Debug 2022-11-03 22:54:19ZigbeeOTA: downloaded main index
Debug 2022-11-03 22:54:19getNewImage for '0x04cf8cdf3c8a2bc8', meta {"fileVersion":22,"fileSize":276030,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750.ota","sha512":"a7fec7851a60696fb4f482f8fbbcfd638631bc460dca328427351baaf3a0c65a85191d450dab96700dd46c2ca9228efcd9a30c909d598718fe60210ee551d4f8"}
Debug 2022-11-03 22:54:19Update of 'corridor_proxmox_socket' failed (AssertionError [ERR_ASSERTION]: No new image available)
Info 2022-11-03 22:54:19MQTT publish: topic 'z2m-zigstar-poe/bridge/response/device/ota_update/update', payload '{"data":{"id":"corridor_proxmox_socket"},"error":"Update of 'corridor_proxmox_socket' failed (No new image available)","status":"error","transaction":"1cjzc-2"}'
Error 2022-11-03 22:54:19Update of 'corridor_proxmox_socket' failed (No new image available)
Debug 2022-11-03 22:54:19AssertionError [ERR_ASSERTION]: No new image available at getNewImage (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:426:5) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:286:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:255:21)
```

[image]
[image]

SAOPP, 2022-11-04:
@Koenkk is there any news about this?

Koenkk, 2022-11-05:
It seems it is already running the latest (https://github.com/Koenkk/zigbee-OTA/blob/73358c9c34a8b7844b5a262feb0503ffbdcc286a/index.json#L1025) (based on `fileVersion`). However if you want to force it you can follow https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files (then it would be interesting to see if it is indeed a new version)

SAOPP, 2022-11-05:
One of them yes, but one socket is using firmware from 2019 and I didn't update it to 2022 build.

closed 2022-12-14 by github-actions[bot]

SAOPP, 2022-12-14:
To the current day, I receive a message that my firmware is newer than in the repository + the socket with the old firmware from 2019 stupidly cannot be updated.

manny199, 2022-12-22:
News about new firmware upgrade?

SAOPP, 2022-12-22:
@manny199 are u faced with the same as mine?

rotdrop, 2023-02-04:
Hi, I have exactly the same problem:

- seemingly at some time two of my plugs got updated to the FW from 01-23-2022
- I have updated my coordinator FW as that firmware update resulting in missing sensor value
- now the two plugs work again and are still on FW 01-23-0222
- however the latest FW available is the one 09-06-2019

Could it be that the manufacturer has removed the new FW because it caused problems, like having to update the coordinator FW?

Otnow, 2023-02-04:
Yes, firmware version `0.0.0_0024` was originally [published](https://github.com/Koenkk/zigbee-OTA/pull/139), but it turned out to be [causing problems](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1259285782), so it was [withdrawn](https://github.com/Koenkk/zigbee-OTA/pull/169).

If you still want to update to version `0.0.0_0024`, you can do it [manually](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files):

[20220210104855_OTA_lumi.plug.mmeu01_0.0.0_0024_20220123_3B6A73.ota](https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220210104855_OTA_lumi.plug.mmeu01_0.0.0_0024_20220123_3B6A73.ota)

SAOPP, 2023-02-05:
Hm, very interesting, so we need is revert updated plug to previous firmware?

Otnow, 2023-02-05:
> Hm, very interesting, so we need is revert updated plug to previous firmware?

If firmware version `0.0.0_0024` [causes problems](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1259285782), then you can roll back to version `0.0.0_0022` in the [following way](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1259324227), otherwise you can leave everything as it is.

SAOPP, 2023-02-05:
Why z2m is still remind me about my 09-06-2019 build plug update?

Otnow, 2023-02-05:
> Why z2m is still remind me about my 09-06-2019 build plug update?

Strange. Can you show what it looks like in the debug log?