Koenkk/zigbee2mqtt #14798: Xiaomi 3 gateway network crash

jjpvdpoel, 2022-11-03:
### What happened?

On 1.28.1 together with Xiaomi 3 gateway, the zigbee herdsman crashes (won't stop, only after two restarts) and doesn't get zdo requests through after a few minutes of running. Back to 1.28.0 fixed the problem.

info  2022-11-03 22:16:04: Starting network scan (includeRoutes 'false')
error 2022-11-03 22:16:14: Failed to execute LQI for 'Coordinator'
debug 2022-11-03 22:16:14: Error: ZdoRequest error
    at Driver.zdoRequest (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:575:19)
    at request (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:275:32)
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:300:28)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
info  2022-11-03 22:16:14: Network scan finished
info  2022-11-03 22:18:32: Stopping zigbee-herdsman...
error 2022-11-03 22:18:38: Failed to stop Zigbee2MQTT

### What did you expect to happen?

Working comunication with gateway

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

6.7.10.0 build 423

### Adapter

Xiaomi 3 gateway

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9933337/log.txt)

jjpvdpoel, 2022-11-04:
Some extra info: all devices being in sleep seems to create the problem.

cacozzzz, 2022-11-04:
The same happened with my setup yesterday after upgrading to 1.28.1.

config:
zigbee2mqtt 1.28.1 docker container
xiaomi gateway 3 - connected as serial tcp EZSP
zigbee module patched by https://github.com/AlexxIT/XiaomiGateway3

Switching back to 1.28.0 fixes the issue.

SadCOW, 2022-11-04:
same problem

Vladimir1604, 2022-11-04:
Same problem(

tikhomirovkd, 2022-11-04:
Yes, the same problem.

BlackRockSoul, 2022-11-04:
Same for me

kirovilya, 2022-11-06:
try the current dev/edge version of z2m

jjpvdpoel, 2022-11-06:
Any info of how to pull a dev version in linux?
git checkout dev < doesn't seem to work.

kirovilya, 2022-11-06:
https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html#linux
if you install z2m via git it should work

jjpvdpoel, 2022-11-06:
This was the way i'd tried, but didn't work. Some complaining about 'dev' doesn't exist after the git checkout command.

babar1905, 2022-11-06:
same trouble

flight4287, 2022-11-07:
https://github.com/Koenkk/zigbee2mqtt/issues/1174#issuecomment-1305192637

mnrgrrt, 2022-11-28:
I had a lot of **problems with my xiaomi gateway 3** and my zigbee network as well. From some devices losing contact to complete freezes. I tried a lot of things to fix it and even switched from zigbee2mqtt to zha. But this didnt work either. Still the same problems. Untill this week I **fixed** my problem **completly**, **by accident**.

It turned out to be a problem with my WiFi network. I use 5g and 2.4g at the same time. I gave both networks exactly the same SSID. And later on I even hid the SSID for the 2.4g network. For all telephones and laptops, camera's etc both of this worked fine. So no suspection there.
This week I changed the SSID of the 2.4g to a unique name to install a Wifi device thats only on 2.4 and had problems pairing. After this the zigbee network did not drop devices or freeze anymore. So happy days over here and I hope this is a solution at least one other person as well.

asergiyenko4screen, 2022-12-03:
@jjpvdpoel looks like 1.28.4 has solved our issue. Maybe we should close it?

closed 2022-12-03 by jjpvdpoel

fredmoro, 2023-07-20:
> I had a lot of **problems with my xiaomi gateway 3** and my zigbee network as well. From some devices losing contact to complete freezes. I tried a lot of things to fix it and even switched from [...]

Amazing! Exactly the same happend to me! I use a Xiaomi Gateway 3 as Zigbee device. A couple of mounths ago I I've bought a new mesh router system, first I configured it with different SSIDs for the 2,4 and 5 Ghz channels.  Then recently I've changed the different SSIDs to the same (I used the SSID which the gateway was attached to - so xiaomi home was able to see the device, it seemed it's working fine). 

Some weeks ago I've noticed that the zigbee2mqtt suddenly stopped working, and then it didn't started anymore. After days of trying revive the zigbee2mqtt, Now I found this topic - changed back the Wifi mesh system to different SSIDs, and now everything is working fine!
(I use Asus RT-AX92U routers)

Thank You!