Koenkk/zigbee2mqtt #14803: Bticino K4027C/L4027C/N4027C/NT4027C Cover inverted states

silvioped, 2022-11-04:
### What happened?

For this device, controls behave right, but the state is always the opposite (close when open, open when close).
I already tried the "invert_cover" specific setting with no luck.
Is there any external converter I could use to fix it?

Thank you

### What did you expect to happen?

the state to be right

### How to reproduce it (minimal and precise)

add the device

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20220928

### Adapter

itead sonoff zigbee 3.0 usb dongle plus

### Debug log

Info 2022-11-04 08:58:50MQTT publish: topic 'zigbee2mqtt/mqtt_venezianaScorrevole', payload '{"linkquality":87,"position":100,"state":"CLOSE","tilt":100}'
Debug 2022-11-04 08:58:50Received Zigbee message from 'mqtt_venezianaScorrevole', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":0}' from endpoint 1 with groupID 0
Debug 2022-11-04 08:58:50Publishing 'set' 'state' to 'mqtt_venezianaScorrevole'
Debug 2022-11-04 08:58:50Received MQTT message on 'zigbee2mqtt/mqtt_venezianaScorrevole/set' with data 'OPEN'

Koenkk, 2022-11-05:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition =  {
    // Newer firmwares (e.g. 001f) Does support partial position reporting
    // Old firmware of this device provides only three values: 0, 100 and 50, 50 means an idefinite position between 1 and 99.
    // If you have an old Firmware set no_position_support to true
    // https://github.com/Koenkk/zigbee-herdsman-converters/pull/2214 - 1st very basic support
    zigbeeModel: [' Shutter SW with level control\u0000'],
    model: 'K4027C/L4027C/N4027C/NT4027C',
    vendor: 'BTicino',
    description: 'Shutter SW with level control',
    fromZigbee: [fz.identify, fz.ignore_basic_report, fz.ignore_zclversion_read, fz.bticino_4027C_binary_input_moving,
        fz.cover_position_tilt],
    toZigbee: [tz.bticino_4027C_cover_state, tz.bticino_4027C_cover_position, tz.legrand_identify,
        tz.legrand_settingEnableLedInDark],
    exposes: [e.cover_position()],
    // meta: {coverInverted: false},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

silvioped, 2022-11-05:
Yes thank you very much!!!

I had to disable the option "invert_cover"

Koenkk, 2022-11-06:
Thanks integrated the fix. You can remove your external converter after the next z2m release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-06 by Koenkk