Koenkk/zigbee2mqtt #14804: TV02 (TS0601 _TZE200_e9ba97vf) the wrong temperature is set when passing parameters to the schedule

semkyskov, 2022-11-04:
### What happened?

The wrong temperature is set above 26 degrees when transferring parameters to the schedule. Up to 26 degrees is set correctly, above it is transmitted incorrectly, as if 0 is taken as 25.6 (25=25, 25.5=25.5, 26=0.4, 27=1.4, ..., 30=4.4)
[image]

### What did you expect to happen?

Temperatures in the range up to 30 degrees will be transferred to the schedule correctly.

### How to reproduce it (minimal and precise)

When programming the schedule, set the temperature to 26 degrees and above.

### Zigbee2MQTT version

 1.28.1

### Adapter firmware version

20210708

### Adapter

zStack3x0 Sonoff

Koenkk, 2022-11-05:
@vladyspavlov  could you check this?

vladyspavlov, 2022-11-05:
> @vladyspavlov  could you check this?

Yes, I'll check

Koenkk, 2022-11-06:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4897

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-06 by Koenkk