Koenkk/zigbee2mqtt #14809: THS317-ET no temperature

colseven, 2022-11-04:
### What happened?

added two THS317-ET but i can't see any temperature
[image]
[image]
[image]

### Zigbee2MQTT version

1.28.1-1

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

danieledwardgeorgehitchcock, 2022-11-04:
Hmm...

Looks like the Z2M device converter uses Endpoint 3 whereas yours uses endpoint 1.

Could you try [this external converter](https://gist.github.com/danieledwardgeorgehitchcock/9ecad05b665373bd3bcea71d13f46371#file-owon_ths317-et-js) and let me know how you get on? you will have to remove your existing Z2M after adding it...

colseven, 2022-11-04:
Excuse me, I have a second temperature sensor (the same type) if I teach this in addition, unfortunately there are no values here either. I've retrained him several times - always with the same result.

[image]

danieledwardgeorgehitchcock, 2022-11-04:
Could you show your cluster screen, or better still share the database.db entries for both of these devices?

colseven, 2022-11-04:
`{"id":44,"type":"EndDevice","ieeeAddr":"0x3c6a2cfffed24582","nwkAddr":5114,"manufId":4412,"manufName":"OWON","powerSource":"Battery","modelId":"THS317-ET","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,32,0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"stackVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c28d50","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"20220721","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1667576214757,"defaultSendRequestWhen":"active"}
{"id":45,"type":"EndDevice","ieeeAddr":"0x3c6a2cfffed2450b","nwkAddr":44480,"manufId":4412,"manufName":"OWON","powerSource":"Battery","modelId":"THS317-ET","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,32,0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"stackVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":31,"batteryPercentageRemaining":200}},"msTemperatureMeasurement":{"attributes":{"measuredValue":4930}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c28d50","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c28d50","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c28d50","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"20220721","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1891685863},"lastSeen":1667576234791,"defaultSendRequestWhen":"active"}`

[image]

danieledwardgeorgehitchcock, 2022-11-04:
All looks good on the device front - and the external converter should work the same.

Is that after a device remove and re-add?

It may take a little while for the device to report back any data...

colseven, 2022-11-04:
I've deleted and reprogrammed the temperature sensor several times. It's been on for over an hour now and isn't sending any temperature or battery status.

colseven, 2022-11-04:
After a complete restart and re-learning, the second sensor now also works. Thanks

closed 2022-11-04 by colseven

danieledwardgeorgehitchcock, 2022-11-04:
> After a complete restart and re-learning, the second sensor now also works. Thanks

Great! I was scatching my head as I could see they were set up the same just that the reporting hadn't been bound.

Glad that it finally worked for you!

jonathanD91, 2022-11-05:
bonjour, j'ai parfaitement le meme probleme. comment je dois faire? Quand je clique sur votre convertisseur externe, j'ai une page 404

danieledwardgeorgehitchcock, 2022-11-05:
Sorry,

I deleted it when I created the Pull Request - I will re-create it and upload it here later..

jonathanD91, 2022-11-05:
ok perfect, I'm waiting impatiently because it seems to work

danieledwardgeorgehitchcock, 2022-11-05:
It looks like my PR has just been merged so, you could wait a couple of hours and switch to the Dev branch of Z2M if you wish?

jonathanD91, 2022-11-05:
ok if I immediately switch to the dev branch, that's good too?

danieledwardgeorgehitchcock, 2022-11-05:
I'd wait for a couple of hours to ensure that the build process has finished but, you might get lucky..?

jonathanD91, 2022-11-05:
Sorry, but do you think it's good now?

danieledwardgeorgehitchcock, 2022-11-05:
Yeah, should be fine now

jonathanD91, 2022-11-05:
It still doesn't work for me

jonathanD91, 2022-11-05:
however I am on the dev branch[
[image]

danieledwardgeorgehitchcock, 2022-11-05:
Try removing the device, restarting Z2M, then re-adding

jonathanD91, 2022-11-05:
I just did but still nothing

jonathanD91, 2022-11-05:
When I add I get the same error message.
[image]

jonathanD91, 2022-11-05:
when i look in owon.js file i don't have endpoint 1 only 3
```    {
        zigbeeModel: ['THS317-ET'],
        model: 'THS317-ET',
        vendor: 'OWON',
        description: 'Temperature sensor',
        fromZigbee: [fz.temperature, fz.battery],
        toZigbee: [],
        exposes: [e.battery(), e.temperature()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(3);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await reporting.temperature(endpoint);
            await reporting.batteryVoltage(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
            device.powerSource = 'Battery';
            device.save();```

colseven, 2022-11-06:
@jonathanD91 
a complete restart helped for me, then i re-trained the sensor and it worked straight away.

jonathanD91, 2022-11-06:
I stopped zigbee2mqtt and restarted several times but nothing

danieledwardgeorgehitchcock, 2022-11-06:
Here's the external converter I created:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const definition = {
    zigbeeModel: ['THS317-ET'],
    model: 'THS317-ET',
    vendor: 'OWON',
    description: 'Temperature sensor',
    fromZigbee: [fz.temperature, fz.battery],
    toZigbee: [],
    exposes: [e.battery(), e.temperature()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1) || device.getEndpoint(3);
        await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await reporting.temperature(endpoint);
        await reporting.batteryVoltage(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
        device.powerSource = 'Battery';
        device.save();
    },
};

module.exports = definition;
```

jonathanD91, 2022-11-06:
@danieledwardgeorgehitchcock I replace my code in owon.js with yours in copy paste?

danieledwardgeorgehitchcock, 2022-11-06:
Add it as an external converter - don't replace the code

jonathanD91, 2022-11-06:
what is the procedure for adding a converter? I don't know sorry

danieledwardgeorgehitchcock, 2022-11-06:
Follow [this guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device) but just paste in what I have added above

jonathanD91, 2022-11-06:
I'm a noob I don't understand how to do it even with the instructions. I don't know where to start on my terminal

danieledwardgeorgehitchcock, 2022-11-06:
In that case then, it's probably best you wait for the change to appear in Dev branch, or wait for the proper release

jonathanD91, 2022-11-06:
ok