Koenkk/zigbee2mqtt #14812: Switch to "auto" for device_mode is not working on Legrand (412171) - DIN contactor module ( Bticino FC80CC)

celestianx, 2022-11-04:
### What happened?

In z2M front-end, when clicking on "auto", the contactor will not switch although z2m consider it done and display state to "auto". 
Switch to "manual" is working perfectly as intended.

### What did you expect to happen?

I expected that when clicking on "auto" mode, the contactor would switch to "auto" and if no "valid" response from the contactor, then get an error message.
The 0x004 code does not seem to be the right code (or it has changed in a recent firmware update?)

### Zigbee2MQTT version

1.28.1-dev

### Adapter firmware version

20220219

### Adapter

zStack3x0 (Sonoff 3.0 Dongle)

### Debug log

```
Debug 2022-11-04 19:00:35Received MQTT message on 'zigbee2mqtt/0x000474000083fc0e/set' with data '{"device_mode":"auto"}'
Debug 2022-11-04 19:00:35Publishing 'set' 'device_mode' to '0x000474000083fc0e'
Info 2022-11-04 19:00:35MQTT publish: topic 'zigbee2mqtt/0x000474000083fc0e', payload '{"device_mode":"auto","led_when_off":"OFF","led_when_on":"OFF","linkquality":54,"power":0,"state":"OFF"}'
Debug 2022-11-04 19:00:54Received Zigbee message from '0x000474000083fc0e', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"apparentPower":0}' from endpoint 1 with groupID 0
```

celestianx, 2022-11-04:
#https://github.com/Koenkk/zigbee2mqtt/issues/14812

celestianx, 2022-11-07:
After some investigations, I believe the device is not fully supported yet. In the official app, the device mode can be configured between "auto" and "switch", respectively "peak/off-peak" and control "on/off". Based on the configuration set, the device allow different types of actions (or "enums") with the contactor.

Here is the screen to select the device mode:
[image]

Once it has been set for "auto", the device will offer 3 "states": "off", "auto" and "on"
[image]

Once it has been set for "switch", the device will offer 2 "states": "off" and "on"
[image]

As of today, the implementation of converter is as such:
```
 {
        zigbeeModel: [' Contactor\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
            '\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        model: '412171',
        description: 'DIN contactor module ( Bticino FC80CC )',
        vendor: 'Legrand',
        extend: extend.switch(),
        fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement, fz.legrand_cluster_fc01, fz.ignore_basic_report, fz.ignore_genOta],
        toZigbee: [tz.legrand_deviceMode, tz.on_off, tz.legrand_identify, tz.electrical_measurement_power],
        exposes: [exposes.switch().withState('state', true, 'On/off (works only if device is in "switch" mode)'),
            e.power().withAccess(ea.STATE_GET), exposes.enum('device_mode', ea.ALL, ['switch', 'auto'])
                .withDescription('switch: allow on/off, auto will use wired action via C1/C2 on contactor for example with HC/HP')],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
            await reporting.onOff(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.activePower(endpoint);
        },
    },
```
While the implementation currently supports:
- The `toZigbee` implements only `tz.legrand_deviceMode` to select the configuration of the conductor (auto, switch)
- The tz.on_off implements the toggle on/off
But once in "auto", there's no "tz.auto_mode" and exposed options should be conditioned by the value of device_mode

Here you can see the frames captures in Wireshark:
[image]
[image]

The frame seems Legrand specific with:
Cluster: 0x1021 (Vantage Controls)
Command: Unknown (0x00)
And data being a byte varying between:
- "00" for "Off", 
- "01" for "Auto" and 
- "02" for "On (Override)"

@Koenkk : I could help support it but would need guidance to:
1.  Pass a payload "Data: XX",
2. How can I conditionally expose "toggle" and some kind of enum?

Thanks!

Koenkk, 2022-11-08:
1. To support this command it needs to be added to https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts
2. Conditional exposes are supported but i would not recommend it for this case. (if I understand correctly, it should be enabled if a certain option is enabled). If the converter is called when this option is not enabled you can  just throw an error.

celestianx, 2022-11-09:
Hi @Koenkk , thank you for your response:

> 2. Conditional exposes are supported but i would not recommend it for this case. (if I understand correctly, it should be enabled if a certain option is enabled). If the converter is called when this option is not enabled you can just throw an error.

When in "auto", it should display "Off", "Auto", "On (Override)" for which these actions targets a specific cluster "0xfc41" which needs to be added in z2m (to your point in 1.)
When in "switch", it should display "Off" and "On" for which these actions targets "'genOnOff' (or what I understand)

Conditional exposes appears to me as a "cleaner" solution as we either leverage already coded "genOnOff", otherwise we need to add logic in a new function to route either to "genOnOff" or  "0xfc41" 

What do you think?

> 1. To support this command it needs to be added to https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

As I am discovering z2m, what I understand from you (supposing we go towards what you suggest. I need to:
1. Add a new cluster definition in "cluster.ts":

```
manuSpecificLegrandDevices3: {
        ID: 0xfc41,
        manufacturerCode: ManufacturerCode.LegrandNetatmo,
        attributes: {},
        commands: {
            command0: {
                ID: 0,
                parameters: [
                    {name: 'data', type: BuffaloZclDataType.BUFFER},
                ],
            },
        },
        commandsResponse: {}
    },
```

Then, add to this to tz.js:
```
legrand_dinContactorState: {
    key: ['din_contactor_state'],
    convertSet: async (entity, key, value, meta) => {
        const state = {
            'off': 0x01,
            'auto': 0x02,
            'on_override': 0x03
        };
        const payload = {data: Buffer.from([mode[value]])};
        await entity.command('manuSpecificLegrandDevices3', 'command0', payload);
        ---- NEED TO SEE IF WE ADD CONDITIONS TO SEND COMMANDS -----
        return {state: {'din_contactor_state': value}};
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('manuSpecificLegrandDevices3', [0x0000], manufacturerOptions.legrand);
    },
},
```

A few questions for me to understand:
1. In the cluster, why would we need to add BuffaloZclDataType.BUFFER (I copied from the cable_outlet_mode as the frame is the same except for the cluster). In the frame we receive, we don't see a "type" declared.
2. What is the convertGet for? Shouldn't any "Get" definition be in the "fromZigbee" module?
3. How can we determine the attributes of a cluster? Is there anyway to easily get these from the device?
4. For me to bind an expose to a cluster, I would need to use:
```
exposes.enum('din_contactor_state', ea.ALL, ['off', 'auto', 'on_override'])
```
And the conditional exposition would have to be handled by adapting ea.ALL accordingly?

celestianx, 2022-11-09:
Also, is there any way to easily build a "dev" environment for me to test things beyond addiing extensions/controller?
What would be you recommendation with a docker?

Koenkk, 2022-11-09:
> Conditional exposes appears to me as a "cleaner" solution as we either leverage already coded "genOnOff", otherwise we need to add logic in a new function to route either to "genOnOff" or "0xfc41"

different exposes for the same property are not supported. What you can do in the toZigbee converter is to check the `meta.state.device_mode` (or whatever the mode property is called) and send the correct command based on that. When in `switch` and `auto` is send, just throw an error that it is only supported in `auto`.

> In the cluster, why would we need to add BuffaloZclDataType.BUFFER (I copied from the cable_outlet_mode as the frame is the same except for the cluster). In the frame we receive, we don't see a "type" declared.

What do you mean with "frame we recieve"? Does the device reply after setting it?

> What is the convertGet for? Shouldn't any "Get" definition be in the "fromZigbee" module?

To read the current state, you can ignore this

> How can we determine the attributes of a cluster? Is there anyway to easily get these from the device?

This cannot be done from Zigbee2MQTT 

> For me to bind an expose to a cluster, I would need to use:

The expose just states what values are possible, but see my suggestion above

> Also, is there any way to easily build a "dev" environment for me to test things beyond addiing extensions/controller?
What would be you recommendation with a docker?

Enter the docker container with `docker exec -it CONTAINER_ID /bin/bash`, edit (make sure to edit the `dist` folder for zigbee-herdsman!), restart docker container with `docker restart CONTAINER_ID`

celestianx, 2022-11-10:
Thank you @Koenkk. I have implemented the support with a few questions:

> What is the convertGet for? Shouldn't any "Get" definition be in the "fromZigbee" module?
To read the current state, you can ignore this

For my own understanding, is it related to "withAccess"? Would you care explaining how it actually works for my own understanding?

Also I would like to understand more "clusters". When defined:
- Attributes are declared to facilitate access to them when using "read" or "write"
- Commands are used when non-standard commands are required
- commandsResponse are used when device provides a response to allow z2m to "translate" it
Is that correct?

Now onto supporting the device, I have updated the code and it actually makes more sense to split exposed features (as it is in the original app):
[image]

Two questions:
1. Is there anyway to grey ou (make it appear as disabled) based on the device_mode value?
2. In the dev console, why my new cluster is not appearing in the cluster list?

Koenkk, 2022-11-10:
> For my own understanding, is it related to "withAccess"? Would you care explaining how it actually works for my own understanding?

Yes it is, if a get has is defined, the access also has to contain `GET` 

> Commands are used when non-standard commands are required

What do you mean with "no-standard commands"?

> commandsResponse are used when device provides a response to allow z2m to "translate" it
Is that correct?

A command response is basically a command but then with the direction bit set in to the different order.

These concepts are not Zigbee2MQTT specific and are well documented in the ZCL: https://github.com/Koenkk/zigbee-herdsman/blob/master/docs/07-5123-08-Zigbee-Cluster-Library.pdf

> Is there anyway to grey ou (make it appear as disabled) based on the device_mode value?

No this is not possible

> In the dev console, why my new cluster is not appearing in the cluster list?

Because only a subset of the standard clusters are here (https://github.com/nurikk/zigbee2mqtt-frontend/blob/3ce5d7b422e3036cf30cb115155d2279db4f9c2b/src/components/cluster-picker/index.tsx)

celestianx, 2022-11-14:
Hi @Koenkk,
I have made 2 PRs to add full support for this device. The PR for zigbee-herdsman-converters fail for some reason about accesses. Could you please tell me what I have done wrong? Thanks!
https://github.com/Koenkk/zigbee-herdsman-converters/actions/runs/3459060338/jobs/5774071064

Koenkk, 2022-11-14:
I'll check the prs

closed 2022-12-22 by github-actions[bot]