Koenkk/zigbee2mqtt #14813: Devices leave the network just after pairing

eskholm, 2022-11-04:
### What happened?

I’ve had this problem for a while now. Months. 

When I add a new device to the network, initial pairing goes problem free. But a few seconds after adding the device successfully i get a message “Device 0xXXXXXXXXXX has left the network” and the device disappears from Z2M. 

I can make the device work by taking the battery out/removing the power to the device as soon as I see the “successfully configured” message. When I re-apply power the device joins as nothing happened and everything works as expected. 

This behaviour is consistent across a lot of different devices. Sonoff, Hue, Aqara, IKEA…

Any ideas?

### What did you expect to happen?

That the device doesnt leave the network right after pairing. 

### How to reproduce it (minimal and precise)

Add device. Wait for it to join. Watch it leave the network. 

### Zigbee2MQTT version

Zigbee2MQTT version 1.28.1 commit: 31f0022

### Adapter firmware version

20220219

### Adapter

Zzh!

closed 2022-12-12 by github-actions[bot]

Aeilstrup, 2023-01-01:
I see multiple issues with the same problem as I have, but all closed and stale without any resolution. that's sad. would have loved to see some sort of fix.

cephos, 2023-03-13:
I also still have the same Problem with Version [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2) commit: [360a777a](https://github.com/Koenkk/zigbee2mqtt/commit/360a777a)

Using a SONOFF ZBDongle-P with version 20220219.