Koenkk/zigbee2mqtt #14814: TV02-Zigbee local_temperature_calibration

iksrd, 2022-11-04:
### What happened?

TV02-Zigbee local_temperature_calibration passes values independently
[image]

### What did you expect to happen?

If I set "0" then "0" will remain

### How to reproduce it (minimal and precise)

Set "0" and just watch for a while

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20211217

### Adapter

zStack3x0

vladyspavlov, 2022-11-05:
I can't reproduce it on my TRV.
Here's a history for the last 5 days, there are no issues.
<details>
<summary>Screenshot</summary>

[image]
</details>

Do you have any automations for this entity? Are you sure that it doesn't change because of any automation?
Could you provide the manufacturer ID of the device, please?

misha1183, 2022-12-03:
Здравствуйте. Можно увеличить значение Local temperature colibration до значений от -15 до 15? TV02 стоит на радиаторе отопления и калибровка нужна на бОльшие значения. Сейчас необходимо -9,2 градуса.

closed 2023-01-10 by github-actions[bot]