Koenkk/zigbee2mqtt #14817: Lonsonho TS110E_2gang dimmer - transition does not work

chojnicki, 2022-11-05:
### What happened?

I'm trying to set brightness with transition, but whatever value I set as `transition` - it is ignored by device or zigbee2mqtt (I'm not sure who's exactly is responsible for transitions). Bulb is changing brightness instantly. 
Note that dimming by device itself (using button directly or when turning off) has transition. 

I'm doing same thing with some other lights for ex. from IKEA and everything works fine, so it's something specific with this Lonsonho.

### What did you expect to happen?

Light should be dimmed with transition

### How to reproduce it (minimal and precise)

Publish payload `{"brightness_l1": 254, "transition": 3}`

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

26720700

### Adapter

ConBee 2

### Debug log

'{"brightness_l1":254,"brightness_l2":255,"last_seen":"2022-11-05T10:23:40.108Z","linkquality":255,"power_outage_memory_l1":"restore","power_outage_memory_l2":"restore","state_l1":"ON","state_l2":"OFF","switch_type_l1":"momentary","switch_type_l2":"state"}'
Zigbee2MQTT:info  2022-11-05 11:23:40: MQTT publish: topic 'zigbee2mqtt/living-room-ceiling-light-dimmer-switch/l1', payload '{"brightness":254,"power_outage_memory":"restore","state":"ON","switch_type":"momentary"}'
Zigbee2MQTT:info  2022-11-05 11:23:40: MQTT publish: topic 'zigbee2mqtt/living-room-ceiling-light-dimmer-switch/l2', payload '{"brightness":255,"power_outage_memory":"restore","state":"OFF","switch_type":"state"}'

Koenkk, 2022-11-05:
Transition is not supported for these device (they use a TuYa api)

chojnicki, 2022-11-05:
> Transition is not supported for these device (they use a TuYa api)

@Koenkk could you maybe elaborate? Can I do something about this, use that api?
Why it is listen in docs then? 
https://www.zigbee2mqtt.io/devices/TS110E_2gang.html#lonsonho-ts110e_2gang

Koenkk, 2022-11-06:
I was wrong about the TuYa api part; it uses the ZCL but it seems it does not honour the transition value. So I would expect it is not supported by the fw on the device. @kiat-huang  can you confirm this?

Robert7997, 2022-11-11:
Using Z2M only, could you make one channel dimmable and force the other to only work as a switch On/Off?  Know this is slightly off topic, but I am struggeling a little here.

chojnicki, 2022-11-11:
@Koenkk also I noticed that with this device I cannot do `brightness_move`, tried also or `brightness_move_l1` etc. Nothing happens. But it is also described in docs.

Koenkk, 2022-11-12:
@chojnicki seems the device also does not support them, does the stepping work? (`brightness_step`) Then I'll update the docs.

chojnicki, 2022-11-12:
@Koenkk nope, also tested with 1 gang version because it has different firmware (less exposes)

Koenkk, 2022-11-14:
I've disabled it for both, the docs will automatically be updated with this with the next release.

Assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-14 by Koenkk