Koenkk/zigbee2mqtt #14823: [New device support]:

Druidblack, 2022-11-05:
### Link

https://aliexpress.ru/item/1005004704966668.html?spm=a2g2w.orderdetails.0.0.5cc24aa6itTK6e&sku_id=12000030168767259

### Database entry

{"id":82,"type":"Router","ieeeAddr":"0xdc8e95fffe5991b7","nwkAddr":42888,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000fF�*i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1667653701183,"defaultSendRequestWhen":"immediate"}

### Comments

I have bought this human presence sensor, because is on devices supported list: https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presense_sensor.html. Unfortunately, when pairing, still have error "unsupported device...". I think the main problem with this sensor is the manufacturer name TZE204_ztc6ggyl". 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ikvncluo'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_lyetpprm'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_wukb7rhc'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_jva8ink8'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_ztc6ggyl'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ztc6ggyl'}],
    model: 'TS0601_smart_human_presense_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [fz.tuya_smart_human_presense_sensor],
    toZigbee: [tz.tuya_smart_human_presense_sensor],
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
};

module.exports = definition;
```

Koenkk, 2022-11-06:
Does it work with: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ikvncluo'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_lyetpprm'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_wukb7rhc'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_jva8ink8'},
            {modelID: 'TS0601', manufacturerName: '_TZE204_ztc6ggyl'},
        {modelID: 'TS0601', manufacturerName: '_TZE204_ztc6ggyl'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ztc6ggyl'}],
    model: 'TS0601_smart_human_presense_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [fz.tuya_smart_human_presense_sensor],
    toZigbee: [tz.tuya_smart_human_presense_sensor],
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.'),
    ],
};

module.exports = definition;
```

Druidblack, 2022-11-06:
It works.
You have duplicated the above converter and duplicated twice the string {modelId: 'TS0601', ManufacturerName: '_TZE204_ztc6ggyl'},

Koenkk, 2022-11-06:
I did, added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-06 by Koenkk

richardwonka, 2023-05-03:
Hey there, I'm pretty sure I have the same device, but the ManufacturerName is `_TZE204_sxm7l9xa` - It looks as if the ManufacturerName has a randomised postfix - as if they are using ManufacturerName for an individual device name or possibly the postfix encodes for re-labels of the same build type.

tomiss288, 2023-05-05:
> Hey there, I'm pretty sure I have the same device, but the ManufacturerName is `_TZE204_sxm7l9xa` - It looks as if the ManufacturerName has a randomised postfix - as if they are using [...]

Are you connected your device ?
I tried to use external converter from that link: https://github.com/Koenkk/zigbee2mqtt/issues/14823#issuecomment-1304793499 , but I have a error:

fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,86]}}

Koenkk, 2023-06-01:
@saschaludwig `_TZE204_sxm7l9xa` is supported in z2m 1.31.0

Igor-Andrusevic, 2023-07-31:
I also have a TS0601 Manufacturer _TZE204_ijxvkhd0 which isn't recognized.
https://www.aliexpress.com/item/1005005682042230.html?spm=a2g0o.order_list.order_list_main.5.3e4418021PBW37
Any advice on how to support that kind of sensor?