Koenkk/zigbee2mqtt #14824: Innr SP 234 not reporting energy consumption

epower53, 2022-11-05:
### What happened?

Innr SP 234 is listed as reporting energy measurements in the docs, but now that I've had one installed for 24 hours, I have yet to see it report energy.

### What did you expect to happen?

The energy topic would be populated with a non-null numeric value and monotonically increasing

### How to reproduce it (minimal and precise)

Pair an Innr SP 234. Wait. No energy reports show up in the z2m interface, and none are pushed to Mosquitto (or whatever broker you use).

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20220928

### Adapter

Sonoff Zigbee Dongle Plus (P edition)

### Debug log

Nov 05 09:30:32.788050 Collective-Core conmon[361840]: Zigbee2MQTT:info  2022-11-05 09:30:32: MQTT publish: topic 'zigbee2mqtt/Family Room TV Outlet', payload '{"current":0.03,"energy":null,"last_seen":1667655032773,"linkquality":142,"power":0,"state":"ON","voltage":122}'
Nov 05 09:30:41.752530 Collective-Core conmon[361840]: Zigbee2MQTT:info  2022-11-05 09:30:41: MQTT publish: topic 'zigbee2mqtt/Family Room TV Outlet', payload '{"current":0.03,"energy":null,"last_seen":1667655041727,"linkquality":138,"power":0,"state":"ON","voltage":122}'

jschornick, 2022-11-13:
I was seeing a similar result (`"energy":null"`) from my SP 234 until I manually added the appropriate attribute under the `Reporting` tab for the device. Specifically I added `currentSummDelivered` from the `seMetering` cluster.

After that change, I see a reasonable looking energy value under the device's `State` and `Exposes` tabs (as well as the dashboard). It also seems to be picked up by Home Assistant without further configuration.

Hope that works for you!

Koenkk, 2022-11-13:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4942

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-13 by Koenkk