Koenkk/zigbee2mqtt #14827: sendZclFrameToEndpointInternal when sending data

and7ey, 2022-11-05:
### What happened?

Zigbee2MQTT started to fail when sending data to several devices:

- Tuya [WHD02](https://www.zigbee2mqtt.io/devices/WHD02.html#tuya-whd02)
- [Aqara D1](https://www.zigbee2mqtt.io/devices/WXKG06LM.html#xiaomi-wxkg06lm)

Other devices work well.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8 (xiaomi gateway 3)

### Debug log

```
Debug 2022-11-05 18:26:02 Received MQTT message on 'zigbee2mqtt/hood_bathroom/set' with data '{"state":"ON"}'
Debug 2022-11-05 18:26:02 Publishing 'set' 'state' to 'hood_bathroom'
Error 2022-11-05 18:26:04Publish 'set' 'state' to 'hood_bathroom' failed: 'Error: Command 0xa4c1385620b4f385/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Debug 2022-11-05 18:26:04 Error: Command 0xa4c1385620b4f385/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:465:19) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-11-05 18:26:04 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'hood_bathroom' failed: 'Error: Command 0xa4c1385620b4f385/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"hood_bathroom"},"type":"zigbee_publish_error"}' 
```

kirovilya, 2022-11-06:
try the current dev/edge version of z2m

and7ey, 2022-11-06:
> try the current dev/edge version of z2m
Thanks. Could you please clarify how to install dev/edge version?

kirovilya, 2022-11-06:
https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

reyz3ll, 2022-11-15:
> try the current dev/edge version of z2m

i have the same problem.

I have tried with the edge version and now it seems to work fine. I will try for a day to see how stable it is.

Thanks!

and7ey, 2022-11-16:
I am on the edge version for a few days, it works well.

ivanp123, 2022-11-21:
Same problem here with stable version 1.28.2
1.28.2-dev (commit #85be7d2) seems to work fine.

ivanp123, 2022-11-24:
Even on dev/edge version the devices stop working after a couple days, requires restarting z2m to work again. 
Now I'm reverting to stable 1.27.2

and7ey, 2022-11-24:
I also faced with the problem today, when it worked well for 8 days. But I've upgraded to HA 2022.11.4, restoring 2022.11.2 now.

ivanp123, 2022-12-09:
After 2 weeks on v1.27.2 I can confirm no more issues, tried with HA 2022.11.2 and 2022.11.4. So I'm almost sure the problema is caused by z2m 1.28.x, no matter what version of HA you have.

andy-81, 2022-12-15:
I am receiving the same error as this with all my devices after a period of time (the time varies). This coincides with the devices no longer reporting in. A restart of the service brings all the devices back online.

Issue appears to still be in 1.28.4 & 1.28.4-dev. I am running via HA add-ons so unable to test reverting back to an older version.

ivanp123, 2023-01-13:
Tried to update to 1.29.1 and the issue appared again, devices stop working after a few days. A restart of z2m takes the devices back online. 
1.27.2 seems to be the last version working with no issues on my EZSP adapter.

Nipal33, 2023-01-24:
> I am receiving the same error as this with all my devices after a period of time (the time varies). This coincides with the devices no longer reporting in. A restart of the service brings all the [...]

I have the same issue.
I used to have a RPI with a Sonoff coordinator it worked wonderfully. I switched to a HA Yellow and since then I have this error. It works well but at some point it will generate this error "failed (sendZclFrameToEndpointInternal error)" and then nothing works (nothing more is received and all actions on a Zigbee device will generate the same error).

To fix it I need to restart the service, when I do so, I will lose some devices. I mean some devices, which looks a bit random, will leave the network. When I repair them they will join back and leave again. After a random number of repairing they will stay and work well until the next "crash".
It is not a proper crash as Z2M is considered running and does not restart with watchdog.

gecko17, 2023-02-02:
I have the same issue! After a few days or a about a week, all my router devices cannot be pinged anymore with the following error (logged as warning...) `warn  2023-01-31 07:45:06: Failed to ping 'Deckenleuchte DG' (attempt 1/1, Read 0x00124b0026b6ae1a/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))`.

The result is that none of my devices are working anymore, except the three devices, which are directly connected to the coordinator. I can see in the logs, that those are still working.

After restarting Z2M, it works again, however I will loose exactly those 3 devices, which are directly connected to the coordinator. Then I need to repair those 3 devices.

This issue seems to be relevant since 1.28.2 and was still relevant with 1.29.2. I am now running 1.30.0 and will see if this resolved the issue...

I am wondering why there is still no response from @Koenkk on this issue here. There seem to be other issues open related to this like https://github.com/Koenkk/zigbee2mqtt/issues/16311.

I am using a Sonoff Zigbee Dongle Plus E (EZSP v8) with version 6.10.3.0 build 297.

and7ey, 2023-02-02:
I have the same problem. It worked perfectly before 1.28.2 with xiaomi gateway 3, but not it works for some time, then something happens and it doesn't work, restart of z2m helps for some time and then again I have to restart it.

Koenkk, 2023-02-02:
@gecko17 because I'm not maintaining the EZSP integration myself. I suggest to provide any feedback of the experimental EZSP integration here: https://github.com/Koenkk/zigbee-herdsman/issues

gecko17, 2023-02-02:
Thanks for pointing to the zigbee-herdsman repo. I didn't know that the issue is related to the EZSP adapter and that this is maintained in zigbee-herdsman :). I just saw that other people also had the same issue reported in this repo here.

So I will report any further issue in zigbee-herdsman. However, I saw that version 1.30.0 also includes a fix in zigbee-herdsman regarding EZSP adapter, so maybe this already resolved the issue? I will monitor it the next days and report back.

gecko17, 2023-02-06:
@Koenkk is there any way to downgrade the Home Assistant Add-on to version 1.27.2-1 (which seems to be quite stable for EZSP v8)?

Koenkk, 2023-02-06:
@gecko17 AFAIK Home Assistant doesn't support dowgrading addons

and7ey, 2023-02-17:
now got the following new error - 
```
Zigbee2MQTT:debug 2023-02-16 21:43:45: Saving state to file /config/zigbee2mqtt/state.json
TypeError: Cannot read properties of undefined (reading 'version')
    at EZSPAdapterBackup.createBackup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/backup.ts:21:56)
    at EZSPAdapter.backup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:589:31)
    at Controller.backup (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:328:47)
    at runNextTicks (node:internal/process/task_queues:61:5)
    at processTimers (node:internal/timers:499:9)
```

closed 2023-03-28 by github-actions[bot]